shivasiddharth/Assistants-Pi #1: env/bin/activate: No such file or directory

cherokee93, 2017-10-27:
Hey,
tried to install your script with on a fresh raspbian lite image, but i encounter this error at the end when I try to run Google Assistant:
```

================Run the Alexa demo to authenticate============
To run the demo, do the following in 3 seperate terminals:
Run the companion service: cd /home/pi/alexa-avs-sample-app/samples/companionService && sudo npm start
Run the AVS Java Client: cd /home/pi/alexa-avs-sample-app/samples/javaclient && sudo mvn exec:exec
Run the wake word agent:
  KITT_AI: cd /home/pi/alexa-avs-sample-app/samples/wakeWordAgent/src && sudo ./wakeWordAgent -e kitt_ai

===============Run Google Assistant demo===========
To run the google assistant demo
Open a new terminal and execute:
source env/bin/activate
google-assistant-demo

After verifying the working of both, restart the Pi to start the services
pi@raspberrypi:~ $ source env/bin/activate
-bash: env/bin/activate: No such file or directory
```

Any tip?
btw thanks for this setup, can't wait to test it out!

cherokee93, 2017-10-27:
And the third console (the wakeword console) says :
```
INFO:Not connected! Make sure AVS client is running
INFO:Not connected! Make sure AVS client is running
INFO:Not connected! Make sure AVS client is running
INFO:Not connected! Make sure AVS client is running
INFO:Not connected! Make sure AVS client is running
```
On first console:
```
pi@raspberrypi:~ $ cd /home/pi/alexa-avs-sample-app/samples/companionService && sudo npm start

> alexa-voice-service-sample-companion-service@1.0.0 start /home/pi/alexa-avs-sample-app/samples/companionService
> node ./bin/www

This node service needs to be running to store token information memory and vend them for the AVS app.

Listening on port 3000

```

shivasiddharth, 2017-10-27:
1. Looks like the google assistant installation did not even start. Did you give a "no" when you got a prompt asking whether you placed the assistant.json file ?
2. For the snowboy to work, you need to run the javaclient in the second terminal. I guess you dont have the javaclient running.
3. Please use Raspbian desktop image. This is not AlexaPi. Only for ALexaPi, you need to use a Raspbian lite version.

cherokee93, 2017-10-27:
1. I said yes there.
2. Now I found the error in the second console:
```
pi@raspberrypi:~/alexa-avs-sample-app/samples/javaclient $ cd /home/pi/alexa-avs-sample-app/samples/javaclient && sudo mvn exec:exec
Warning: JAVA_HOME environment variable is not set.
[INFO] Scanning for projects...
[INFO]
[INFO] ------------------------------------------------------------------------
[INFO] Building Alexa Voice Service Sample Java Client 20160207.6
[INFO] ------------------------------------------------------------------------
[INFO]
[INFO] --- exec-maven-plugin:1.2.1:exec (default-cli) @ sample-java-client ---
Error: Could not find or load main class com.amazon.alexa.avs.App
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 01:40 min
[INFO] Finished at: 2017-10-27T14:19:43+00:00
[INFO] Final Memory: 7M/18M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.codehaus.mojo:exec-maven-plugin:1.2.1:exec (default-cli) on project sample-java-client: Command execution failed. Process exited with an error: 1 (Exit value: 1) -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
```
3. Ok, I'll use raspbian Desktop image. Can this be done headless via ssh?

shivasiddharth, 2017-10-27:
Use VNC server it would be much better.

cherokee93, 2017-10-27:
I'll give it a try and report later

cherokee93, 2017-10-27:
@shivasiddharth 
All fine until this problem pops up (yes I have put my credentials file in /home/pi/):

```
(env) pi@raspberrypi:~ $ google-assistant-demo 
usage: google-assistant-demo [-h] [--credentials OAUTH2_CREDENTIALS_FILE]
google-assistant-demo: error: argument --credentials: No such file or directory: "/home/pi/.config/google-oauthlib-tool/credentials.json"
(env) pi@raspberrypi:~ $
```

Tried to load the file directly:
```

(env) pi@raspberrypi:~ $ google-assistant-demo --credentials /home/pi/credentials.json
Traceback (most recent call last):
  File "/home/pi/env/bin/google-assistant-demo", line 11, in <module>
    sys.exit(main())
  File "/home/pi/env/local/lib/python2.7/site-packages/google/assistant/__main__.py", line 64, in main
    **json.load(f))
TypeError: __init__() got an unexpected keyword argument 'installed'
```

shivasiddharth, 2017-10-27:
run this within env
`google-oauthlib-tool --client-secrets /home/pi/assistant.json --scope https://www.googleapis.com/auth/assistant-sdk-prototype --save --headless`

You will be asked to authenticate again after that it should be fine..

cherokee93, 2017-10-27:
Works perfectly, thanks a lot. Awesome!

shivasiddharth, 2017-10-27:
The google assistant can do some wonderful things... refer to this https://github.com/shivasiddharth/GassistPi for its features.. Since the issue is solved i am closing this for now..

closed 2017-10-27 by shivasiddharth