shivasiddharth/Assistants-Pi #5: installation issue: PyAudio

vadalp, 2017-11-14:
Hi,  
  I started with a new install of Raspbian Stretch full. Followed your YouTube installation procedure. I do have an AIY kit.

I got an error of PyAudio failed due to BDist Wheel issue. As everything is handled by the script, i saw the error and it flew by.
End result is that Alexa worked. Google Assistant didn't. Google Assistant complained the IS_MUTE flag was set and the input device was busy.

So, i decided to install PyAudio using pip

>pip install PyAudio. 

It went fine but, I guess the whole thing broke after a reboot.

Companion service didn't even start. Once again, the status report complained about input devices being busy.

do you have any idea what is going on?

Did you install with Jesse or Stretch?

V

shivasiddharth, 2017-11-15:
The build failure is a warning nothing to be worried about. Just after that, you will see a line that says pyaudio installed. So you need not install it again. The reason for google assistant's misbehaviour is due to the audio configuration. AIY script deletes the .asoundrc and Raspbian auto-populates the .asoundrc file.

After installing the audio configs and before starting the installation, just make sure that the contents of .asoundrc are same as asound.conf

vadalp, 2017-11-15:
So I tried a clean install. 

The audio-test.sh failed. but it is not clear exactly what is going on. May be you can clear it up.

So, the Assistants-Pi/Startup.wav and /usr/share/sounds/alsa/*.wav files dont play. Give out this error.
===========
Playing WAVE '/usr/share/sounds/alsa/Front_Left.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
aplay: set_params:1299: Sample format non available
Available formats:
- S32_LE
===========
But it plays the mic_test.wav that was recorded.
Playing WAVE 'mic-test.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Stereo

Only format it plays is S32_LE. I am going to proceed with the installation. Sounds more like some settings that can be changed later.

vadalp, 2017-11-16:
So after completing the final steps. I still got the same errors as before this time i captured the error

ALSA lib pcm.c:8403:(snd_pcm_set_params) Rate doesn't match (requested 16000Hz, get 48000Hz)
[16693:16708:ERROR:audio_input_processor.cc(755)] Input error
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED
ON_ASSISTANT_ERROR:
  {'is_fatal': True}
[16693:16709:ERROR:audio_input_processor.cc(755)] Input error
ON_ASSISTANT_ERROR:
  {'is_fatal': True}

shivasiddharth, 2017-11-16:
Copy the contents of asound.conf into asoundrc and try again.

vadalp, 2017-11-16:
I copied the asound.conf content from this site into .asoundrc.

Reboot resulted in the following:
Now Google assistant works perfectly but Alexa wakeword service keeps restarting. Hence,it says "Hello" every few minutes.

shivasiddharth, 2017-11-16:
Did you authenticate the Alexa, before restarting? It is easy to miss out on that ! If not done, then stop the alexa related services using:
```
  sudo systemctl enable companionapp.service
  sudo systemctl enable client.service
  sudo systemctl enable wakeword.service
```
 and manually run the following scripts.
companionstart.sh
clientstart.sh
They are inside the  Assistants-Pi scripts directory

vadalp, 2017-11-16:
Yes. Alexa was authenticated at the time of install. Infact before updating .asoundrc, Alexa services were running fine. but no voice interaction could happen.

shivasiddharth, 2017-11-16:
Then please post the outputs of:
sudo systemctl status companionapp.service
sudo systemctl status client.service
sudo systemctl status wakeword.service

vadalp, 2017-11-16:
● companionapp.service - Companion Service
   Loaded: loaded (/lib/systemd/system/companionapp.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2017-11-16 10:04:30 EST; 55min ago
 Main PID: 372 (bash)
   CGroup: /system.slice/companionapp.service
           ├─372 /bin/bash /home/pi/Assistants-Pi/scripts/companionstart.sh
           ├─398 sudo npm start
           ├─445 npm
           ├─774 sh -c node ./bin/www
           └─775 node ./bin/www

Nov 16 10:04:30 raspberrypi systemd[1]: Started Companion Service.
Nov 16 10:04:31 raspberrypi sudo[398]:       pi : TTY=unknown ; PWD=/home/pi/alexa-avs-sample-app/samples/compan
Nov 16 10:04:31 raspberrypi sudo[398]: pam_unix(sudo:session): session opened for user root by (uid=0)
Nov 16 10:04:39 raspberrypi bash[372]: > alexa-voice-service-sample-companion-service@1.0.0 start /home/pi/alexa
Nov 16 10:04:39 raspberrypi bash[372]: > node ./bin/www
Nov 16 10:04:39 raspberrypi bash[372]: This node service needs to be running to store token information memory a
Nov 16 10:04:41 raspberrypi bash[372]: Listening on port 3000
Nov 16 10:59:32 raspberrypi bash[372]: Successfully retrieved access token for session id:

xxxxxxxxxxxxxxxxxxxxx

● client.service - Client Service
   Loaded: loaded (/lib/systemd/system/client.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2017-11-16 10:04:30 EST; 56min ago
 Main PID: 373 (bash)
   CGroup: /system.slice/client.service
           ├─ 373 /bin/bash /home/pi/Assistants-Pi/scripts/clientstart.sh
           ├─ 406 sudo mvn exec:exec
           ├─ 443 /usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/bin/java -classpath /usr/share/maven/boot/plexus-clas
           └─1004 java -Xbootclasspath/p:/root/.m2/repository/org/mortbay/jetty/alpn/alpn-boot/8.1.6.v20151105/a

Nov 16 10:59:32 raspberrypi bash[373]: Logged in.
Nov 16 10:59:32 raspberrypi bash[373]: Welcome to AVS! To interact with this app, type one of the following comm
Nov 16 10:59:32 raspberrypi bash[373]: Login/Logout: login or logout
Nov 16 10:59:32 raspberrypi bash[373]: To listen: <space>/l/listen, or just press enter!
Nov 16 10:59:32 raspberrypi bash[373]: Next song: n/next
Nov 16 10:59:32 raspberrypi bash[373]: Play/Pause: p/play/pause
Nov 16 10:59:32 raspberrypi bash[373]: Previous song: b/back
Nov 16 10:59:32 raspberrypi bash[373]: Switch locale: en-US/en-GB/de-DE
Nov 16 10:59:32 raspberrypi bash[373]: Quit: q/quit/exit
Nov 16 10:59:32 raspberrypi bash[373]: Display this help text again: h/help

xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

● wakeword.service - Wakeword Service
   Loaded: loaded (/lib/systemd/system/wakeword.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2017-11-16 11:01:20 EST; 44s ago
 Main PID: 1155 (bash)
   CGroup: /system.slice/wakeword.service
           ├─1155 /bin/bash /home/pi/Assistants-Pi/scripts/wakeword.sh
           └─1156 sleep 45

Nov 16 11:01:20 raspberrypi systemd[1]: Started Wakeword Service.

shivasiddharth, 2017-11-16:
your companionapp and client service are running normally as they should.. Which is the actual alexa itself So there is no issue with Alexa, its the just the snowboy which needs to be troubleshooted. 

stop the snowboy service using:
`sudo systemctl status wakeword.service`
open a terminal run the following and getback to me with the output:
```
cd /home/pi/alexa-avs-sample-app/samples
cd wakeWordAgent/src && sudo ./wakeWordAgent -e kitt_ai
```

vadalp, 2017-11-16:
So,
   I stopped the service. 
The directory changes to ~/alexa-avs-sample-app/samples/wakeWordAgent/src
but complains sudo: ./wakeWordAgent/: command not found.

if you are looking for snowboy directory in kitt_ai directory. It does exist.

shivasiddharth, 2017-11-16:
Can you run the installer script again. this time choose to install Alexa alone. Since most of the stuff has already been made it should breeze through this time. During installation watch for the part when it starts cloning snowboy.

vadalp, 2017-11-16:
After Alexa authorization, i tried running the wakeword command. I get the following error:

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ERROR:Failed to open PortAudio stream.Device unavailable
ERROR:KittAiSnowboyWakeWordEngine: Initialization error:Failed to open PortAudio stream.
ERROR:WakeWordAgent: exception in constructor: Failed to open PortAudio stream.
ERROR:main: Exception happened: Failed to open PortAudio stream.
INFO:Shutting down Wake Word Agent

shivasiddharth, 2017-11-16:
Good !! We are almost there. I really appreciate your patience. Sometimes, raspbian midifies the audio configs, so please post the outputs of following: 
```
sudo nano /etc/asound.conf
sudo nano ~/.asoundrc
```

vadalp, 2017-11-16:
sudo nano /etc/asound.conf
no such file exists.

sudo nano ~/,asoundrc
options snd_rpi_googlemihat_soundcard index=0

pcm.softvol {
    type softvol
    slave.pcm dmix
    control {
        name Master
        card 0
    }
}

pcm.micboost {
    type route
    slave.pcm dsnoop
    ttable {
        0.0 30.0
        1.1 30.0
    }
}

pcm.!default {
    type asym
    playback.pcm "plug:softvol"
    capture.pcm "plug:micboost"
}

ctl.!default {
    type hw
    card 0
}

shivasiddharth, 2017-11-16:
Paste the contents of asoundrc into asound.conf. Type the follwoing:
`sudo nano /etc/asound.conf`
paste the following into that and save:
```
options snd_rpi_googlemihat_soundcard index=0

pcm.softvol {
type softvol
slave.pcm dmix
control {
name Master
card 0
}
}

pcm.micboost {
type route
slave.pcm dsnoop
ttable {
0.0 30.0
1.1 30.0
}
}

pcm.!default {
type asym
playback.pcm "plug:softvol"
capture.pcm "plug:micboost"
}

ctl.!default {
type hw
card 0
}
```

this should fix your audio issues and snowboy should run now.

vadalp, 2017-11-16:
Thanks for your patience. Finally it worked!!!

may be you need to tweak the script to copy the asound.conf file to /etc directory.

closed 2017-11-16 by vadalp

shivasiddharth, 2017-11-16:
its already done.. I have intimated the AIY guys also about the .asoundrc issue (16000hz, 48000Hz). https://github.com/google/aiyprojects-raspbian/pull/172