shivasiddharth/Assistants-Pi #6: USB Mic (Playstation Eye) not working

smcsekar, 2017-11-16:
Sid,
A noob here. While testing audio during the installation, aux out speaker works fine but mic recording does not playback (no sound). Shows .wav file directory does not exist. What to do? sorry if i sound silly...

shivasiddharth, 2017-11-16:
open a terminal, run the following command and get back with the error that you see:
`arecord -d 10 -D hw:1,0 -r 16000 -f S16_LE -c1 /home/pi/mic-test.wav`

smcsekar, 2017-11-16:
Started a fresh install from scratch today.. Got the same mic error. Tried the command you suggested got the following below

Great!! Proceeding to test the microphones......

A 10 second audio sample will be recorded for testing. Are you ready?
[y/n] >> y
Select your audio config that you used:
1) AIY-HAT-or-CUSTOM-HAT
2) USB-DAC-or-USB-SOUNDCARD-or-USB-MIC

Please select an option [1-2] 2
Selection: USB-DAC-or-USB-SOUNDCARD-or-USB-MIC
=============Recording Mic Audio Sample=============
Recording WAVE '/home/pi/mic-test.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
arecord: set_params:1305: Channels count non available

Finished recording the samples.

Playing back the recorded audio sample......

/home/pi/mic-test.wav: No such file or directory
Did you hear the recorded audio sample?
[y/n] >> n
Execute arecord -l in terminal and verify the card id with the ones mentioned in asound.conf file and .asoundrc file. Exiting...
pi@raspberrypi:~ $ arecord -d 10 -D hw:1,0 -r 16000 -f S16_LE -c1 /home/pi/mic-test.wav
Recording WAVE '/home/pi/mic-test.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
arecord: set_params:1305: Channels count non available
pi@raspberrypi:~ $

shivasiddharth, 2017-11-16:
I included this audio test just to make sure that your mic works. This does not mean that the installation will not work. You can proceed with the installation. Try this command but.
`arecord -d 10 -D hw:1,0 -r 16000 -f S16_LE  /home/pi/mic-test.wav`

closed 2017-12-05 by shivasiddharth