shivasiddharth/Assistants-Pi #7: Issue with google-assistant-demo

pri-coder, 2017-11-22:
Hi,
I followed all the steps and was successfully able to run Alexa. 

I am having an issue with the google-assistant-demo. Here is the output I get,

env) pi@raspberrypi:~ $ google-assistant-demo
[4469:6245:ERROR:audio_input_processor.cc(755)] Input error
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED
ON_ASSISTANT_ERROR:
  {'is_fatal': True}
[4469:6634:ERROR:audio_input_processor.cc(755)] Input error
ON_ASSISTANT_ERROR:
  {'is_fatal': True}

What could be causing this? Can you help please?

shivasiddharth, 2017-11-22:
Can you provide some info on what is your audio setup (Mic and Audio out)? Also please post the outputs of the following:
```
sudo nano /etc/asound.conf
sudo nano ~/.asoundrc
```

pri-coder, 2017-11-22:
Mic is Playstation Eye and Audio out is through the 3.5mm jack.
`sudo nano /etc/asound.conf 
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
`

`sudo nano ~/.asoundrc
pcm.jack {
  type hw
  card ALSA
}
pcm.array {
  type hw
  card CameraB409241
}
pcm.softvol {
  type softvol
  slave.pcm "jack"
  control {
    name Master
    card 0
  }
}
pcm.cap {
  type plug
  slave.pcm "array"
  slave.channels 4
  ttable {
    0.0 30.0
    1.1 30.0
  }
}
pcm.!default {
    type asym
    playback.pcm "plug:softvol"
    capture.pcm {
      type plug
      slave.pcm "cap"
    }
}
ctl.!default {
  type hw
  card 0
}
ctl.softvol {
  type hw
  card 0
}`

shivasiddharth, 2017-11-22:
Remove the existing contents of the asoundrc, copy the contents of asound.conf and paste it in asoundrc.

pri-coder, 2017-11-22:
I rebooted and now Google Assistant is working but having an issue with Alexa. It keeps saying Hello from time to time but does not respond when I say Alexa. I tried starting up the companion service and I got a Port 3000 is already in use error.

pri-coder, 2017-11-22:
I started the install from scratch and now both the assistants are working as expected. Thanks!

closed 2017-11-22 by pri-coder