shivasiddharth/Assistants-Pi #8: Play music from YouTube

pri-coder, 2017-11-22:
Is there an option to enable playing music from YouTube on a voice command with this install?

shivasiddharth, 2017-11-22:
Here is all that you can do:

1. Voice control of GPIOs without IFTTT, api.ai, Actions SDK.
2. Voice control of NodeMCU without IFTTT and MQTT.
3. Radio streaming.
4. Voice control of servo connected to RPi GPIO.
5. Safe shutdown RPi using voice command.
6. Stream Music from YouTube.

But these are restricted to Google assistant alone. Amazon did not give me permission to customize Alexa. .. For more details on how the commands work, please visit this page. https://github.com/shivasiddharth/GassistPi

pri-coder, 2017-11-22:
Ok. Do I need to install GassistPi or can I make the config changes to the Assistants-Pi project?

shivasiddharth, 2017-11-22:
This installation uses GassistPi only. So there is nothing you need to do. Everything is already done and is ready to be used.

pri-coder, 2017-11-22:
Oh ok. Do I need to enable anything to make the Play voice command work?

pri-coder, 2017-11-22:
Saying Play based in your example doesn't play anything from YouTube. It just plays a 2 second music sample.

shivasiddharth, 2017-11-22:
Ok, my mistake. I had forgotten to change a line  in the installation script here. I have fixed it now.  Open a terminal and type:
`mpsyt set player vlc, set playerargs ,exit`
now the play command should work fine.

pri-coder, 2017-11-22:
Ran these commands but the play command still does not work

shivasiddharth, 2017-11-22:
Open the actions.py file which in /home/pi/GassistPi/src directory.  Scroll to` def YouTube(phrase):` and within that down below, you will have a line `playshell.stdin.write(bytes('/' + track + '\n1\n'))` replace that with `playshell.stdin.write(bytes('/' + track + '\n1\n','utf-8'))`

An oversight which had forgotten to correct.

pri-coder, 2017-11-23:
I have the Assistants Pi code base which does not have an src folder

pri-coder, 2017-11-23:
Found the GassistPi in home/Pi as well and made the changes you suggested. Play command now works. Thanks!

closed 2017-11-23 by pri-coder