shivasiddharth/Assistants-Pi #9: Alexa demo works but google does not

KGunther, 2017-11-24:
Hi,
   Sorry to bother you but I am having problems. I did the installation from a fresh Raspbian Stretch install.

When I try to run the Google demo I get the following error:
pi@raspberrypi:~ $ source env/bin/activate
(env) pi@raspberrypi:~ $ google-assistant-demo
ALSA lib pcm.c:8403:(snd_pcm_set_params) Rate doesn't match (requested 16000Hz, get 48000Hz)
[16409:16424:ERROR:audio_input_processor.cc(755)] Input error
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED
ON_ASSISTANT_ERROR:
  {'is_fatal': True}
[16409:16425:ERROR:audio_input_processor.cc(755)] Input error
ON_ASSISTANT_ERROR:
  {'is_fatal': True}

I looked at some of the similar issues and you asked for /etc/asound.conf, this file does not exist. Here is my .asoundrc file:
pcm.!default {
	type hw
	card 0
}

ctl.!default {
	type hw
	card 0
}

shivasiddharth, 2017-11-24:
I suppose, you are using AIY hat ?
Paste the following in both asound.conf and asoundrc.

options snd_rpi_googlemihat_soundcard index=0

pcm.softvol {
    type softvol
    slave.pcm dmix
    control {
        name Master
        card 0
    }
}

pcm.micboost {
    type route
    slave.pcm dsnoop
    ttable {
        0.0 30.0
        1.1 30.0
    }
}

pcm.!default {
    type asym
    playback.pcm "plug:softvol"
    capture.pcm "plug:micboost"
}

ctl.!default {
    type hw
    card 0
}

KGunther, 2017-11-25:
It is working now, thank you very much. I will save an image of the working system and then try custom wakewords. Your support is very much appreciated

closed 2017-11-27 by KGunther