shivasiddharth/Assistants-Pi #12: Folder locations and folder usage with new AssistantsPi setup

supreme83, 2018-01-08:
continued from GassistPi issue..............Unfortunatley Google AIY was limited usage so i decided to try the https://github.com/shivasiddharth/Assistants-Pi setup. It works great with both alexa and aiy but is it installing the same directories? I ask because GassistPi did not seem to need to be deleted and works along side the Assistants setup but AIY was constantly triggered by my tv and wasting api quota for aiy, so i first renamed entire GassistPi file to Gassist.backup which made both Alexa and AIY stop working on next boot(no recognition on "Alexa" or "OK Google" (no custom wake word set). I renamed the file to original name and things worked on next boot.....I then instead renamed the GassistPi/src file to GassistPi/backup.src and Amazon Alexa works but only with my newly created custom wake word "Oasis". Again, my question is about the GAssistPi folder (i ask because these are both your developments and i see AssistantsPi did use the same "/home/pi/alexa-avs-sample-app" directory as alexa-avs-sample folder git would have. Is GassistPi part of the AssistantsPi setup or an unrelated folder?

shivasiddharth, 2018-01-08:
Background:
I wanted to modify Alexa so that i could combine both the Google Assistant and Alexa. I had contacted Amazon for obtaining permission to modify it just like how i did the Google Assistant but they refused permission for modifications and redistribution, they would not even allow me to modify a file and redistribute.
As a result of the refused permission, i scripted the Assistants-Pi. Assistants-Pi is an installer script which integrates alexa-avs-sample-app and GassistPi. Basically Assistants-Pi is an integrator, so the Assistants-Pi folder will have stuff that is common to both Alexa and GassistPi like audio drivers, startup and installer scripts. But the core scripts/codes will be in the respective alexa-avs-sample-app and GassistPi folders.
In short, GassistPi+alexa-avs-sample-app = Assistants-Pi

supreme83, 2018-01-09:
i understand that GassistPi+alexa-avs-sample-app = Assistants-Pi  ...what i'm asking about is file structure for the 3 gits as they relate to this project....In short. having an existing Gassist install in /home/pi/GassistPi and then using the installer script for AssistantsPI. noticeably installed /home/pi/alexa-avs-sample-app and /home/pi/Assistants-Pi ....Did that use my existing /home/pi/GassistPi/src folder for actions.py or is that actions.py replaced by something inside of one of the other directories? For any of my sound drivers to work GassistPi filename can not be changed to i can only assume the file is used in the current scripting? is this true or is actions.py defined elsewhere now?

shivasiddharth, 2018-01-09:
If you already have a GassistPi folder, it will use that only. There will be no other additional directories. GassistPi installed using the GassistPi git and the GassistPi installed using Assistants-Pi git will both use the same directory only. Hope that clears your doubts. If you take a look into the Assistants-Pi installer script you can see that it clones a branch of GassistPi git and so obviously, it will have the same folder name.

supreme83, 2018-01-09:
ok thanks. that is what i was asking. I did figure out today to look at the script and saw that it was indeed cloning the gassistpi git to the same directory. Thanks again. You can close this

closed 2018-01-09 by supreme83