shivasiddharth/Assistants-Pi #13: Kodi control

iristvproject, 2018-01-12:
Sir, I'm not installed this on my rpi 3b but before starting I want to confirm whether I can control kodi using Google assistant in this build. And if yes, what are the few examples that I can try? 

Thanks in advance.

shivasiddharth, 2018-01-12:
This build supports Kodi control. Please take a look at this README doc for syntax.:https://github.com/shivasiddharth/GassistPi

iristvproject, 2018-01-12:
After installation completed while checking the demo app we encountered few problems. 
1. Companion service worked but javavlient didn't run... It says mvn is not found. 
2. Google model id error, I copied from model id file and pasted but it says its wrong. 

Any advice?

closed 2018-01-12 by iristvproject