shivasiddharth/Assistants-Pi #15: Volume control for Google Voice HAT

TedMichalik, 2018-01-12:
I have both Alexa and the Google Assistant running on the Google Voice kit. The volume of the Alexa response is twice as loud as that of the Google Assistant. I tried lowering the volume in the Raspbian desktop, but that also lowered the Google Assistant volume to a barely audible level. Is there a way to adjust the volume of either assistant that is independent of the other?

shivasiddharth, 2018-01-13:
Alexa's volume is managed by VLC. open a vlc player, reduce the volume and restart alexa, hopefully that should have some effect.

jabulonboaz, 2018-02-03:
I had the same issue and fixed it by lowering the volume at the preferences and then asked google to “hey google volume maximum”. Now both assistants are at the same level.

commit 8cda70cb (2018-10-12): Migrated the project to AVS SDK

Fixes #3, Fixes #15, Fixes #69 , Fixes #66 , Fixes #65 , Fixes #64 , Fixes #62 , Fixes #61 , Fixes #60 , Fixes #59 , Fixes #58 , Fixes #57 , Fixes #56 , Fixes #55 , Fixes #54 , Fixes #53 , Fixes #52 , Fixes #51 , Fixes #49 , Fixes #46 , Fixes #38 , Fixes #37 , Fixes #33 , Fixes #20 , Fixes #19 , Fixes #18 , Fixes #16
closed 2018-10-12 by shivasiddharth