shivasiddharth/Assistants-Pi #16: building failure

rithunrosh, 2018-01-14:
hello suiddharth recently i had run your code, but the following error has been occured , and i did not get any device token . so please help me.

 Scanning for projects...
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building Alexa Voice Service Sample Java Client 20160207.7
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- exec-maven-plugin:1.2.1:exec (default-cli) @ sample-java-client ---
Error: Could not find or load main class com.amazon.alexa.avs.App
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 4.433 s
[INFO] Finished at: 2018-01-14T19:18:02+00:00
[INFO] Final Memory: 8M/20M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.codehaus.mojo:exec-maven-plugin:1.2.1:exec (default-cli) on project sample-java-client: Command execution failed. Process exited with an error: 1 (Exit value: 1) -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
pi@raspberrypi:~/alexa-avs-sample-app/samples/javaclient $

shivasiddharth, 2018-01-22:
Try executing the following one after another in terminal:
```
cd /home/pi/alexa-avs-sample-app/samples
cd javaclient
mvn clean
mvn install
mvn validate
mvn exec:exec
```

rithunrosh, 2018-01-23:
Sorry Sidharth , it again shows that " BUILDING FAILURE"

shivasiddharth, 2018-01-23:
I have updated the git with fixes for the installation issues. Delete the existing Assistants-Pi folder, reclone the git run the installer again.

jabulonboaz, 2018-01-24:
Hello Sidharth  AVS now works but Google assistant won't start and when i follow all the instructions i get an error on audio too. In a terminal when i go into evn/bin/active and i run (BTW this wont execite from raspberrypi directory) "googlesamples-assistant-hotword.........with my id and model id then i get this "ERROR: audio_input_processor.cc(748) input error.
Any ideas? I do not understand it was working fine all other times in every build with the same steps, i did the same steps as always even yesturday the Google assistant build was working but tonight Google assitant wont verify or star. When tryed to re install and execute the installer script to re install Google again it starts it gets to the point of the authentication URL link i authenticate then it runs for a bit and it just closes now. It did not do this the first time i run it today however.   
Also even though AVS works fine now (so thank you) however it does not say "hello" like it use to at boot maybe this is related to the first problem of google assistant not starting at boot nether being able to verify it running? 
Any chance to look into the google assitant issue ?

shivasiddharth, 2018-01-24:
@jabulonboaz , The audio issues are because of empty asound.conf and auto populated asoundrc. Copy the contents of the asound.conf file that i have given with this project, paste it into the system's asound.conf and .asoundrc. After that, open a terminal and run
`speaker-test`
Stop the test and restart, after that things should be back to normal.

jabulonboaz, 2018-01-24:
@Sidharth thank you for this, could u pls kindly remind me the command to get my audio HW config, is it "asound -l" or "aplay -l" something like it? And then with the values i get how do i use them to populate your asound.conf ? Do i replace all "0"s with "1"s for instance if i get 1/1?
Apologies but i am not at your level.

shivasiddharth, 2018-01-24:
@jabulonboaz , 
I never asked you to change the asound.conf file i have given. Please dont do that. All that i wanted you to do is the following:

First edit the asound.conf using:
`sudo nano /etc/asound.conf`
Paste into the window the contents of this file: https://github.com/shivasiddharth/Assistants-Pi/blob/master/audio-drivers/AIY-HAT/scripts/asound.conf  
Save that and there after, edit the .asoundrc using:
`sudo nano ~/.asoundrc`
Paste into the window the contents of this file: https://github.com/shivasiddharth/Assistants-Pi/blob/master/audio-drivers/AIY-HAT/scripts/asound.conf  
Save that and then execute the following to initialize alsa:
`speaker-test`
Now if you restart, things should be back to normal.

jabulonboaz, 2018-01-24:
I hear a very loud white noise

shivasiddharth, 2018-01-24:
@jabulonboaz That is what the `speaker-test` command will give. Please read through this reference material http://manpages.ubuntu.com/manpages/trusty/man1/speaker-test.1.html
If you want something more pleasant run:
`speaker-test -t wav`
Our goal is just to initialize alsa, so the command is immaterial.

jabulonboaz, 2018-01-24:
Success! Thank u! Both assistants work now.
How can i lower Alexa's volume please she is too loud, do i need to instal VLC to control it? 
And how can i make the AYI hat button to work too so that i can trigger Google assistant with the button and hotword?
Thank u again!

jabulonboaz, 2018-01-24:
I started playing with the audio preferences trying to bring Alexa audio down but now i can hear her neither Google. I can hear the chime like sound when i say Alexa.

shivasiddharth, 2018-01-24:
Open a terminal and type `vlc` that will open the vlc player change the volume there and a restart should bring alexa to the new volume (I have not tried it, so if it works it works, else not). The pushbutton has been re-purposed to stop music streaming for GassistPi. That button kills vlc, so in all likely-hood it could crash Alexa (Again i have not tested it). 
In the next update to GassistPi, all internet content streaming will be switched to another player called mpv, just to isolate vlc and Alexa for Assistants-Pi users so that the pushbutton may not interfere with Alexa.

shivasiddharth, 2018-01-24:
It can now be confirmed that the build failure error has been fixed. I shall leave the thread open for @rithunrosh to test. confirm and close the issue.

jabulonboaz, 2018-01-24:
But nothing works now nor the assistant nor AVS, there is no sound. So if someone has an AIY hat and presses the button it breaks everything and it crashes? How can i get them to work again please? 
I tried to do a test again but its not working ether:
pi@raspberrypi:~ $ speaker-test
speaker-test 1.1.3

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Sample format not available for playback: Invalid argument
Setting of hwparams failed: Invalid argument
pi@raspberrypi:~ $ ^C
pi@raspberrypi:~ $

jabulonboaz, 2018-01-24:
Hello SId Google assistant stopped working i get this error
   pi@raspberrypi:~ $ speaker-test

speaker-test 1.1.3

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Sample format not available for playback: Invalid argument
Setting of hwparams failed: Invalid argument

shivasiddharth, 2018-01-24:
@jabulonboaz , disable the stopbutton service using:
`sudo systemctl disable stopbutton.service`
Reboot to disable the button.
Check your .asoundrc and see if it has changed. I would really advise you to not to "Play around" something which could crash the project. I really dont have time for helping fix sabotages.

jabulonboaz, 2018-01-24:
Hello @shivasiddharth believe me if i knew it would break my project i would have never pressed the button, i which i knew that before. I did not exactly play around i was trying to lower Alexa's volume as it is too loud and it does not respond to volume voice commands. Have been trying to make it work for some time now since the build for Alexa stopped working. 
I tried sudo systemctl disable stopbutton.service  and the rebooted but same result Goggle assistant does not work now. any thought?

shivasiddharth, 2018-01-24:
@jabulonboaz Golden rule of thumb if GA does not work:
Check the .asoundrc and asound.conf and then if they are off follow this.
https://github.com/shivasiddharth/Assistants-Pi/issues/16#issuecomment-360086100

jabulonboaz, 2018-01-24:
As i have explained i did that as you instructed they are ok but Google assistant is not working:
GNU nano 2.7.4                File: /etc/asound.conf                Modified

options snd_rpi_googlemihat_soundcard index=0

pcm.softvol {
    type softvol
    slave.pcm dmix
    control {
        name Master
        card 0
    }
}

pcm.micboost {
    type route
    slave.pcm dsnoop
    ttable {
        0.0 30.0
        1.1 30.0
    }
}

pcm.!default {
    type asym
    playback.pcm "plug:softvol"
    capture.pcm "plug:micboost"
}

ctl.!default {
    type hw
    card 0
}

pcm.!default {
        type hw
        card 0
}

ctl.!default {
        type hw
        card 0
}

shivasiddharth, 2018-01-24:
@jabulonboaz Come on, see that again. How is that ok? Its not ok!  See the last 9 lines. Is that what you see here: https://github.com/shivasiddharth/Assistants-Pi/blob/master/audio-drivers/AIY-HAT/scripts/asound.conf I am going to stop assisting you at this point.

jabulonboaz, 2018-01-24:
Apologies for some reason the copy passing add extra lines that is the correct pasting and it is identical with your file: 
  GNU nano 2.7.4                File: /etc/asound.conf                Modified

options snd_rpi_googlemihat_soundcard index=0

pcm.softvol {
    type softvol
    slave.pcm dmix
    control {
        name Master
        card 0
    }
}

pcm.micboost {
    type route
    slave.pcm dsnoop
    ttable {
        0.0 30.0
        1.1 30.0
    }
}

pcm.!default {
    type asym
    playback.pcm "plug:softvol"
    capture.pcm "plug:micboost"
}

ctl.!default {
    type hw
    card 0
}

shivasiddharth, 2018-01-24:
@jabulonboaz Can you please check .asoundrc as well? In all likely-hood what you previously posted must be from .asoundrc. Only that will get appended. If .asoundrc and asound.conf are different and do not match, then you copy the contents from asound.conf and paste it in .asoundrc, not the other way around.

jabulonboaz, 2018-01-24:
Both files are identical, i attach a picture for you here. [image]

shivasiddharth, 2018-01-24:
Paste here the outputs of:
```
aplay -l
arecord -l
sudo systemctl status gassistpi-ok-google.service
```

shivasiddharth, 2018-01-24:
@jabulonboaz What is that first line in the .asoundrc? "GNU nano 2.74..... blah blah blah ". Guess you have copied the entire window name of asound.conf file. Please be careful when you paste stuff and make sure to check and double check.

jabulonboaz, 2018-01-24:
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ 

pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-hifi-0 []
  Subdevices: 0/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ arecord -l

● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-01-24 12:37:21 UTC; 40min ago
 Main PID: 434 (python)
   CGroup: /system.slice/gassistpi-ok-google.service
           └─434 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id plucky-alliance-188518 --device_model_id

Jan 24 12:37:27 raspberrypi python[434]: ALSA lib pcm.c:8403:(snd_pcm_set_params) Rate doesn't match (requested 16000Hz, get 480
Jan 24 12:37:28 raspberrypi python[434]: device_model_id: plucky-alliance-188518-20180124000250
Jan 24 12:37:28 raspberrypi python[434]: device_id: BC586C7133CDFF977B9CA950737EC6C1
Jan 24 12:37:28 raspberrypi python[434]: https://embeddedassistant.googleapis.com/v1alpha2/projects/plucky-alliance-188518/devic
Jan 24 12:37:29 raspberrypi python[434]: ON_MUTED_CHANGED:
Jan 24 12:37:29 raspberrypi python[434]:   {'is_muted': False}[434:809:ERROR:audio_input_processor.cc(748)] Input error
Jan 24 12:37:29 raspberrypi python[434]: ON_MUTED_CHANGED:
Jan 24 12:37:29 raspberrypi python[434]:   {'is_muted': False}
Jan 24 12:37:29 raspberrypi python[434]: ON_START_FINISHED
Jan 24 12:37:31 raspberrypi python[434]: [434:866:ERROR:audio_input_processor.cc(748)] Input error

jabulonboaz, 2018-01-24:
I have no idea @shivasiddharth what GNU nano is it just there when i type  sudo nano ~/.asoundrc

jabulonboaz, 2018-01-24:
It is all working now @shivasiddharth and i have disconnected the button switch just in case.
After i did a reboot i cheeked again the 2 audio files and one of the had an extra paragraph in it i deleted it and compared the 2 side by side and then the audio test works. I rebooted again and the config is in place. i am not touching it gain... thank you.
I will install a potentiometer on my speaker to be able to control the volume though as AVS sound cannot be lowered with a voice command. correct?

shivasiddharth, 2018-01-24:
@jabulonboaz In the next update, i shall migrate to a different media player, so then you should be able to use the pushbutton without disrupting alexa. Feel free to try to lower the volume by voice and see if it works (I have not tried it). Since everything is working, i would suggest you to make a copy of the sd card (there are guides in Raspbian forums to help you out for that).

jabulonboaz, 2018-01-24:
@shivasiddharth thank you one more time, i cant thank you enough for this wonderful work and assistance. 
I have installed a variable resistance to control the value AVS volume as it does not respond to audio commands it seams. 
Yes this is a brilliant idea i will clone my SD card with this project!  
Looking forward for the new media player improvement so I can reconnect the button.  
Q: As it stands now it is my understanding from your Git and Youtube channel that only GiassiPi supports music streaming correct? 
If i may suggest 3 future improvements for this project 
1. Stream google play music or youtube (or Spotify)  like in GiassisPi porject in AssistansPi for GA and AVS
2. Have the AYI hat switch to be functional and the LED come on for both AVS and GA. 
3. Enable AVS to be able to control the associated smarthome devices like GA can- AVS on AsstansPi cant detect any of my devices - (But perhaps i need to delete them and re-associate them to my Amazon account for this to work but it annoying - GA on AssistancePi does that automatically.) 
FYI As a side note in the past and now AVS is unable to pick up my address form the Alexa app under settings for my Pi so i get no local weather even though i get correct time.  Also when AVS streams news from say Reutes it wont respond to any command like "Alexa stop" until it finishes 
Any time at your disposal if you need me to test anything for you with the AIY hat.

rithunrosh, 2018-01-26:
Hello Sidharth, I didn't get any token from the Amazon.

shivasiddharth, 2018-01-26:
@rithunrosh , Can you format you SD card and try again.?

v1n0ds, 2018-04-29:
@shivasiddharth After cloning the latest files, everything went fine until executing the last step which ended up showing **Build Failure**
Below are error messages

[ERROR] Failed to execute goal org.apache.maven.plugins:maven-resources-plugin:2.6:resources (default-resources) on project sample-java-client: /home/pi/alexa-avs-sample-app/samples/javaclient/target/classes/res/version.properties (Permission denied) -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException

Please help as i'm using it to build my project.
Thanks

v1n0ds, 2018-04-29:
@rithunrosh How did you rectify **build failure** error? 
please help

commit 8cda70cb (2018-10-12): Migrated the project to AVS SDK

Fixes #3, Fixes #15, Fixes #69 , Fixes #66 , Fixes #65 , Fixes #64 , Fixes #62 , Fixes #61 , Fixes #60 , Fixes #59 , Fixes #58 , Fixes #57 , Fixes #56 , Fixes #55 , Fixes #54 , Fixes #53 , Fixes #52 , Fixes #51 , Fixes #49 , Fixes #46 , Fixes #38 , Fixes #37 , Fixes #33 , Fixes #20 , Fixes #19 , Fixes #18 , Fixes #16
closed 2018-10-12 by shivasiddharth