shivasiddharth/Assistants-Pi #17: not starting on boot ?

nikki-reprap, 2018-01-14:
Hi When my system boots it says "hello" (I have only installed alexa) and then it does nothing I am using the google aiy hat. This is the output

pi@raspberrypi:~ $ sudo systemctl status gassistpi-ok-google.service
● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled)
   Active: failed (Result: start-limit) since Sun 2018-01-14 22:42:59 UTC; 1min 7s ago
  Process: 642 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id project-i                                                                                                      d --device_model_id saved-model-id (code=exited, status=203/EXEC)
 Main PID: 642 (code=exited, status=203/EXEC)

Jan 14 22:42:58 raspberrypi systemd[1]: gassistpi-ok-google.service: main process exited, code=exi...EXEC
Jan 14 22:42:58 raspberrypi systemd[1]: Unit gassistpi-ok-google.service entered failed state.
Jan 14 22:42:59 raspberrypi systemd[1]: gassistpi-ok-google.service holdoff time over, scheduling ...art.
Jan 14 22:42:59 raspberrypi systemd[1]: Stopping Google Assistant-Voice...
Jan 14 22:42:59 raspberrypi systemd[1]: Starting Google Assistant-Voice...
Jan 14 22:42:59 raspberrypi systemd[1]: gassistpi-ok-google.service start request repeated too qui...art.
Jan 14 22:42:59 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.
Jan 14 22:42:59 raspberrypi systemd[1]: Unit gassistpi-ok-google.service entered failed state.
Hint: Some lines were ellipsized, use -l to show in full.

shivasiddharth, 2018-01-22:
You have presented the status of wrong service. Can you kindly provide the status of Alexa's companionapp, client and wakeword service

shivasiddharth, 2018-02-05:
Closing this due to inactivity

closed 2018-02-05 by shivasiddharth