shivasiddharth/Assistants-Pi #23: Alexa Not Starting at Boot

nitinchauhan83, 2018-01-25:
Hi Shiva,

Thank you for your hard work in creating this project.

I have been trying to make google and alexa work together. First I installed Gassitpi, which works flawlessly. Now I have been trying to install Alexa, but running into an error.  I was able to follow instructions until very end, but I am getting an error at the last step for wakeword.sh. While executing: 
cd wakeWordAgent/src && sudo ./wakeWordAgent -e kitt_ai
I am getting error:
sudo: ./wakeWordAgent: command not found
After this error, I ran remaining commands and rebooted my Pi assuming that things will work. Google assistant boots up but Alexa doesn't. However at every 45 sec interval I am hearing "Hello", which i believe is from Alexa.

Regards,
Nitin

PS: I am first time user of Pi and linux environment

nitinchauhan83, 2018-01-25:
npm ERR! alexa-voice-service-sample-companion-service@1.0.0 start: `node ./bin/www`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the alexa-voice-service-sample-companion-service@1.0.0 start script.
npm ERR! This is most likely a problem with the alexa-voice-service-sample-companion-service package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node ./bin/www
npm ERR! You can get their info via:
npm ERR!     npm owner ls alexa-voice-service-sample-companion-service
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 4.9.59-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! cwd /home/pi/alexa-avs-sample-app/samples/companionService
npm ERR! node -v v4.8.2
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/alexa-avs-sample-app/samples/companionService/npm-debug.log
npm ERR! not ok code 0

shivasiddharth, 2018-01-25:
Please paste the outputs of:
```
sudo systemctl status companionapp.service
sudo systemctl status client.service
sudo systemctl status wakeword.service
```

nitinchauhan83, 2018-01-25:
For: sudo systemctl status companionapp.service

companionapp.service - Companion Service
   Loaded: loaded (/lib/systemd/system/companionapp.service; enabled; vendor pre
   Active: active (running) since Thu 2018-01-25 00:36:24 PST; 19min ago
 Main PID: 388 (bash)
   CGroup: /system.slice/companionapp.service
           ├─388 /bin/bash /home/pi/Assistants-Pi/scripts/companionstart.sh
           ├─410 sudo npm start
           ├─434 npm
           ├─781 sh -c node ./bin/www
           └─782 node ./bin/www

Jan 25 00:36:25 raspberrypi sudo[410]: pam_unix(sudo:session): session opened fo
Jan 25 00:36:30 raspberrypi bash[388]: > alexa-voice-service-sample-companion-se
Jan 25 00:36:30 raspberrypi bash[388]: > node ./bin/www
Jan 25 00:36:31 raspberrypi bash[388]: This node service needs to be running to 
Jan 25 00:36:33 raspberrypi bash[388]: Listening on port 3000
Jan 25 00:37:15 raspberrypi bash[388]: Successfully retrieved access token for s
Jan 25 00:40:21 raspberrypi systemd[1]: [/lib/systemd/system/companionapp.servic
Jan 25 00:40:21 raspberrypi systemd[1]: [/lib/systemd/system/companionapp.servic
Jan 25 00:40:21 raspberrypi systemd[1]: [/lib/systemd/system/companionapp.servic
Jan 25 00:49:38 raspberrypi bash[388]: Successfully retrieved access token for s
lines 1-21/21 (END)...skipping...
● companionapp.service - Companion Service
   Loaded: loaded (/lib/systemd/system/companionapp.service; enabled; vendor pre
   Active: active (running) since Thu 2018-01-25 00:36:24 PST; 19min ago
 Main PID: 388 (bash)
   CGroup: /system.slice/companionapp.service
           ├─388 /bin/bash /home/pi/Assistants-Pi/scripts/companionstart.sh
           ├─410 sudo npm start
           ├─434 npm
           ├─781 sh -c node ./bin/www
           └─782 node ./bin/www

Jan 25 00:36:25 raspberrypi sudo[410]: pam_unix(sudo:session): session opened fo
Jan 25 00:36:30 raspberrypi bash[388]: > alexa-voice-service-sample-companion-se
Jan 25 00:36:30 raspberrypi bash[388]: > node ./bin/www
Jan 25 00:36:31 raspberrypi bash[388]: This node service needs to be running to 
Jan 25 00:36:33 raspberrypi bash[388]: Listening on port 3000
Jan 25 00:37:15 raspberrypi bash[388]: Successfully retrieved access token for s
Jan 25 00:40:21 raspberrypi systemd[1]: [/lib/systemd/system/companionapp.servic
Jan 25 00:40:21 raspberrypi systemd[1]: [/lib/systemd/system/companionapp.servic
Jan 25 00:40:21 raspberrypi systemd[1]: [/lib/systemd/system/companionapp.servic
Jan 25 00:49:38 raspberrypi bash[388]: Successfully retrieved access token for s
~
~
~
~
~

nitinchauhan83, 2018-01-25:
for: sudo systemctl status client.service

client.service - Client Service
   Loaded: loaded (/lib/systemd/system/client.service; enabled; vendor preset: e
   Active: active (running) since Thu 2018-01-25 00:36:24 PST; 21min ago
 Main PID: 393 (bash)
   CGroup: /system.slice/client.service
           ├─ 393 /bin/bash /home/pi/Assistants-Pi/scripts/clientstart.sh
           ├─ 411 sudo mvn exec:exec
           ├─ 433 /usr/lib/jvm/java-8-oracle/bin/java -classpath /usr/share/mave
           └─1190 java -Xbootclasspath/p:/root/.m2/repository/org/mortbay/jetty/

Jan 25 00:37:16 raspberrypi bash[393]: To listen: <space>/l/listen, or just pres
Jan 25 00:37:16 raspberrypi bash[393]: Next song: n/next
Jan 25 00:37:16 raspberrypi bash[393]: Play/Pause: p/play/pause
Jan 25 00:37:16 raspberrypi bash[393]: Previous song: b/back
Jan 25 00:37:16 raspberrypi bash[393]: Switch locale: en-US/en-GB/de-DE/en-IN/ja
Jan 25 00:37:16 raspberrypi bash[393]: Quit: q/quit/exit
Jan 25 00:37:16 raspberrypi bash[393]: Display this help text again: h/help
Jan 25 00:40:21 raspberrypi systemd[1]: [/lib/systemd/system/client.service:10] 
Jan 25 00:40:21 raspberrypi systemd[1]: [/lib/systemd/system/client.service:10] 
Jan 25 00:40:21 raspberrypi systemd[1]: [/lib/systemd/system/client.service:10] 
lines 1-20/20 (END)

nitinchauhan83, 2018-01-25:
For: sudo systemctl status wakeword.service

wakeword.service - Wakeword Service
   Loaded: loaded (/lib/systemd/system/wakeword.service; enabled; vendor preset:
   Active: active (running) since Thu 2018-01-25 00:59:02 PST; 40s ago
 Main PID: 4186 (bash)
   CGroup: /system.slice/wakeword.service
           ├─4186 /bin/bash /home/pi/headless-alexa-avs-sample-app/scripts/wakew
           └─4187 sleep 45

Jan 25 00:59:02 raspberrypi systemd[1]: Started Wakeword Service.

shivasiddharth, 2018-01-25:
Everything is running fine. To fix the wakeword, open the wakeword.sh file. when you double click, it will ask to execute in terminal or open, choose to open. Change the last line from
`cd wakeWordAgent/src && sudo ./wakeWordAgent -e kitt_ai`
to 
`cd wakeWordAgent/src && ./wakeWordAgent -e kitt_ai`

close, save and run the following in the terminal to restart the service
`sudo systemctl restart wakeword.service`

Now your wakeword service should work.

nitinchauhan83, 2018-01-25:
Hi Shiva,

I tried that but same issue. below is the error:
pi@NitinPi:~ $ #!/bin/bash
pi@NitinPi:~ $ cd /home/pi/alexa-avs-sample-app/samples
pi@NitinPi:~/alexa-avs-sample-app/samples $ sleep 45
pi@NitinPi:~/alexa-avs-sample-app/samples $ aplay /home/pi/Assistants-Pi/Startup.wav
Playing WAVE '/home/pi/Assistants-Pi/Startup.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
pi@NitinPi:~/alexa-avs-sample-app/samples $ cd wakeWordAgent/src && ./wakeWordAgent -e kitt_ai
bash: ./wakeWordAgent: No such file or directory

i checked the directory but could not find "./wakeWordAgent". however "WakeWordAgent.h" exist.

shivasiddharth, 2018-01-25:
Looks like the wakewordagent has not got compiled. You need to re run the installation script. What version of OS are you running and what version of Raspberry Pi are you running ?

nitinchauhan83, 2018-01-25:
Raspberry Pi: Raspberry Pi 3 Model B Quad-Core 1.2 GHz 1 GB RAM
OS: Raspbian - I will share model number after few hours.

But thank you for looking into it. Really appreciate.

nitinchauhan83, 2018-01-26:
I am trying to reinstall Assistants-Pi. Do I need uninstall the previous version?

nitinchauhan83, 2018-01-26:
pi@NitinPi:~ $ cd /home/pi/alexa-avs-sample-app/samples/companionService && sudo npm start

> alexa-voice-service-sample-companion-service@1.0.0 start /home/pi/alexa-avs-sample-app/samples/companionService
> node ./bin/www

This node service needs to be running to store token information memory and vend them for the AVS app.

Port 3000 is already in use

npm ERR! alexa-voice-service-sample-companion-service@1.0.0 start: `node ./bin/www`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the alexa-voice-service-sample-companion-service@1.0.0 start script.
npm ERR! This is most likely a problem with the alexa-voice-service-sample-companion-service package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node ./bin/www
npm ERR! You can get their info via:
npm ERR!     npm owner ls alexa-voice-service-sample-companion-service
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 4.9.59-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! cwd /home/pi/alexa-avs-sample-app/samples/companionService
npm ERR! node -v v4.8.2
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/alexa-avs-sample-app/samples/companionService/npm-debug.log
npm ERR! not ok code 0
pi@NitinPi:~/alexa-avs-sample-app/samples/companionService $

nitinchauhan83, 2018-01-26:
It worked. Thank you, Shiva! Hats off to you

shivasiddharth, 2018-01-26:
Ok, closing this issue then !

closed 2018-01-26 by shivasiddharth