shivasiddharth/Assistants-Pi #27: google-assistant-demo command giving error

ivaturianil, 2018-01-29:
@shivasiddharth , I am getting below error when i run google-assistant-demo command. I tried to register the model using "googlesamples-assistant-devicetool & register-model" but getting error like "Failed to register model: 400". My assistant.json is in home/pi location. Any help in this regard is appreciated. 
usage: google-assistant-demo [-h] [--credentials OAUTH2_CREDENTIALS_FILE]
                             --device_model_id DEVICE_MODEL_ID
                             [--project_id PROJECT_ID]             
google-assistant-demo: error: the following arguments are required: --device_model_id

ivaturianil, 2018-01-29:
@shivasiddharth , I tried to reinstall using the installer.sh and now i am getting "FileNotFoundError". Will check this again.

shivasiddharth, 2018-01-29:
Not sure as to which guide you are following. There is no demo command now after the 20th Dec GA SDK update. Also in the latest versions, i have asked people no not to rename the credentials file.

ivaturianil, 2018-01-30:
@shivasiddharth , I am following https://www.hackster.io/shiva-siddharth/simultaneously-run-alexa-and-google-assistant-on-pi-12e8df till now and i didn't realize that you have updated the github. 
You saved lot of time for me. Will follow github. Many thanks for your response.

ivaturianil, 2018-01-31:
@shivasiddharth , I have re-installed the project and still having problem with Google Assistant. It is not starting after bootup  (Alexa does start with "Hello" word). When I run "google-assistant-demo --project_id anil-rasp-goolexa --device_model_id anil-rasp-goolexa-20180130235146" command google assistant starting without any sound. After this command both (GAssist & Alexa) are working fine and the volume is good(no need to set again). I have verified the sound files and both are same. I tried to use single quotes for project & model-id but no luck. Attached the screen shot with all some details. Any help to fix the issue is appreciated. 

[image]

shivasiddharth, 2018-01-31:
You are saying that everything is working and also saying that there some issue. Can you please be clear as to what is the issue.

ivaturianil, 2018-01-31:
Issue : Google Assistant is not starting after pi restarted(Alexa is starting automatically).
I need to start Google Assistant with "google-assistant-demo --project_id anil-rasp-goolexa --device_model_id anil-rasp-goolexa-20180130235146" command. 
I was not able to find why google Assistant is not starting automatically. I am looking for help to fix this issue.

shivasiddharth, 2018-01-31:
Please post the status of the service using:
`sudo systemctl status gassistpi-ok-google.service `

ivaturianil, 2018-02-01:
@shivasiddharth , Sorry to bug you these many times. I am really trying to fix this issue with little help from you(trying diff options). But none of them are working. This time to fix the issue I ran ¨update Gassist script¨ using https://github.com/shivasiddharth/GassistPi/blob/update-script/GassistPi-Update-Script.sh link. Now GAssist is working fine but not Alexa. 

Below are the status details for Alexa.

pi@raspberrypi:~ $ sudo systemctl status companionapp.service
● companionapp.service - Companion Service
   Loaded: loaded (/lib/systemd/system/companionapp.service; enabled; vendor pre
   Active: failed (Result: exit-code) since Wed 2018-01-31 21:51:25 EST; 1min 4s
  Process: 664 ExecStart=/bin/bash /home/pi/Assistants-Pi/scripts/companionstart
 Main PID: 664 (code=exited, status=1/FAILURE)

Jan 31 21:51:24 raspberrypi systemd[1]: companionapp.service: Main process exite
Jan 31 21:51:24 raspberrypi systemd[1]: companionapp.service: Unit entered faile
Jan 31 21:51:24 raspberrypi systemd[1]: companionapp.service: Failed with result
Jan 31 21:51:25 raspberrypi systemd[1]: companionapp.service: Service hold-off t
Jan 31 21:51:25 raspberrypi systemd[1]: Stopped Companion Service.
Jan 31 21:51:25 raspberrypi systemd[1]: companionapp.service: Start request repe
Jan 31 21:51:25 raspberrypi systemd[1]: Failed to start Companion Service.
Jan 31 21:51:25 raspberrypi systemd[1]: companionapp.service: Unit entered faile
Jan 31 21:51:25 raspberrypi systemd[1]: companionapp.service: Failed with result
...skipping...
● companionapp.service - Companion Service
   Loaded: loaded (/lib/systemd/system/companionapp.service; enabled; vendor pre
   Active: failed (Result: exit-code) since Wed 2018-01-31 21:51:25 EST; 1min 4s
  Process: 664 ExecStart=/bin/bash /home/pi/Assistants-Pi/scripts/companionstart
 Main PID: 664 (code=exited, status=1/FAILURE)

Jan 31 21:51:24 raspberrypi systemd[1]: companionapp.service: Main process exite
Jan 31 21:51:24 raspberrypi systemd[1]: companionapp.service: Unit entered faile
Jan 31 21:51:24 raspberrypi systemd[1]: companionapp.service: Failed with result
Jan 31 21:51:25 raspberrypi systemd[1]: companionapp.service: Service hold-off t
Jan 31 21:51:25 raspberrypi systemd[1]: Stopped Companion Service.
Jan 31 21:51:25 raspberrypi systemd[1]: companionapp.service: Start request repe
Jan 31 21:51:25 raspberrypi systemd[1]: Failed to start Companion Service.
Jan 31 21:51:25 raspberrypi systemd[1]: companionapp.service: Unit entered faile
Jan 31 21:51:25 raspberrypi systemd[1]: companionapp.service: Failed with result
~
~
~
~
~
~
~
~

pi@raspberrypi:~ $ sudo systemctl status client.service
● client.service - Client Service
   Loaded: loaded (/lib/systemd/system/client.service; enabled; vendor preset: e
   Active: inactive (dead) since Wed 2018-01-31 21:51:25 EST; 1min 19s ago
  Process: 665 ExecStart=/bin/bash /home/pi/Assistants-Pi/scripts/clientstart.sh
 Main PID: 665 (code=killed, signal=TERM)

Jan 31 21:51:24 raspberrypi systemd[1]: Started Client Service.
Jan 31 21:51:24 raspberrypi sudo[668]:       pi : TTY=unknown ; PWD=/home/pi/ale
Jan 31 21:51:25 raspberrypi sudo[668]: pam_unix(sudo:session): session opened fo
Jan 31 21:51:25 raspberrypi systemd[1]: Stopping Client Service...
Jan 31 21:51:25 raspberrypi sudo[668]: pam_unix(sudo:session): session closed fo
Jan 31 21:51:25 raspberrypi systemd[1]: Stopped Client Service.
Jan 31 21:51:25 raspberrypi systemd[1]: Dependency failed for Client Service.
Jan 31 21:51:25 raspberrypi systemd[1]: client.service: Job client.service/start
...skipping...
● client.service - Client Service
   Loaded: loaded (/lib/systemd/system/client.service; enabled; vendor preset: e
   Active: inactive (dead) since Wed 2018-01-31 21:51:25 EST; 1min 19s ago
  Process: 665 ExecStart=/bin/bash /home/pi/Assistants-Pi/scripts/clientstart.sh
 Main PID: 665 (code=killed, signal=TERM)

Jan 31 21:51:24 raspberrypi systemd[1]: Started Client Service.
Jan 31 21:51:24 raspberrypi sudo[668]:       pi : TTY=unknown ; PWD=/home/pi/ale
Jan 31 21:51:25 raspberrypi sudo[668]: pam_unix(sudo:session): session opened fo
Jan 31 21:51:25 raspberrypi systemd[1]: Stopping Client Service...
Jan 31 21:51:25 raspberrypi sudo[668]: pam_unix(sudo:session): session closed fo
Jan 31 21:51:25 raspberrypi systemd[1]: Stopped Client Service.
Jan 31 21:51:25 raspberrypi systemd[1]: Dependency failed for Client Service.
Jan 31 21:51:25 raspberrypi systemd[1]: client.service: Job client.service/start
~
~
~
~
~
~
~
~
~
lines 1-14/14 (END)
pi@raspberrypi:~ $ sudo systemctl status wakeword.service
● wakeword.service - Wakeword Service
   Loaded: loaded (/lib/systemd/system/wakeword.service; enabled; vendor preset:
   Active: inactive (dead) since Wed 2018-01-31 21:51:25 EST; 1min 32s ago
  Process: 666 ExecStart=/bin/bash /home/pi/Assistants-Pi/scripts/wakeword.sh (c
 Main PID: 666 (code=killed, signal=TERM)

Jan 31 21:51:24 raspberrypi systemd[1]: Started Wakeword Service.
Jan 31 21:51:25 raspberrypi systemd[1]: Stopping Wakeword Service...
Jan 31 21:51:25 raspberrypi systemd[1]: Stopped Wakeword Service.
Jan 31 21:51:25 raspberrypi systemd[1]: Dependency failed for Wakeword Service.
Jan 31 21:51:25 raspberrypi systemd[1]: wakeword.service: Job wakeword.service/s
...skipping...
● wakeword.service - Wakeword Service
   Loaded: loaded (/lib/systemd/system/wakeword.service; enabled; vendor preset:
   Active: inactive (dead) since Wed 2018-01-31 21:51:25 EST; 1min 32s ago
  Process: 666 ExecStart=/bin/bash /home/pi/Assistants-Pi/scripts/wakeword.sh (c
 Main PID: 666 (code=killed, signal=TERM)

Jan 31 21:51:24 raspberrypi systemd[1]: Started Wakeword Service.
Jan 31 21:51:25 raspberrypi systemd[1]: Stopping Wakeword Service...
Jan 31 21:51:25 raspberrypi systemd[1]: Stopped Wakeword Service.
Jan 31 21:51:25 raspberrypi systemd[1]: Dependency failed for Wakeword Service.
Jan 31 21:51:25 raspberrypi systemd[1]: wakeword.service: Job wakeword.service/s
~
~
~
~
~
~
~
~
~
~
~
~

pi@raspberrypi:~ $

Rooster69, 2018-02-01:
@ivaturianil Sorry to hijack your thread, but I think this should be useful

@shivasiddharth I'm having identical problem as @ivaturianil. The problem that I'm having is with the Gassispi not auto-start. Every time that I reboot the Pi, I have to manually start the service by command as followed:

source env/bin/activate
googlesamples-assistant-hotword --project_id pi-assistance-xxxxxx --device_model_id pi-assistance-xxxxxx-20180131185556

when I ran the command you have suggested above, I'm getting the following:

● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: **failed** (Result: exit-code) since Thu 2018-02-01 14:24:00 UTC; 13min ago
  Process: 891 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id pi-assistance-**xxxxxx**--device_model_id pi-assistance-**xxxxxx**-20180131185556 (code=exited, status=1/FAILURE)
 Main PID: 891 **(code=exited, status=1/FAILURE)**

Feb 01 14:24:00 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Feb 01 14:24:00 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Feb 01 14:24:00 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Feb 01 14:24:00 raspberrypi systemd[1]: gassistpi-ok-google.service: Start request repeated too quickly.
Feb 01 14:24:00 raspberrypi systemd[1]: **Failed to start Google Assistant-Voice.**
Feb 01 14:24:00 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Feb 01 14:24:00 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.

note: where I substitute xxxxxx with my project ID numbers.

Rooster69, 2018-02-02:
Update: by updating the Google assistance using [GassistPi-Update-Script.sh](https://github.com/shivasiddharth/GassistPi/blob/update-script/GassistPi-Update-Script.sh). it fix the Google auto start issue. but at the same time, it brick Alexa.

Re-setup Alexa, everything is now good.

ivaturianil, 2018-02-02:
@Rooster69 , To be specific, you have used "Assistants-Pi" option 1 i.e., install only Alexa. 
Could you please confirm.

shivasiddharth, 2018-02-02:
@Rooster69 , Thanks for your input, i shall checkout the dependency issue.

Rooster69, 2018-02-02:
The initial install was 3) Both. But Google-Assistant start with error as auto-start.

Ran the update with your script, Google-Assistant works on auto-start but bricked Alexa (stop working)

Re-install Alexa nu using option 2) Both Alexa and Google-Assistant is now working.

Rooster69, 2018-02-02:
Google stream stop working

Me: "hey google, stream romantic music" 
GassisPi "getting youtube links" .. sound.. chime from speacker.. "Hi"

I think at this point, the app has crashed.

shivasiddharth, 2018-02-02:
@Rooster69 , Did you add the YouTube API in the actions.py file ? You can find further details on that [here](https://github.com/shivasiddharth/GassistPi)

Rooster69, 2018-02-02:
I have already enabled the API and it was working from before. I think I would have to re-install everything from the beginning using one install at a time.

I will report back.

Rooster69, 2018-02-03:
Start from fresh install will not help.
I have to use and older version of the **2017-11-29-raspbian-stretch.img** and disable the update in order to install properly.

After the install, everything works fine including youttube,

Take an image of the card for backup purpose.

Ran the update script, reboot, youtube and Alexa stop working. Re-install Alexa, Alexa works fine.

I think the update script brick youtube.

I have re-image the backup to the SD card and everything works fine again as long as I don't update.

@shivasiddharth Any suggestion that could try?

shivasiddharth, 2018-02-03:
@Rooster69 , There is hardly anything i could do/ suggest. Alexa right now is in shambles, thanks to Amazon's ridiculous policies for modification. Neither would they allow us to do changes/modifications to the code and redistribute, nor would they accept pull requests from the community. They themselves have right now changed the avs-sample-app as a development project.
At this point all that i can do is port the Alexa SDK, which is more robust. But that will take time.

Rooster69, 2018-02-03:
Thanks for the great job, I will be waiting for your next fix. Keep up the good job.

ivaturianil, 2018-02-05:
@shivasiddharth & @Rooster69 : i have re-installed only Alexa using Assistants-pi library.  Now both assistants are working fine. Thanks for your help in setting/fixing things.

shivasiddharth, 2018-02-05:
Closing this issue then !

closed 2018-02-05 by shivasiddharth

Rooster69, 2018-02-05:
@ivaturianil and @shivasiddharth 
I have tried your steps and GassistPi is still not loading.
pi@raspberrypi:~ $ sudo systemctl status gassistpi-ok-google.service
● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2018-02-05 09:14:04 EST; 4min 19s ago
  Process: 907 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id pi-assistance-xxxxxx --device_model_id pi-assistance-xxxxxx-20180203114103 (code=exited, status=1/FAILURE)
 Main PID: 907 (code=exited, status=1/FAILURE)

Feb 05 09:14:04 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=1/FAILURE
Feb 05 09:14:04 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Feb 05 09:14:04 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Feb 05 09:14:04 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Feb 05 09:14:04 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Feb 05 09:14:04 raspberrypi systemd[1]: gassistpi-ok-google.service: Start request repeated too quickly.
Feb 05 09:14:04 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.
Feb 05 09:14:04 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Feb 05 09:14:04 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
pi@raspberrypi:~ $

shivasiddharth, 2018-02-13:
For future reference. 
If you use the Google Assistant update script, then open a terminal and run:
`sudo apt-get install npm` 
GA removes the npm and hence this alexa failure. If you run the above command, Alexa will be restored on restart, no need to re-run the installer.

shivasiddharth, 2018-02-13:
@Rooster69 , Can you try adding the ids in systemd file within quotes ? And also please attach the log file. You can find the log in /tmp folder.

Rooster69, 2018-02-14:
Hi Sid,
I have followed your instructions but still have problem. Please see attache new clean log file.