shivasiddharth/Assistants-Pi #32: alexa sample voice service  java......is not working on my linux device

Kishu18, 2018-02-02:
it says
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/nodejs', '/usr/bin/npm', 'start' ]
2 info using npm@1.4.21
3 info using node@v4.8.2
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info prestart alexa-voice-service-sample-companion-service@1.0.0
6 info start alexa-voice-service-sample-companion-service@1.0.0
7 verbose unsafe-perm in lifecycle true
8 info alexa-voice-service-sample-companion-service@1.0.0 Failed to exec start script
9 error alexa-voice-service-sample-companion-service@1.0.0 start: `node ./bin/www`
9 error Exit status 1
10 error Failed at the alexa-voice-service-sample-companion-service@1.0.0 start script.
10 error This is most likely a problem with the alexa-voice-service-sample-companion-service package,
10 error not with npm itself.
10 error Tell the author that this fails on your system:
10 error     node ./bin/www
10 error You can get their info via:
10 error     npm owner ls alexa-voice-service-sample-companion-service
10 error There is likely additional logging output above.
11 error System Linux 4.9.59-v7+
12 error command "/usr/bin/nodejs" "/usr/bin/npm" "start"
13 error cwd /home/pi/alexa-avs-sample-app/samples/companionService
14 error node -v v4.8.2
15 error npm -v 1.4.21
16 error code ELIFECYCLE
17 verbose exit [ 1, true ]

shivasiddharth, 2018-02-02:
What is your Linux Device ?

Kishu18, 2018-02-02:
raspberry pi's raspbian.......plz help......I have my google assistant working,,,,,,but alexa is not installing

Kishu18, 2018-02-02:
my google assistant too does not play songs if I say it to do so......please show me a way to fix this

shivasiddharth, 2018-02-02:
Which version of Raspbian are you using ? For google assistant issues, attach the log file. Log file can be found in the /tmp directory.

Kishu18, 2018-02-05:
Ok i will try

closed 2018-02-06 by Kishu18