shivasiddharth/Assistants-Pi #37: Java psutil module missing

TedMichalik, 2018-02-04:
Reimaged my RPi3 and did a fresh install of both assistants. Installation tests worked. Alexa is working. Google assistant won't start. Log shows this:

Feb  4 15:32:44 voice systemd[1]: Started Google Assistant-Voice.
Feb  4 15:32:45 voice python[1855]: Traceback (most recent call last):
Feb  4 15:32:45 voice python[1855]:   File "/home/pi/GassistPi/src/main.py", line 25, in <module>
Feb  4 15:32:45 voice python[1855]:     import psutil
Feb  4 15:32:45 voice python[1855]: ImportError: No module named 'psutil'
Feb  4 15:32:45 voice systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=1/FAILURE

shivasiddharth, 2018-02-05:
Looks like the assistant is running outside the environment. Can you check the Exec path in the systemd file ?

TedMichalik, 2018-02-05:
~ $ cat Assistants-Pi/systemd/gassistpi-ok-google.service 
[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'sigma-firmament-185618'  --device_model_id 'sigma-firmament-185618-20180204144721'

WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target

shivasiddharth, 2018-02-05:
By any chance, are you running the AIY image ?

TedMichalik, 2018-02-06:
No, it is the standard Raspbian desktop image.  I will try rebuilding the whole thing tomorrow to see if maybe I did something wrong.

TedMichalik, 2018-02-06:
Did a fresh install of Raspbian. Before starting your install, I ran "pydoc modules" and see that the psutil module is not installed. Starting your build now.

TedMichalik, 2018-02-06:
I found that the AIY-HAT/scripts/install-alsa-config.sh script is still in DOS format. Fixed it with dos2unix.

After running the prep-system.sh script, I ran "pydoc modules". Still no psutil module, and I got these new error messages after the module list:
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/dist-packages/phatbeat/__init__.py", line 46, in _exit
    GPIO.cleanup()
AttributeError: 'NoneType' object has no attribute 'cleanup'
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/dist-packages/phatbeat/__init__.py", line 46, in _exit
    GPIO.cleanup()
AttributeError: 'NoneType' object has no attribute 'cleanup'

shivasiddharth, 2018-02-06:
Can you check if the .asoundrc and asound.conf are same ? Raspbian tends to append values to the .asoundrc

TedMichalik, 2018-02-06:
I ran diff on the two files and they are identical:
pi@voice:~ $ diff .asoundrc /etc/asound.conf
pi@voice:~ $ cat .asoundrc
options snd_rpi_googlemihat_soundcard index=0

pcm.softvol {
    type softvol
    slave.pcm dmix
    control {
        name Master
        card 0
    }
}

pcm.micboost {
    type route
    slave.pcm dsnoop
    ttable {
        0.0 30.0
        1.1 30.0
    }
}

pcm.!default {
    type asym
    playback.pcm "plug:softvol"
    capture.pcm "plug:micboost"
}

ctl.!default {
    type hw
    card 0
}

shivasiddharth, 2018-02-06:
System prep does not install the pip modules. The actual installer installs the pip. And also, i think you are checking for the modules outside the environment, the pip modules are installed within the environment.

Reg the audio issue, can you post the error that you are seeing ?

shivasiddharth, 2018-02-06:
Can you check the config file and make sure that the onboard audio has been disabled ?

TedMichalik, 2018-02-06:
Regarding the audio issue, when I got to Step 6 of the README and ran audio-test.sh there was no sound and I couldn't find any error messages in the logs. After I rebooted, it worked Ok.

The Alexa demo worked Ok after running the installer.sh script.
 
The onboard audio is not active:
pi@voice:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

TedMichalik, 2018-02-06:
Regarding the python modules, psutil does not show up after running the installer.sh script:
pi@voice:~ $ source env/bin/activate
(env) pi@voice:~ $ pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
aftership (0.2)
cachetools (2.0.1)
certifi (2018.1.18)
cffi (1.11.4)
chardet (3.0.4)
click (6.7)
enum34 (1.1.6)
feedparser (5.2.1)
futures (3.1.1)
google-api-python-client (1.6.5)
google-assistant-grpc (0.1.0)
google-assistant-library (0.1.0)
google-assistant-sdk (0.4.2)
google-auth (1.3.0)
google-auth-httplib2 (0.0.3)
google-auth-oauthlib (0.2.0)
googleapis-common-protos (1.5.3)
grpcio (1.9.1)
httplib2 (0.10.3)
idna (2.6)
kodi-json (1.0.0)
monotonic (1.4)
oauth2client (4.1.2)
oauthlib (2.0.6)
pip (9.0.1)
pkg-resources (0.0.0)
protobuf (3.5.1)
pyasn1 (0.4.2)
pyasn1-modules (0.2.1)
PyAudio (0.2.11)
pycparser (2.18)
python-dateutil (2.6.1)
requests (2.18.4)
requests-oauthlib (0.8.0)
RPi.GPIO (0.6.3)
rsa (3.4.2)
setuptools (38.5.1)
six (1.11.0)
sounddevice (0.3.10)
tenacity (4.8.0)
uritemplate (3.0.0)
urllib3 (1.22)

shivasiddharth, 2018-02-06:
Move into the environment and use `pip install psutil`. will take a look as to why that module was left out

TedMichalik, 2018-02-06:
After installing psutil, the demo worked. I made the changes to make the assistants auto start and rebooted. Alexa works, but the Google assistant still fails. Here are the log messages:

Feb  6 18:35:14 voice systemd[1]: Started Google Assistant.
Feb  6 18:35:14 voice python[1459]: Traceback (most recent call last):
Feb  6 18:35:14 voice python[1459]:   File "/home/pi/GassistPi/src/main.py", line 33, in <module>
Feb  6 18:35:14 voice python[1459]:     from actions import say
Feb  6 18:35:14 voice python[1459]:   File "/home/pi/GassistPi/src/actions.py", line 9, in <module>
Feb  6 18:35:14 voice python[1459]:     from gmusicapi import Mobileclient
Feb  6 18:35:14 voice python[1459]: ImportError: No module named 'gmusicapi'
Feb  6 18:35:14 voice systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=1/FAILURE
Feb  6 18:35:14 voice systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Feb  6 18:35:14 voice systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Feb  6 18:35:15 voice systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Feb  6 18:35:15 voice systemd[1]: Stopped Google Assistant-Voice.
Feb  6 18:35:15 voice systemd[1]: Started Google Assistant-Voice.
Feb  6 18:35:15 voice python[1466]: Traceback (most recent call last):
Feb  6 18:35:15 voice python[1466]:   File "/home/pi/GassistPi/src/stopbutton.py", line 7, in <module>
Feb  6 18:35:15 voice python[1466]:     from actions import stop
Feb  6 18:35:15 voice python[1466]:   File "/home/pi/GassistPi/src/actions.py", line 9, in <module>
Feb  6 18:35:15 voice python[1466]:     from gmusicapi import Mobileclient

shivasiddharth, 2018-02-07:
Seems like gmusicapi is not installed as well. Move into the environment and use `pip install gmusicapi`.

TedMichalik, 2018-02-07:
More were missing. Also had to install googletrans, gtts, pafy, pychromecast. After that both Alexa and Google Assistant work. 

Getting these error messages in the log, but the assistant works:
Feb  6 19:22:43 voice python[2666]: WARNING:root:pafy: youtube-dl not found; falling back to internal backend. This is not as well maintained as the youtube-dl backend. To hide this message, set the environmental variable PAFY_BACKEND to "internal".
Feb  6 19:22:43 voice python[2666]: INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
Feb  6 19:22:43 voice python[2666]: INFO:gmusicapi.Mobileclient1:version: 11.0.0
Feb  6 19:22:43 voice python[2666]: INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
Feb  6 19:22:43 voice python[2666]: INFO:gmusicapi.Mobileclient1:initialized
Feb  6 19:22:43 voice python[2666]: INFO:gmusicapi.Mobileclient1:logged out
Feb  6 19:22:43 voice python[2666]: INFO:gmusicapi.Mobileclient1:failed to authenticate

shivasiddharth, 2018-02-07:
Move into the environment and use pip install youtube-dl that will get rid of 
WARNING:root:pafy: youtube-dl not found; falling back to internal backend. This is not as well maintained as the youtube-dl backend. To hide this message, set the environmental variable PAFY_BACKEND to "internal".
Looks like the gmusicapi's credentials have not been set or incorrectly set.

closed 2018-10-12 by shivasiddharth

georgeion, 2021-02-12:
> Move into the environment and use `pip install youtube-dl `that will get rid of
> WARNING:root:pafy: youtube-dl not found; falling back to internal backend. This is not as well maintained as the youtube-dl backend. To hide this message, set the environmental variable PAFY_BACKEND to "internal".
> Looks like the gmusicapi's credentials have not been set or incorrectly set. That has nothing to do with the assistant's working. Its just that you will not be able to stream music from google music

I have tried everything that you said. When running:
 **googlesamples-assistant-hotword --device-model-id rpi—assitant-684d4-pi-google-assistant-ie0** is have this output: 
**bash: googlesamples-assistant-hotword: command not found**
I was trying on RPi4 and RPi 400. the same error. 
When unning pip list the outcone is this:
**appdirs                  1.4.4
beautifulsoup4           4.9.3
cachetools               4.2.1
certifi                  2020.12.5
cffi                     1.14.4
chardet                  4.0.0
click                    6.7
cryptography             3.3.1
decorator                4.4.2
enum34                   1.1.10
futures                  3.1.1
gmusicapi                13.0.0
google-assistant-grpc    0.2.1
google-assistant-library 1.0.0
google-assistant-sdk     0.6.0
google-auth              1.24.0
google-auth-oauthlib     0.4.2
googleapis-common-protos 1.52.0
gpsoauth                 0.4.3
grpcio                   1.35.0
httplib2                 0.19.0
idna                     2.10
lxml                     4.6.2
MechanicalSoup           1.0.0
mutagen                  1.45.1
oauth2client             4.1.3
oauthlib                 3.1.0
pathlib2                 2.3.5
pip                      21.0.1
pkg-resources            0.0.0
proboscis                1.2.6.0
protobuf                 3.14.0
psutil                   5.8.0
pyasn1                   0.4.8
pyasn1-modules           0.2.8
pycparser                2.20
pycryptodomex            3.10.1
pyOpenSSL                20.0.1
pyparsing                2.4.7
python-dateutil          2.8.1
requests                 2.25.1
requests-oauthlib        1.3.0
rsa                      4.7
setuptools               53.0.0
six                      1.15.0
sounddevice              0.3.15
soupsieve                2.2
tenacity                 4.12.0
urllib3                  1.26.3
validictory              1.1.2
youtube-dl               2021.2.10**

shivasiddharth, 2021-02-13:
The command for the sample is wrong. It should be `google-assistant-demo` and not `googlesamples-assistant-hotword`. You are here on a three year old issue. Quite a lot has changed. This project no longer carries the features of GassistPi  if you are looking for music streaming, etc.

georgeion, 2021-02-13:
> The command for the sample is wrong. It should be `google-assistant-demo` and not `googlesamples-assistant-hotword`. You are here on a three year old issue. Quite a lot has changed. This project no longer carries the features of GassistPi if you are looking for music streaming, etc.

Hi mate. I have tried that too. This is the outcome:
(env) pi@raspberrypi:~ $ google-assistant-demo --project-id rpi--assitant-684d4  --device-model-id rpi—assitant-684d4-pi-google-assistant-ie0g2f
Traceback (most recent call last):
  File "/home/pi/env/bin/google-assistant-demo", line 8, in <module>
    sys.exit(main())
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/__main__.py", line 122, in main
    with Assistant(credentials, device_model_id) as assistant:
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 98, in __init__
    device_model_id.encode('ASCII')))
UnicodeEncodeError: 'ascii' codec can't encode character '\u2014' in position 3: ordinal not in range(128)

georgeion, 2021-02-13:
> The command for the sample is wrong. It should be `google-assistant-demo` and not `googlesamples-assistant-hotword`. You are here on a three year old issue. Quite a lot has changed. This project no longer carries the features of GassistPi if you are looking for music streaming, etc.

I am just trying the hotword. I found out this is not supported by Google anymore. Snowboy it will be an option.

shivasiddharth, 2021-02-13:
Its just that the Google Assistant Library is deprecated, but "Hey Google" wakewording still works. Demo: (https://youtu.be/TLP1Ysu8d3g?t=1138)  Snowboy has shutdown. So I added Picovoice to the GassistPi project.

In your case, there seems to be an issue with the model/project id that you are using. I would suggest you to double check that. Thats what is throwing that ASCII error.

georgeion, 2021-02-13:
Cheers, mate! Much appreciated!

commit 8cda70cb (2018-10-12): Migrated the project to AVS SDK

Fixes #3, Fixes #15, Fixes #69 , Fixes #66 , Fixes #65 , Fixes #64 , Fixes #62 , Fixes #61 , Fixes #60 , Fixes #59 , Fixes #58 , Fixes #57 , Fixes #56 , Fixes #55 , Fixes #54 , Fixes #53 , Fixes #52 , Fixes #51 , Fixes #49 , Fixes #46 , Fixes #38 , Fixes #37 , Fixes #33 , Fixes #20 , Fixes #19 , Fixes #18 , Fixes #16