shivasiddharth/Assistants-Pi #39: Google Assistant Not Starting at Boot

Ricky3125, 2018-02-05:
I am able to run google assistant demo in source env/bin/activate, googlesamples-assistant-hotword --project_id 'mypi3-env-190008' --device_model_id '2011205041740' . I am not able to get Google Assistant to run on boot.  I am using a Raspberry PI3 B and fresh install of Raspbian.  After  I do sudo systemctl enable gassistpi-ok-google.service  and  sudo systemctl start gassistpi-ok-google.service. I then do sudo systemctl status gassistpi-ok-google.service and get/see the following:

● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2018-02-05 12:10:46 PST; 1s ago
  Process: 11404 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id mypi3-env-190008 --device_model_id mypi3-env-190008-2011205041740 (code=exited, status=1/FAIL
 Main PID: 11404 (code=exited, status=1/FAILURE)

Feb 05 12:10:46 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Feb 05 12:10:46 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Feb 05 12:10:46 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Feb 05 12:10:46 raspberrypi systemd[1]: gassistpi-ok-google.service: Start request repeated too quickly.
Feb 05 12:10:46 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.
Feb 05 12:10:46 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Feb 05 12:10:46 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.

Any help or suggestions would be appreciated.  
Thank you in advance
Rick

shivasiddharth, 2018-02-05:
The project-id and model-id should be within single quotes.

Ricky3125, 2018-02-08:
I have tried a few different things and still not able to get  'gassistpi-ok-google.service' to start on boot.  As I said I am able to run the 'google assistant demo' to run with no issues.

**First test I did is as shown:** 
In this test I tried with no quotes (should fail and did)
In the gassistpi-ok-google.service file this is what I have for this test:

ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id mypi3-env-190008  --device_model_id mypi3-env-190008-2011205041740

● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2018-02-08 09:31:21 PST; 17s ago
  Process: 6781 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id mypi3-env-190008 --device_model_id mypi3-env-190008-2011205041740 
 Main PID: 6781 (code=exited, status=1/FAILURE)

Feb 08 09:31:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Feb 08 09:31:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Feb 08 09:31:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Feb 08 09:31:21 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Feb 08 09:31:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Start request repeated too quickly.
Feb 08 09:31:21 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.
Feb 08 09:31:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Feb 08 09:31:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.

**The next test I did it failed as well.**

In the gassistpi-ok-google.service file this is what I have for this test (note the single quotes, this should work. Correct ?):

ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'mypi3-env-190008'  --device_model_id 'mypi3-env-190008-2011205041740'

● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2018-02-08 09:38:08 PST; 7s ago
  Process: 7960 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id mypi3-env-190008 --device_model_id mypi3-env-190008-2011205041740 
 Main PID: 7960 (code=exited, status=1/FAILURE)

Feb 08 09:38:07 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Feb 08 09:38:08 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Feb 08 09:38:08 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Feb 08 09:38:08 raspberrypi systemd[1]: gassistpi-ok-google.service: Start request repeated too quickly.
Feb 08 09:38:08 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.
Feb 08 09:38:08 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Feb 08 09:38:08 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.

**And the last test I did was this:**
In the gassistpi-ok-google.service file this is what I have for this test (note the double quotes, single quotes.  Should have failed (?) and did.).
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id "'mypi3-env-190008'"  --device_model_id "'mypi3-env-190008-2011205041740'"

● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2018-02-08 10:19:25 PST; 3min 48s ago
  Process: 2247 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'mypi3-env-190008' --device_model_id 'mypi3-env-190008-2011205041740' (code=exited, status=1/FAILURE)
 Main PID: 2247 (code=exited, status=1/FAILURE)

Feb 08 10:19:25 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Feb 08 10:19:25 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Feb 08 10:19:25 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Feb 08 10:19:25 raspberrypi systemd[1]: gassistpi-ok-google.service: Start request repeated too quickly.
Feb 08 10:19:25 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.
Feb 08 10:19:25 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Feb 08 10:19:25 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.

Although the quotes do show in the status, I believe that by using the double quotes that it is attempting to pass the "single quotes" as part of the string(parameters) which would cause it to fail.  I do have some knowledge regarding programming, but not phython (Perl, Delphi, C++ and a little PHP). Thank you for your assistance (sorry for the long story, I would like to get this running as a service).
Rick

shivasiddharth, 2018-02-08:
Can you please attach the log file. You can find that in the /tmp folder.

Ricky3125, 2018-02-10:
There were no log files being generated, I did some further test based on this users issue (https://github.com/shivasiddharth/GassistPi/issues/194 and 

At the command line I used:
/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id mypi3-env-190008  --device_model_id mypi3-env-190008-2011205041740   

And recieved this error:

/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id mypi3-env-190008  --device_model_id mypi3-env-190008-2011205041740
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 25, in <module>
    import psutil
ImportError: No module named 'psutil'

I reinstalled psutil but the same error occurred. Next I went to the /home/pi/GassistPi/src/main.py file and commented (#) the following lines:
25 - #import psutil

33 - #from actions import say
34 - #from actions import Action
35 - #from actions import YouTube_No_Autoplay
36 - #from actions import YouTube_Autoplay
37 - #from actions import stop
38 - #from actions import radio
39 - #from actions import ESP
40 - #from actions import track
41 - #from actions import feed
42 - #from actions import kodiactions
43 - #from actions import mutevolstatus
44 - #from actions import play_playlist
45 - #from actions import play_songs
46 - #from actions import play_album
47 - #from actions import play_artist
48 - #from actions import refreshlists
49 - #from actions import chromecast_play_video
50 - #from actions import chromecast_control

The "google assistant" started but errored shortly after the first "wake command" and it generated a GassistPi.log file:

2018-02-09 15:26:17,800 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-02-09 15:26:17,819 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-02-09 15:28:29,491 DEBUG urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-02-09 15:28:29,523 DEBUG urllib3.util.retry Converted retries value: 3 -> Retry(total=3, connect=None, read=None, redirect=None, status=None)
2018-02-09 15:28:29,542 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): embeddedassistant.googleapis.com
2018-02-09 15:30:42,747 DEBUG urllib3.connectionpool https://embeddedassistant.googleapis.com:443 "GET /v1alpha2/projects/mypi3-env-190008/devices/7FC7E94D0A486AA7C4B18CEA876A8C03 HTTP/1.1" 200 None
2018-02-09 15:30:53,443 ERROR __main__ name 'psutil' is not defined
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 495, in <module>
    main()
  File "/home/pi/GassistPi/src/main.py", line 234, in main
    process_event(event, assistant.device_id)
  File "/home/pi/GassistPi/src/main.py", line 125, in process_event
    if ismpvplaying():
  File "/home/pi/GassistPi/src/main.py", line 83, in ismpvplaying
    for pid in psutil.pids():
NameError: name 'psutil' is not defined

I am thinking something incorrectly installed and is causing the issue because as soon as I un-comment any of the above lines, it quits generating the log file and will not start.   I am thinking about doing a new installation of Raspbian and Assistant-PI  should this not be enough information to resolve the issue..

[GassistPi.log](https://github.com/shivasiddharth/Assistants-Pi/files/1712819/GassistPi.log)

Regards Rick

shivasiddharth, 2018-02-10:
I have already discussed about this in #37 Hope it helps. And, you should not comment out all those functions. They are still very much used in the body of the script. When the script tries to call an un-imported function, it will crash.

Ricky3125, 2018-02-10:
After installing the following modules in (source env/bin/activate) I now have both assistants running on boot:
pip install psutil
pip install gmusicapi
pip install googletrans
pip install gtts
pip install pafy
pip install pychromecast

I had only commented out all the functions in order to try and diagnose the issue. Thank you for all of your help. It was and is greatly appreciated !!!
Regards
Rick

closed 2018-02-10 by Ricky3125