shivasiddharth/Assistants-Pi #40: Audio input processor error

Kishu18, 2018-02-06:
https://embeddedassistant.googleapis.com/v1alpha2/projects/anirudh-bot195/devices/EC75642F5B5FD3877F989377DB11EBC5 200
[7588:7610:ERROR:audio_input_processor.cc(748)] Input error
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED
[7588:7612:ERROR:audio_input_processor.cc(748)] Input error

Kishu18, 2018-02-06:
I tried all the troubleshooting options but that didn't work too

shivasiddharth, 2018-02-06:
Check if your .asoundrc and asound.conf are same.

Kishu18, 2018-02-06:
yes they are same

Kishu18, 2018-02-07:
if you have teamviewer can you please takeover my pc and look into the issue.......please

shivasiddharth, 2018-02-07:
Sorry, i dont resort those practices. Please post the outputs of `arecord -l` , `aplay-l` , your asound.conf and .asoundrc file outputs and also please state the audio setup that you are using.

Kishu18, 2018-02-07:
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Kishu18, 2018-02-07:
pi@raspberrypi:~ $ aplay-l
bash: aplay-l: command not found

Kishu18, 2018-02-07:
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug

Kishu18, 2018-02-07:
cm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
    slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"

Kishu18, 2018-02-07:
I am using a usb mic which was in the description liink provided by beebom

shivasiddharth, 2018-02-07:
Closing all of the user's threads for spamming.

closed 2018-02-07 by shivasiddharth