shivasiddharth/Assistants-Pi #41: alexa token retrival failure

Kishu18, 2018-02-06:
when i typed the command in new terminal Run the companion service: cd /home/pi/alexa-avs-sample-app/samples/companionService && sudo npm start

Listening on port 3000
Successfully retrieved registration code for AnirudhBot / 123456789
Error retrieving access token: InvalidSessionId, No refresh tokens stored for this session id: f69192a7-8eb1-4611-9134-a2782c29909a
Successfully retrieved registration code for AnirudhBot / 123456789
Error retrieving access token: InvalidSessionId, No refresh tokens stored for this session id: a6a841d3-ac49-43d1-beae-ba5a53e4f1c8
_http_outgoing.js:356
    throw new Error('Can\'t set headers after they are sent.');
    ^

Error: Can't set headers after they are sent.
    at ServerResponse.OutgoingMessage.setHeader (_http_outgoing.js:356:11)
    at ServerResponse.header (/home/pi/alexa-avs-sample-app/samples/companionService/node_modules/express/lib/response.js:700:10)
    at ServerResponse.send (/home/pi/alexa-avs-sample-app/samples/companionService/node_modules/express/lib/response.js:154:12)
    at ServerResponse.json (/home/pi/alexa-avs-sample-app/samples/companionService/node_modules/express/lib/response.js:240:15)
    at ServerResponse.send (/home/pi/alexa-avs-sample-app/samples/companionService/node_modules/express/lib/response.js:142:21)
    at /home/pi/alexa-avs-sample-app/samples/companionService/app.js:97:17
    at IncomingMessage.<anonymous> (/home/pi/alexa-avs-sample-app/samples/companionService/authentication.js:322:17)
    at emitNone (events.js:72:20)
    at IncomingMessage.emit (events.js:166:7)
    at endReadableNT (_stream_readable.js:923:12)

npm ERR! alexa-voice-service-sample-companion-service@1.0.0 start: `node ./bin/www`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the alexa-voice-service-sample-companion-service@1.0.0 start script.
npm ERR! This is most likely a problem with the alexa-voice-service-sample-companion-service package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node ./bin/www
npm ERR! You can get their info via:
npm ERR!     npm owner ls alexa-voice-service-sample-companion-service
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 4.9.59-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! cwd /home/pi/alexa-avs-sample-app/samples/companionService
npm ERR! node -v v4.8.2
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/alexa-avs-sample-app/samples/companionService/npm-debug.log
npm ERR! not ok code 0
pi@raspberrypi:~/alexa-avs-sample-app/samples/companionService $

Kishu18, 2018-02-06:
in the second terminal it says

 Trying again in 2 seconds. Please make sure it is up and running.
20:07:18.028 [Thread-28] ERROR com.amazon.alexa.avs.auth.companionservice.CompanionServiceAuthManager - There was a problem connecting to the Companion Service. Trying again in 2 seconds. Please make sure it is up and running.
20:07:19.573 [Thread-40] ERROR com.amazon.alexa.avs.auth.companionservice.CompanionServiceAuthManager - There was a problem connecting to the Companion Service. Trying again in 2 seconds. Please make sure it is up and running.
20:07:19.854 [Thread-32] ERROR com.amazon.alexa.avs.auth.companionservice.CompanionServiceAuthManager - There was a problem connecting to the Companion Service. Trying again in 2 seconds. Please make sure it is up and running.
20:07:19.862 [Thread-36] ERROR com.amazon.alexa.avs.auth.companionservice.CompanionServiceAuthManager - There was a problem connecting to the Companion Service. Trying again in 2 seconds. Please make sure it is up and running.
20:07:19.862 [Thread-38] ERROR com.amazon.alexa.avs.auth.companionservice.CompanionServiceAuthManager - There was a problem connecting to the Companion Service. Trying again in 2 seconds. Please make sure it is up and running.
20:07:19.864 [Thread-34] ERROR com.amazon.alexa.avs.auth.companionservice.CompanionServiceAuthManager - There was a problem connecting to the Companion Service. Trying again in 2 seconds. Please make sure it is up and running.
20:07:19.871 [Thread-42] ERROR com.amazon.alexa.avs.auth.companionservice.CompanionServiceAuthManager - There was a problem connecting to the Companion Service. Trying again in 2 seconds. Please make sure it is up and running.

Kishu18, 2018-02-06:
in the third terminal it shows

pi@raspberrypi:~ $ KITT_AI: cd /home/pi/alexa-avs-sample-app/samples/wakeWordAgent/src && sudo ./wakeWordAgent -e kitt_ai
bash: KITT_AI:: command not found

Kishu18, 2018-02-06:
please help me

shivasiddharth, 2018-02-06:
1. You have manually started the companion app even after having the service running.
2. Error is because of having multiple companionapp running.
3. The command you have used is wrong.

Kishu18, 2018-02-07:
what should i do now

Kishu18, 2018-02-07:
if you have teamviewer can you please takeover my pc and look into the issue.......please

shivasiddharth, 2018-02-07:
This is not a product tech support to take remote access, this is DIY. I have already given the reasons for what you are seeing on the screen. Upon restart of Pi post the outputs of 
```
sudo systemctl status companionapp.service
sudo systemctl status client.service
sudo systemctl status wakeword.service
```

Kishu18, 2018-02-07:
● wakeword.service - Wakeword Service
   Loaded: loaded (/lib/systemd/system/wakeword.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-02-07 14:54:00 UTC; 7min ago
 Main PID: 441 (bash)
   CGroup: /system.slice/wakeword.service
           ├─ 441 /bin/bash /home/pi/Assistants-Pi/scripts/wakeword.sh
           ├─1026 sudo ./wakeWordAgent -e kitt_ai
           └─1030 ./wakeWordAgent -e kitt_ai

Feb 07 14:55:10 raspberrypi bash[441]: INFO:WakeWordAgent: thread started
Feb 07 14:56:08 raspberrypi bash[441]: INFO:KittAiSnowboyWakeWordEngine: Wake Word Detected
Feb 07 14:56:08 raspberrypi bash[441]: INFO:===> WakeWordAgent: wake word detected <===
Feb 07 14:56:08 raspberrypi bash[441]: INFO:WakeWordAgent: State set to WAKE_WORD_DETECTED(3)
Feb 07 14:56:08 raspberrypi bash[441]: INFO:WakeWordAgent: State set to SENT_WAKE_WORD_DETECTED(4)
Feb 07 14:56:08 raspberrypi bash[441]: INFO:WakeWordAgent: IPC Command received:3
Feb 07 14:56:08 raspberrypi bash[441]: INFO:WakeWordAgent: State set to WAKE_WORD_PAUSE_REQUESTED(5)
Feb 07 14:56:08 raspberrypi bash[441]: INFO:KittAiSnowboyWakeWordEngine: handling pause
Feb 07 14:56:08 raspberrypi bash[441]: INFO:KittAiSnowboyWakeWordEngine: releasing portAudio
Feb 07 14:56:08 raspberrypi bash[441]: INFO:WakeWordAgent: State set to WAKE_WORD_PAUSED(6)

Kishu18, 2018-02-07:
pi@raspberrypi:~ $ sudo systemctl status companionapp.service
● companionapp.service - Companion Service
   Loaded: loaded (/lib/systemd/system/companionapp.service; enabled; vendor pre
   Active: active (running) since Wed 2018-02-07 14:54:00 UTC; 5min ago
 Main PID: 439 (bash)
   CGroup: /system.slice/companionapp.service
           ├─439 /bin/bash /home/pi/Assistants-Pi/scripts/companionstart.sh
           ├─451 sudo npm start
           ├─477 npm
           ├─663 sh -c node ./bin/www
           └─664 node ./bin/www

Feb 07 14:54:00 raspberrypi systemd[1]: Started Companion Service.
Feb 07 14:54:01 raspberrypi sudo[451]:       pi : TTY=unknown ; PWD=/home/pi/ale
Feb 07 14:54:01 raspberrypi sudo[451]: pam_unix(sudo:session): session opened fo
Feb 07 14:54:10 raspberrypi bash[439]: > alexa-voice-service-sample-companion-se
Feb 07 14:54:10 raspberrypi bash[439]: > node ./bin/www
Feb 07 14:54:12 raspberrypi bash[439]: This node service needs to be running to 
Feb 07 14:54:16 raspberrypi bash[439]: Listening on port 3000
Feb 07 14:55:24 raspberrypi bash[439]: Successfully retrieved registration code 
lines 1-19/19 (END)...skipping...
● companionapp.service - Companion Service
   Loaded: loaded (/lib/systemd/system/companionapp.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-02-07 14:54:00 UTC; 5min ago
 Main PID: 439 (bash)
   CGroup: /system.slice/companionapp.service
           ├─439 /bin/bash /home/pi/Assistants-Pi/scripts/companionstart.sh
           ├─451 sudo npm start
           ├─477 npm
           ├─663 sh -c node ./bin/www
           └─664 node ./bin/www

Feb 07 14:54:00 raspberrypi systemd[1]: Started Companion Service.
Feb 07 14:54:01 raspberrypi sudo[451]:       pi : TTY=unknown ; PWD=/home/pi/alexa-avs-sample-app/samples/companionService ; USER=root ; COMMAND=/usr/bin/npm start
Feb 07 14:54:01 raspberrypi sudo[451]: pam_unix(sudo:session): session opened for user root by (uid=0)
Feb 07 14:54:10 raspberrypi bash[439]: > alexa-voice-service-sample-companion-service@1.0.0 start /home/pi/alexa-avs-sample-app/samples/companionService
Feb 07 14:54:10 raspberrypi bash[439]: > node ./bin/www
Feb 07 14:54:12 raspberrypi bash[439]: This node service needs to be running to store token information memory and vend them for the AVS app.
Feb 07 14:54:16 raspberrypi bash[439]: Listening on port 3000
Feb 07 14:55:24 raspberrypi bash[439]: Successfully retrieved registration code for AnirudhBot / 123456789

Kishu18, 2018-02-07:
● wakeword.service - Wakeword Service
   Loaded: loaded (/lib/systemd/system/wakeword.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-02-07 14:54:00 UTC; 7min ago
 Main PID: 441 (bash)
   CGroup: /system.slice/wakeword.service
           ├─ 441 /bin/bash /home/pi/Assistants-Pi/scripts/wakeword.sh
           ├─1026 sudo ./wakeWordAgent -e kitt_ai
           └─1030 ./wakeWordAgent -e kitt_ai

Feb 07 14:55:10 raspberrypi bash[441]: INFO:WakeWordAgent: thread started
Feb 07 14:56:08 raspberrypi bash[441]: INFO:KittAiSnowboyWakeWordEngine: Wake Word Detected
Feb 07 14:56:08 raspberrypi bash[441]: INFO:===> WakeWordAgent: wake word detected <===
Feb 07 14:56:08 raspberrypi bash[441]: INFO:WakeWordAgent: State set to WAKE_WORD_DETECTED(3)
Feb 07 14:56:08 raspberrypi bash[441]: INFO:WakeWordAgent: State set to SENT_WAKE_WORD_DETECTED(4)
Feb 07 14:56:08 raspberrypi bash[441]: INFO:WakeWordAgent: IPC Command received:3
Feb 07 14:56:08 raspberrypi bash[441]: INFO:WakeWordAgent: State set to WAKE_WORD_PAUSE_REQUESTED(5)
Feb 07 14:56:08 raspberrypi bash[441]: INFO:KittAiSnowboyWakeWordEngine: handling pause
Feb 07 14:56:08 raspberrypi bash[441]: INFO:KittAiSnowboyWakeWordEngine: releasing portAudio
Feb 07 14:56:08 raspberrypi bash[441]: INFO:WakeWordAgent: State set to WAKE_WORD_PAUSED(6)
~

Kishu18, 2018-02-07:
these are the results of the code input above

shivasiddharth, 2018-02-07:
Everything is working as it should be. There is is no issue with the Alexa. Closing this thread now.

closed 2018-02-07 by shivasiddharth