shivasiddharth/Assistants-Pi #44: Alexa works headless/wakeword on Pi zeroW with Respeaker 2mic -Not an issue

jabulonboaz, 2018-02-09:
Note: i do not know how to submit this as a contribution to this project so i wrote it here i hope that @Sid will use this in a coments section for others to benefit from and thank uou Sid!
I have successfully instaled  using AssistantsPi Alexa on a Pi zero W using the Pi3 instal process and it works with wakeword activation - no need for the button.
I have not only completed the install on a Pi3 but allso on a PiOw and it works like a charm. I have only installed Alexa from the options of assistant pi as with Google assistant and Alexa running simultaneously it won't work on Pi0 but it does on Pi3 btw.
The way i done it was to install the AssitantsPi by choosing option 2 Alexa only and choose 1 for AIY hat as soundcard option. That leaves the asound.conf empty. I have an AIY hat so i could test the sound but u don't need to have one i believe you can just go ahead and finish the install any way without confirming the sound check. Then when i finished installing on the pi3 (btw i skipped the very last command of the instructions on the Assistant Pi git that activated Google assistant) i removed the AIY hat and placed the Respeaker 2 mic on the Pi3. I Then downloaded the Rspeaker 2mic drivers from their git https://github.com/respeaker/seeed-voicecard and the rebooted.
Miraculously the respeaker drivers had deleted the only other appearing audio card that was the AIY hat's and the asound.conf file was populated. BTW Copy pasting the contents of the asound to asoundrc made no difference.
Alexa is working perfectly the only thing there is no "hello" sound when she starts up it must be because i skipped the last command that was supposed to activate the Google assistant i assume it contained some .wav for Alexa or maybe simply the instal script runs in a way that if u install only alexa you get no "hello" at startup. Any way i then took the very SD card with that Pi3 build and another miracle happened when i put it and the respeaker on a Pi zero W it works there too and i have sound from all channels! Only thing on the Pi0 it takes a good 3 to 5 minutes to boot up as the Pi0 is slow.
Any way very please with Alexa from AssistantsPi works headless and with the hotword activation -no button needed, and the package of the device is so small it can fit into anything. My next step it to use Snowboy for custom wake words and activate some of the GPiO pins to be able to control a relay from the PiO. Thank you Sid!

shivasiddharth, 2018-02-12:
Soon i will start supporting Re-Speaker HATS.

jabulonboaz, 2018-02-12:
Looking forward to this optimisation @Shivasiddharth and if you ever get the time and if possible to implement the Respeaker 2mic hat LEDs when Alexa listens and responds for Pi0W and both GA and Alexa for Pi3vB. 
Final request is for audio triggered shutdown, volume control of Alexa and local GPIO pin control for AssistantsPi.
Thank u again for giving to us this fantastic project.

shivasiddharth, 2018-02-12:
I have added a note in this git saying that GA on this and GassistPi is same, also not to mention the number of times that i have repeated that in the issues, i have given syntaxes for all the commands in the other git, i have mentioned as to how you can run it manually as well. All that one has to do is follow the notes in README.

jabulonboaz, 2018-02-12:
Fair enough i will read through the GassitPi carefully and run it manually on AssitantsPi and provide the output because non of the local triggering is working but i am sure i mist be doing something wrong myself.

jabulonboaz, 2018-02-12:
@shivasiddharth Here are my traces as you requested as you can see GA detects the words "trigger shutdown" correctly but it just not shutdown the Pi and i don't hear the audio confirmation. 
I was able however to trigger "trigger bathroom lights" and get the correct audio response from the local GA. i think the GPIO trigger functionality started working after i installed the GA update from the URL you have on your git. However i got some error message and i was not sure what command to use to run it so i tried "curl" and i put the URL in "" and ' ' and also tried "sudo apt-get install " and the URL with out and with "" and ' '  but then i got some error messages at the end that could not find some directories:
sudo apt-get install 'https://github.com/shivasiddharth/GassistPi/blob/update-script/GassistPi-Update-Script.sh'

/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'xxxxxxxxxxxxxxxx'  --device_model_id 'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'trigger shutdown'}
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'trigger shutdown'}
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}
 
apologies but i am not an expert hope you can help.

shivasiddharth, 2018-02-12:
The way the GA recognizes commands changes from time to time. Open the actions.py file and scroll down below towards the end. under `def Action(phrase):`  remove the space between `shut down` save and restart the assistant.

You need to first download the update script.  open the link, click on the "RAW" button that you see on right. Then right click in the browser and choose save as. Then:
sudo chmod +x path to the file
sudo path to the file

jabulonboaz, 2018-02-12:
@shivasiddharth ok thank you for this i will update according to your instructions.
FYI it had already occurred to me and after carefully reading your readme notes i have managed to find "actions.py" and changed "shut down" to "shutdown" but still no luck. I have also changed a couple of the lights GPIO triggers to e.g. from "kitchen lights" to "WiFi" and GA recognises them correctly as it does the new "shutdown" but there is no audio response. I left "bathroom lights" unchanged and for that i get the audio response fine.
Thank you for this i will do what u said and report back.

jabulonboaz, 2018-02-13:
@shivasiddharth I did it but no luck GA has the exact same behaviour only thi time for the "bathroom lights"  trigger GA sounds like a human rather than a robot. "shut down does not work even though its recognised and same as i described above for the other changes i made they are recognised but no responses.
Also Alex does not start up and has stooped working after the GassistPi upgrade..
Excuse my ignorance but what am i missing here?

shivasiddharth, 2018-02-13:
Issue with the update script has already been discussed here. #27 After updating, you still need to check the shutdown command and change it in the actions.py. Closing this thread..

closed 2018-02-13 by shivasiddharth

jabulonboaz, 2018-02-13:
hello sorry i am afraid i will have to reopen i just added all my data and i do not follow, i did exactly what you asked @shivasiddharth and it is obvious that there is a bug or an issue here. Please kindly read my edited response and clarify?

jabulonboaz, 2018-02-13:
----------------------------------------Code alterations:

GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
#Number of entities in 'var' and 'PINS' should be the same
var = ('Wi-Fi', 'bathroom lights', 'WiFi')#Add whatever names you want. This is$
gpio = (12,13,24)#GPIOS for 'var'. Add other GPIOs that you want

#GPIO Device Control
def Action(phrase):
if 'shutdown' in phrase:
say('Shutting down Raspberry Pi')
time.sleep(10)
os.system("sudo shutdown -h now")

----------------------------------------Debug results:

ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
{'text': 'trigger shutdown'}
ON_CONVERSATION_TURN_FINISHED:
{'with_follow_on_turn': False}

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
{'text': 'trigger Wi-Fi on'}
ON_CONVERSATION_TURN_FINISHED:
{'with_follow_on_turn': False}

ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
{'text': 'trigger bathroom lights on'}
Turning On bathroom lights
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
version 1.23.8; written and copyright by Michael Hipp and others
free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:02] Decoding of say.mp3 finished.

jabulonboaz, 2018-02-13:
@shivasiddharth yes yes ok i am ok with the Alexa issue i read it ok i will install it again i am pointing out thar the GA issue is still here even after the update even if i change the shut down to shutdown etc

 ----------------------------------------Code alterations:

GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
#Number of entities in 'var' and 'PINS' should be the same
var = ('Wi-Fi', 'bathroom lights', 'WiFi')#Add whatever names you want. This is$
gpio = (12,13,24)#GPIOS for 'var'. Add other GPIOs that you want

#GPIO Device Control
def Action(phrase):
if 'shutdown' in phrase:
say('Shutting down Raspberry Pi')
time.sleep(10)
os.system("sudo shutdown -h now")

----------------------------------------Debug results:

ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
{'text': 'trigger shutdown'}
ON_CONVERSATION_TURN_FINISHED:
{'with_follow_on_turn': False}

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
{'text': 'trigger Wi-Fi on'}
ON_CONVERSATION_TURN_FINISHED:
{'with_follow_on_turn': False}

ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
{'text': 'trigger bathroom lights on'}
Turning On bathroom lights
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
version 1.23.8; written and copyright by Michael Hipp and others
free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:02] Decoding of say.mp3 finished.

shivasiddharth, 2018-02-13:
Try changing the `say('Shutting down Raspberry Pi')` to `say("Shutting down Raspberry Pi")` .

jabulonboaz, 2018-02-13:
@shivasiddharth sure i will give it a go thank u.
Are these lines in acrion.py too?

shivasiddharth, 2018-02-13:
Yes.

jabulonboaz, 2018-02-13:
@shivasiddharth nope... no luck i am afraid it did not work there is something wrong here and also changing any of the trigger names i don't get a response only thing i get a response for  is "bathroom lights" which i left intact. the words are recorded all fine as you will see bellow but no action or response.

shivasiddharth, 2018-02-13:
Attach your log file and actions.py file here. "Attach" dont paste.

jabulonboaz, 2018-02-13:
@shivasiddharth how do i attach the logs please?

shivasiddharth, 2018-02-13:
You can find the log file in the /tmp/ directory of Pi. Do you see an "Attach" option down below, in the window where you type in all these comments ? Click on that and then select those files.

jabulonboaz, 2018-02-13:
@shivasiddharth there are many folders in the /tmp/ directory wich one should i attache? some folders have "SSH" in their name.
This morning after playing with actions.py i saw its content suddenly deleted so then i decided to restore assitantspi from a clone SD. Now i am back from work and i run the GassistPi update URL again then the Alexa restore commands  sudo apt-get install npm and then restarted. After i went into the /GassistPi/src/actions.py  but not via SSH through VNC and found the file and edited it like that:
var = ('test', 'bathroom lights', 'internet')#Add whatever names you want. This is case is insensitive

 if 'shutdown' in phrase:
        say("Shutting down Raspberry Pi")

and this time everything worked! i do not know what changed or what was wrong before so i still want to send you the logs here to see if you find anything useful if you tell me which folder and file in tmp to load.
I will not go and try to change the word "trigger" as per your instructions in the main.py and assistant.py. Where exactly is the assistant.py located please?

jabulonboaz, 2018-02-13:
one strange thing was that when i was trying to go not to actions.py via SSH and sudo nano it was empty and when i opened the file in a VNC session its was ok

jabulonboaz, 2018-02-13:
i believe now that last time i was altering another actions.py file that is in AssistansPi folder.  BTW where is  the assitant.py file located please?

jabulonboaz, 2018-02-13:
I found assistant.py bout there is not "trigger" word in it.  Should i just change main.py?

shivasiddharth, 2018-02-13:
within the /tmp dierectory, there will be a file called Gassistpi.log that is the log file. Regarding the editing of the trigger word, its again there in the readme. When you use the update script, old GassistPi folder will be renamed and the date to updation will be appended to the old folder name. The latest scripts will be cloned onto the GassistPi folder. If you make any change to the old folder, it will not work.

jabulonboaz, 2018-02-13:
@shivasiddharth thank you.
"It has been intentionally included to prevent control actions due to false positive commands. If you wish to change the trigger word, you can replace the 'trigger'in the main.py and assistant.py code with your desired trigger word."
This is the quote from the read me file, but it is not clear to me what to change in "assitant.py" as i did "find" and when i open it there is not mentioning of the word "trigger" or there is more than one and if yes which directory and sub-folders is it in pls?
I will send you some logs any way.

shivasiddharth, 2018-02-13:
As you may ave noticed, this version of GassistPi does not have any script called assistant.py. It was a script that was used for custom wakeword activation. pre-20th Dec. I will make changes to the readme reflecting that.

jabulonboaz, 2018-02-13:
@shivasiddharth can i have more than one trigger words ?

jabulonboaz, 2018-02-15:
@shivasiddharth i do not know where i can ask this question so i place it here and please feel free to point me to the right direction.

How can i point GA to run a script i have instead of say triggering a GPIO pin in the actions.py file please?
E.g "hey Google trigger wifi on" and then a wifi_on.sh script is executed.

Also i have changed the "trigger" word to "home" and everything works with it but the "shuttdown" but will generate some logs for u over the wend like i promised.