shivasiddharth/Assistants-Pi #49: Manual Start of Google Assistant gives "No module named 'google.assistant.library'

rutherb, 2018-03-01:
Hi Shiva,

I have followed the correct instructions for the post December 20th installation from this link https://github.com/shivasiddharth/Assistants-Pi#readme but when I try to start the Google Assistant manually using the /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'replace this with your project id'  --device_model_id 'replace this with the model id' command I get an error as below:-

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 29, in <module>
    from google.assistant.library import Assistant
ImportError: No module named 'google.assistant.library'

Can you help please?

shadowscott, 2018-03-13:
I had this same problem. I had to follow instructions here, however @shivasiddharth mentioned that this is not the way to do it. I would like to know why it isn't installed as part of the scripts either. 

https://github.com/googlesamples/assistant-sdk-python

rutherb, 2018-03-14:
Thanks for the info shadowscott - I will refresh Raspbian and follow above instructions and post update.

shadowscott, 2018-03-14:
Sure thing. I will say that I reinstalled Raspbian and installed the google assistant from GassistPi instead. That worked flawlessly without hitch. I'm now trying to get Alexa installed as well (either through AssistantPi which hasn't worked, or maybe AlexaPi or some other method)

If you manage to get both working on the Google AIY Voice Kit, please let me know!!!

nicusorflorin, 2018-06-03:
any updates on this?, have the same problem

rutherb, 2018-06-03:
Hi - i upgraded to a later Raspberry Pi and reinstalled the Google assistant using these instructions https://github.com/googlesamples/assistant-sdk-python and it installed OK so I think it may have been the version of the PI I was using that caused the issue.

commit 8cda70cb (2018-10-12): Migrated the project to AVS SDK

Fixes #3, Fixes #15, Fixes #69 , Fixes #66 , Fixes #65 , Fixes #64 , Fixes #62 , Fixes #61 , Fixes #60 , Fixes #59 , Fixes #58 , Fixes #57 , Fixes #56 , Fixes #55 , Fixes #54 , Fixes #53 , Fixes #52 , Fixes #51 , Fixes #49 , Fixes #46 , Fixes #38 , Fixes #37 , Fixes #33 , Fixes #20 , Fixes #19 , Fixes #18 , Fixes #16
closed 2018-10-12 by shivasiddharth