shivasiddharth/Assistants-Pi #52: Cannot test google assistant in env

shadowscott, 2018-03-13:
Ran through the installer for both Google and Alexa. Alexa wake work tested fine. However, I cannot get to test the Google portion. Any help?

pi@raspberrypi:~ $ source env/bin/activate
(env) pi@raspberrypi:~ $ googlesamples-assistant-hotword --project_id 'pi-8a638' --device_model_id 'pi-8a638-googlepi-35rp2i'
-bash: googlesamples-assistant-hotword: command not found

shivasiddharth, 2018-03-13:
Are you using Pi Zero ?

shadowscott, 2018-03-13:
Hey @shivasiddharth . No. I'm using Pi 3.

shadowscott, 2018-03-13:
Managed to get it to work after installing a bunch of stuff from here: https://github.com/googlesamples/assistant-sdk-python

Only problem is now that I have it up and running (and tested both assistants) neither of them will run on boot. Can't get this to function at all.

shivasiddharth, 2018-03-13:
Problem is if you followed the official SDK, you will be on a newer version than this project and so obviously, you cannot use my files to auto start GA. But ALexa should autostart on boot without a hitch. Try removing the sudo in the wakesowrd.sh file.

shadowscott, 2018-03-13:
I only installed googlesamples-assistant-hotwork and googlesamples-assistant-library because I could not execute the test in the virtualenv. Had to install a bunch of python scripts there and set up virtual environment. Then was finally able to execute the google-samples-hotword which worked fine. Removed sudo and same thing, Starts up saying "hello" but i can't get Alexa to reply. And google assistant isn't working either.

shadowscott, 2018-03-13:
So @shivasiddharth I got an update. I followed your GassistPi tutorial and got Google Assistant working fine there. So that is operational. Now I'm running Assistants-Pi and selecting only the Alexa option. What portions should I not be running at the bottom regarding the services/file updates since I already have it installed from GassistPi? Also, after installing and rebooting I hear google playing it's intro and I do get a Hello/Hi (can't remember) from Alexa, however she doesn't seem to be responsive when using her wake word despite the test as part of your script with the listening ports showing that the wake word is picking up. Thought?

shivasiddharth, 2018-03-14:
May i know as to what is your audio setup ?

shadowscott, 2018-03-14:
Using AIY hat from google kit.

shivasiddharth, 2018-03-14:
Can you check if your .asoundrc and asound.conf are the same ?

shadowscott, 2018-03-14:
Yes, they are the same.

commit 8cda70cb (2018-10-12): Migrated the project to AVS SDK

Fixes #3, Fixes #15, Fixes #69 , Fixes #66 , Fixes #65 , Fixes #64 , Fixes #62 , Fixes #61 , Fixes #60 , Fixes #59 , Fixes #58 , Fixes #57 , Fixes #56 , Fixes #55 , Fixes #54 , Fixes #53 , Fixes #52 , Fixes #51 , Fixes #49 , Fixes #46 , Fixes #38 , Fixes #37 , Fixes #33 , Fixes #20 , Fixes #19 , Fixes #18 , Fixes #16
closed 2018-10-12 by shivasiddharth