shivasiddharth/Assistants-Pi #59: Could not find or load main class com.amazon.alexa.avs.App

oneilbogle1, 2018-05-10:
i tried 

pi@raspberrypi:~/alexa-avs-sample-app/samples/javaclient $  sudo mvn exec:exec

Warning: JAVA_HOME environment variable is not set.
[INFO] Scanning for projects...
[INFO]
[INFO] ------------------------------------------------------------------------
[INFO] Building Alexa Voice Service Sample Java Client 20160207.7
[INFO] ------------------------------------------------------------------------
[INFO]
[INFO] --- exec-maven-plugin:1.2.1:exec (default-cli) @ sample-java-client ---
Error: Could not find or load main class com.amazon.alexa.avs.App
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 4.850 s
[INFO] Finished at: 2018-05-10T15:11:33+00:00
[INFO] Final Memory: 8M/20M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.codehaus.mojo:exec-maven-plugin:1.2.1:exec (default-cli) on project sample-java-client: Command execution failed. Process exited with an error: 1 (Exit value: 1) -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException

i tried setting the variable by  nano ~/.bashrc

export JAVA_HOME="/usr/lib/jvm/java-8-openjdk-armhf"
export PATH=$PATH:$JAVA_HOME/bin

did not wrk can someone help

smowk, 2018-05-18:
same problem here

ricklambrechts, 2018-06-04:
If you still facing this issue. This did it for me.

1. Check your Java version.
`java -version`<br>
Example output:
pi@raspberrypi:~ $ java -version
openjdk version "1.8.0_171"
OpenJDK Runtime Environment (build 1.8.0_171-8u171-b11-1~deb9u1-b11)
OpenJDK Client VM (build 25.171-b11, mixed mode)

2. With your java version, check the alpn-version [here](http://www.eclipse.org/jetty/documentation/current/alpn-chapter.html#alpn-versions).
For example with 1.8.0_171, the alpn-version is '8.1.12.v20180117'.

3. With the version you will execute it.
`sudo mvn exec:exec -Dalpn-boot.version=<alpn-version>` <br>
For example with java 1.8.0_171:
`sudo mvn exec:exec -Dalpn-boot.version=8.1.12.v20180117`

Hope this helps.

commit 8cda70cb (2018-10-12): Migrated the project to AVS SDK

Fixes #3, Fixes #15, Fixes #69 , Fixes #66 , Fixes #65 , Fixes #64 , Fixes #62 , Fixes #61 , Fixes #60 , Fixes #59 , Fixes #58 , Fixes #57 , Fixes #56 , Fixes #55 , Fixes #54 , Fixes #53 , Fixes #52 , Fixes #51 , Fixes #49 , Fixes #46 , Fixes #38 , Fixes #37 , Fixes #33 , Fixes #20 , Fixes #19 , Fixes #18 , Fixes #16
closed 2018-10-12 by shivasiddharth