shivasiddharth/Assistants-Pi #62: Mic test failed and folders missing

exthazor, 2018-05-24:
I don't have the systemd folder inside my GassistPi. Also the audio test succeeds but the mic test fails. I'm also missing the audio-drivers and the sample-audio-files folder. 

The error during recording test is more like "ALSA lib pcm_hw.c1713:(_snd_pcm_hw_open) Invalid value for card 
arecord: main:788: audio open error: No such file or directory. It finished recording in a blink of an eye lol.

exthazor, 2018-05-24:
And yes, in spite of the mic test failing, I went through the other stages of installation only to find that those folders were missing after installation.

idandro, 2018-07-26:
Looks like you have an incomplete copy of GassistPi git repo. Try and start from beginning again, particularly with the step where you copy (clone) the repository, that is: 

git clone https://github.com/shivasiddharth/GassistPi    

I just tried that and sure enough there is a systemd folder in it which contains start-up scripts necessary to automatically start Google Assistant at boot time. As to your mic failing it looks like you did not configure your sound card correctly. If you're using a USB ADC / sound card for your mic input make sure you have it plugged in and that it shows up when you issue 'lsusb' command. Here is what mine looks like:

(without USB microphone) -
pi@gaspi:~ $ lsusb
Bus 001 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

(with USB mic plugged in) -
pi@gaspi:~ $ lsusb
Bus 001 Device 009: ID e4b7:0812  
Bus 001 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

NOTE that second run of the command contains an extra line at the top "Bus 001 Device 009: ID e4b7:0812" - in my case it doesn't even have a proper device name for it is a super cheap generic USB microphone, nevertheless it does show up and so should yours. So make sure you've got these two things covered first and then try and go through the installation instructions again. 

Good Luck!

commit 8cda70cb (2018-10-12): Migrated the project to AVS SDK

Fixes #3, Fixes #15, Fixes #69 , Fixes #66 , Fixes #65 , Fixes #64 , Fixes #62 , Fixes #61 , Fixes #60 , Fixes #59 , Fixes #58 , Fixes #57 , Fixes #56 , Fixes #55 , Fixes #54 , Fixes #53 , Fixes #52 , Fixes #51 , Fixes #49 , Fixes #46 , Fixes #38 , Fixes #37 , Fixes #33 , Fixes #20 , Fixes #19 , Fixes #18 , Fixes #16
closed 2018-10-12 by shivasiddharth