shivasiddharth/Assistants-Pi #69: Exception in building Alexa

Shane-Lester, 2018-10-10:
I got this exception (second time I tried- different one the first time)

>[INFO]
[INFO] ------------------------------------------------------------------------
[INFO] Building Alexa Voice Service Sample Java Client 20160207.7
[INFO] ------------------------------------------------------------------------
[INFO]
[INFO] --- exec-maven-plugin:1.2.1:exec (default-cli) @ sample-java-client ---
07:03:55.348 [main] ERROR com.amazon.alexa.avs.AVSController - Error creating Wake Word IPC ok.
java.net.BindException: Address already in use (Bind failed)
        at java.net.PlainSocketImpl.socketBind(Native Method) ~[?:1.8.0_181]
        at java.net.AbstractPlainSocketImpl.bind(AbstractPlainSocketImpl.java:387) ~[?:1.8.0_181]
        at java.net.ServerSocket.bind(ServerSocket.java:375) ~[?:1.8.0_181]
        at java.net.ServerSocket.<init>(ServerSocket.java:237) ~[?:1.8.0_181]
        at com.amazon.alexa.avs.wakeword.WakeWordIPCSocket.<init>(WakeWordIPCSocket.java:34) ~[classes/:?]
        at com.amazon.alexa.avs.wakeword.WakeWordIPCFactory.createWakeWordIPC(WakeWordIPCFactory.java:21) ~[classes/:?]
        at com.amazon.alexa.avs.AVSController.<init>(AVSController.java:137) [classes/:?]
        at com.amazon.alexa.avs.App.<init>(App.java:50) [classes/:?]
        at com.amazon.alexa.avs.App.<init>(App.java:39) [classes/:?]
        at com.amazon.alexa.avs.App.main(App.java:34) [classes/:?]
07:04:05.394 [main] ERROR com.amazon.alexa.avs.AVSController - There was a problem connecting to the wake word engine.
java.lang.NullPointerException
        at com.amazon.alexa.avs.AVSController$3.call(AVSController.java:249) ~[classes/:?]
        at com.amazon.alexa.avs.AVSController$3.call(AVSController.java:245) ~[classes/:?]
        at com.amazon.alexa.avs.http.AbstractRetryPolicy.tryCall(AbstractRetryPolicy.java:38) ~[classes/:?]
        at com.amazon.alexa.avs.AVSController.initializeMicrophone(AVSController.java:262) [classes/:?]
        at com.amazon.alexa.avs.AVSController.<init>(AVSController.java:146) [classes/:?]
        at com.amazon.alexa.avs.App.<init>(App.java:50) [classes/:?]
        at com.amazon.alexa.avs.App.<init>(App.java:39) [classes/:?]
        at com.amazon.alexa.avs.App.main(App.java:34) [classes/:?]
Device: MyAlexaAssist, DSN: 123456789
Alexa Voice Service - v20160207.7
Exception in thread "Thread-23" java.lang.NoClassDefFoundError: sun/security/ssl/SupportedEllipticCurvesExtension
        at sun.security.ssl.ClientHandshaker.getKickstartMessage(ClientHandshaker.java:1438)
        at sun.security.ssl.Handshaker.kickstart(Handshaker.java:1087)
        at sun.security.ssl.SSLSocketImpl.kickstartHandshake(SSLSocketImpl.java:1497)
        at sun.security.ssl.SSLSocketImpl.performInitialHandshake(SSLSocketImpl.java:1361)
        at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1413)
        at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1397)
        at sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:559)
        at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:185)
        at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1564)
        at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1492)
        at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
        at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:347)
        at com.amazon.alexa.avs.auth.companionservice.CompanionServiceClient.callService(CompanionServiceClient.java:224)
        at com.amazon.alexa.avs.auth.companionservice.CompanionServiceClient.getRegistrationCode(CompanionServiceClient.java:157)
        at com.amazon.alexa.avs.auth.companionservice.CompanionServiceAuthManager.requestRegistrationCode(CompanionServiceAuthManager.java:74)
        at com.amazon.alexa.avs.auth.companionservice.CompanionServiceAuthManager.startNewProvisioningRequest(CompanionServiceAuthManager.java:66)
        at com.amazon.alexa.avs.auth.companionservice.CompanionServiceAuthManager.startRemoteProvisioning(CompanionServiceAuthManager.java:61)
        at com.amazon.alexa.avs.auth.AuthSetup.lambda$startProvisioningThread$1(AuthSetup.java:90)
        at java.lang.Thread.run(Thread.java:748)

closed 2018-10-12 by shivasiddharth

Mohamed19966, 2019-02-24:
what did you do to solve it  ?

kmpathak1610, 2019-02-25:
I have solved that problem. But now when I am reconfiguring some user
permissions issues is coming. its saying /root/modelid.txt not found.

Pls help me in solving this issue.

Regards,
Krishna

Mohamed19966, 2019-02-25:
how did you solve it

commit 8cda70cb (2018-10-12): Migrated the project to AVS SDK

Fixes #3, Fixes #15, Fixes #69 , Fixes #66 , Fixes #65 , Fixes #64 , Fixes #62 , Fixes #61 , Fixes #60 , Fixes #59 , Fixes #58 , Fixes #57 , Fixes #56 , Fixes #55 , Fixes #54 , Fixes #53 , Fixes #52 , Fixes #51 , Fixes #49 , Fixes #46 , Fixes #38 , Fixes #37 , Fixes #33 , Fixes #20 , Fixes #19 , Fixes #18 , Fixes #16