shivasiddharth/Assistants-Pi #75: Problem with routine say(word) in actions.pi

cnoorman, 2018-10-23:
Hi, I have installed GassistPi through AssistantsPi on a RPI-3B using the original Google AIY hat. 
All the regular Google Assistant features work fine. However as soon as I try to use a feature that involves the use of the routine **say** (for instance _Hey-Google trigger shutdown_), the system freezes. When I start  the assistant manually I see it crashing in line 59 of gtoken.py.

`ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "trigger shutdown"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "trigger shutdown"}
Shutdown_CNO
Shutting down Raspberry Pi
en
ERROR:__main__:'NoneType' object has no attribute 'group'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 667, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 517, in main
    Action(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line `1say528,` in Action
    say('Shutting down Raspberry Pi')
  File "/home/pi/GassistPi/src/actions.py", line 221, in say
    newword=trans(words,language)
  File "/home/pi/GassistPi/src/actions.py", line 212, in trans
    transword= translator.translate(words, dest=lang)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 172, in translate
    data = self._translate(text, dest, src)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 75, in _translate
    token = self.token_acquirer.do(text)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 180, in do
    self._update()
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 59, in _update
    code = unicode(self.RE_TKK.search(r.text).group(1)).replace('var ', '')
AttributeError: 'NoneType' object has no attribute 'group'
`
Same error for various other requests in which the assistant should talk through the translating routine **say**.

BTW I notes the following during startup of the assistant:
`
[014af378] pulse audio output error: PulseAudio server connection failure: Connection refused
[014eb538] pulse audio output error: PulseAudio server connection failure: Connection refused
`
Can this be the cause, and how to prevent this?
Thanks -- Chris

shivasiddharth, 2018-10-23:
Pulseaudio has nothing to do with this. This has already been discussed in the GassistPi git https://github.com/shivasiddharth/GassistPi/issues/534

cnoorman, 2018-10-24:
OK, I missed that. Replacing the file gtoken.py as mentioned works.
Thx!

closed 2018-10-24 by cnoorman