shivasiddharth/Assistants-Pi #78: 2 tests failed out of 920

liberliutw, 2018-10-30:
when I tried to build Assistants-Pi to my system but meet some issue. hope someone can help me to fix it.

1) run sudo /home/pi/Assistants-Pi/scripts/installer.sh on my system, I didn't saw any 	
wrong message but in the last step popup failed out of 920, please reference attached info
---------------------------------------------------------------------------------------------------
99% tests passed, 2 tests failed out of 920

Total Test time (real) = 1788.06 sec

The following tests FAILED:
	788 - SettingsTest.addGlobalSettingsObserverTest (OTHER_FAULT)
	791 - SettingsTest.removeSingleSettingObserverWithInvalidKeyTest (Timeout)
Errors while running CTest
Makefile:116: recipe for target 'test' failed
make: *** [test] Error 8
========================Finished Installing Amazon Alexa========================
---------------------------------------------------------------------------------------------------

2)  enable alexa.service but failed
---------------------------------------------------------------------------------------------------
pi@raspberrypi:~ $ sudo systemctl status alexa.service
? alexa.service - Companion Service
   Loaded: loaded (/usr/lib/systemd/system/alexa.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2018-10-30 16:48:14 CST; 11min ago
 Main PID: 18555 (code=exited, status=217/USER)

10? 30 16:48:13 raspberrypi systemd[1]: alexa.service: Failed with result 'exit-code'.
10? 30 16:48:14 raspberrypi systemd[1]: alexa.service: Service hold-off time over, scheduling restart.
10? 30 16:48:14 raspberrypi systemd[1]: Stopped Companion Service.
10? 30 16:48:14 raspberrypi systemd[1]: alexa.service: Start request repeated too quickly.
10? 30 16:48:14 raspberrypi systemd[1]: Failed to start Companion Service.
10? 30 16:48:14 raspberrypi systemd[1]: alexa.service: Unit entered failed state.
10? 30 16:48:14 raspberrypi systemd[1]: alexa.service: Failed with result 'exit-code'.
10? 30 16:49:08 raspberrypi systemd[1]: [/usr/lib/systemd/system/alexa.service:9] Failed to parse output specifier, ignoring: tty,
pi@raspberrypi:~ $ 
---------------------------------------------------------------------------------------------------

cnoorman, 2018-10-31:
It looks to me that the first error is caused by the fact that the authentication with Amazon went wrong.
Make sure that the ClientID that you use is not the one from the top of the product information page, but the one generated within the security profile (see [here](https://github.com/alexa/avs-device-sdk/wiki/Create-Security-Profile#create-a-security-profile)). Also you should delete the `cblAuthDelegate.db` file in `/home/pi/Assistants-Pi/Alexa/db` before trying to authenticate again.

The second error is because of errors in the `alexa.service` file (see [here](https://github.com/shivasiddharth/GassistPi/issues/607#issue-375429734)). Have a look at it and you'll see that variable `__USER__` did not translate properly. Just replace all `__USER__` by `pi` and the service script will be OK.
Furthermore there is a typo in line `StandardOutput=tty,`. The comma should not be there, but that is not a problem. 
I do not understand why your `alexa.service` file is in folder `/usr/lib/systemd/system/`? With me it's in `/lib/systemd/system/` but I don't think that that is a problem (?).

Realise that running Alexa as a service will only work if you solve the first problem!

liberliutw, 2018-11-01:
Thanks for your prompt reply, I'm following the insertion and authenticate again use client Id & productId under "Other devices and platforms" section, but get below failure info again.

--------------------------------------------------------------------------------------------------
99% tests passed, 3 tests failed out of 920

Total Test time (real) = 3282.60 sec

The following tests FAILED:
	789 - SettingsTest.removeGlobalSettingsObserverTest (OTHER_FAULT)
	791 - SettingsTest.removeSingleSettingObserverWithInvalidKeyTest (Timeout)
	792 - SettingsTest.removeSingleSettingObserverWithCorrectKeyTest (Timeout)
Errors while running CTest
Makefile:116: recipe for target 'test' failed
make: *** [test] Error 8
--------------------------------------------------------------------------------------------------

--------------------------------------------------------------------------------------------------
789/920 Test #789: SettingsTest.removeGlobalSettingsObserverTest ............................................................***Exception: Other  1.61 sec
        Start 790: SettingsTest.addSingleSettingObserverWithInvalidKeyTest
790/920 Test #790: SettingsTest.addSingleSettingObserverWithInvalidKeyTest ..................................................   Passed    3.04 sec
        Start 791: SettingsTest.removeSingleSettingObserverWithInvalidKeyTest
791/920 Test #791: SettingsTest.removeSingleSettingObserverWithInvalidKeyTest ...............................................***Timeout 1500.02 sec
        Start 792: SettingsTest.removeSingleSettingObserverWithCorrectKeyTest
792/920 Test #792: SettingsTest.removeSingleSettingObserverWithCorrectKeyTest ...............................................***Timeout 1500.02 sec
        Start 793: SettingsTest.defaultSettingsCorrect
--------------------------------------------------------------------------------------------------

pls advise me which part I'm missing

cnoorman, 2018-11-01:
I ran the test script again on my working Alexa installation:
```
cd /home/pi/Assistants-Pi/Alexa/
sudo ./test.sh
```
and got exactly the same errors. No idea why. 
Did you check if you have 
`/home/pi/Assistants-Pi/Alexa/startsample.sh` ?
If so try to start it 
`sudo /home/pi/Assistants-Pi/Alexa/startsample.sh`

liberliutw, 2018-11-02:
After I tried run installer.sh  again and script create some DB files in /home/pi/Assistants-Pi/Alexa/db folder, original I used pi3+ got failure message so I change to use pi2+ to build again but it still popup same message says 
```
99% tests passed, 1 tests failed out of 920

Total Test time (real) = 1848.14 sec

The following tests FAILED:
	792 - SettingsTest.removeSingleSettingObserverWithCorrectKeyTest (Timeout)
Errors while running CTest
Makefile:116: recipe for target 'test' failed
make: *** [test] Error 8
========================Finished Installing Amazon Alexa========================
```

if run` /home/pi/Assistants-Pi/Alexa/startsample.sh` I can hear the voice from the speaker says "Hi", but when I say ALEXA nothing responses.

liberliutw, 2018-11-07:
I used ReSpeaker-4-Mic-Array-for-Raspberry-Pi for MIC.
https://github.com/SeeedDocument/ReSpeaker-4-Mic-Array-for-Raspberry-Pi/blob/master/ReSpeaker-4-Mic-Array-for-Raspberry-Pi.md

I follow this instruction and downgrade Raspbian to 2018-03-14 for pi 3+, now Alexa can respond my voice, it still needs manually mode a lot of work but that is fine. a least it already works.

how to Alexa-voice-service instruction 
https://developer.amazon.com/docs/alexa-voice-service/set-up-raspberry-pi.html

Raspbian os download link
http://ftp.jaist.ac.jp/pub/raspberrypi/raspbian/images/raspbian-2018-03-14/

liberliutw, 2018-11-07:
I used Raspbian to 2018-03-14 on pi 2+and run instal.sh get missing: GTEST_LIBRARY GTEST_MAIN_LIBRARY error message, after `sudo apt-get install cmake libgtest-dev` already fixed issue Alexa service works.

siddeshwarpatil, 2018-12-26:
tried again and get the following errors
99% tests passed, 2 tests failed out of 920

Total Test time (real) = 290.23 sec

The following tests FAILED:
	253 - TimeTest.testStringConversion (Failed)
	256 - TimeTest.testTimeConversion (Failed)
Errors while running CTest
Makefile:116: recipe for target 'test' failed
make: *** [test] Error 8
Finished Installing Alexa. Proceeding to install Google Assistant

leftfielder, 2019-06-22:
I am getting the same error message. The output follows below. Any idea on how to solve this issue?
Thanx in advance.
99% tests passed, 4 tests failed out of 920

Total Test time (real) = 2022.97 sec

The following tests FAILED:
	253 - TimeTest.testStringConversion (Failed)
	256 - TimeTest.testTimeConversion (Failed)
	788 - SettingsTest.addGlobalSettingsObserverTest (OTHER_FAULT)
	794 - SettingsTest.clearDataTest (Timeout)
Errors while running CTest
Makefile:116: recipe for target 'test' failed
make: *** [test] Error 8
========================Finished Installing Amazon Alexa========================

priceey, 2019-11-18:
I too get the same error message. Did anyone get a work around?