shivasiddharth/Assistants-Pi #79: Assistants don't auto start when i power on my raspberry pi i have a raspberry pi 3

erosalpe02, 2018-11-02:


erosalpe02, 2018-11-02:
And i have a quwstion where i find model id for install the gapi ?

cnoorman, 2018-11-03:
Should be saved in `/home/pi/modelid.txt` .

erosalpe02, 2018-11-03:
Now my only problem is : my assistants don't start on boot and alexa after few minutes if i start it by command it stop to activate when i say alexa

cnoorman, 2018-11-03:
Show us results of:
```
cat /lib/systemd/system/alexa.service 
cat /lib/systemd/system/gassistpi-ok-google.service
sudo systemctl status alexa.service
sudo systemctl status gassistpi-ok-google.service
```

erosalpe02, 2018-11-03:
cat /lib/systemd/system/alexa.service :
[Unit]
Description=Companion Service
After=network.target

[Service]
Environment=PATH=/home/__USER__/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/us     r/bin:/sbin:/bin
ExecStart=/usr/bin/python -u /home/__USER__/Assistants-Pi/Alexa/alexa.py
WorkingDirectory=/home/__USER__/
StandardOutput=tty,
StandardError=tty
Restart=always
User=__USER__

[Install]
WantedBy=multi-user.target

cat /lib/systemd/system/gassistpi-ok-google.service : 
[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/__USER__/env
Environment=PATH=/home/__USER__/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/us     r/bin:/sbin:/bin
ExecStart=/home/__USER__/env/bin/python -u /home/__USER__/GassistPi/src/main.py      --hahahome-93727-gapi-7wiuf2 'saved-model-id'

WorkingDirectory=/home/__USER__/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=__USER__

[Install]
WantedBy=multi-user.target

sudo systemctl status alexa.service : 

● alexa.service - Companion Service
   Loaded: loaded (/lib/systemd/system/alexa.service; enabled; vendor preset: enabled
   Active: failed (Result: exit-code) since Fri 2018-11-02 21:45:14 CET; 17h ago
  Process: 451 ExecStart=/usr/bin/python -u /home/__USER__/Assistants-Pi/Alexa/alexa.
 Main PID: 451 (code=exited, status=217/USER)

nov 02 21:45:13 raspberrypi systemd[1]: alexa.service: Unit entered failed state.
nov 02 21:45:13 raspberrypi systemd[1]: alexa.service: Failed with result 'exit-code'
nov 02 21:45:14 raspberrypi systemd[1]: alexa.service: Service hold-off time over, sc
nov 02 21:45:14 raspberrypi systemd[1]: Stopped Companion Service.
nov 02 21:45:14 raspberrypi systemd[1]: alexa.service: Start request repeated too qui
nov 02 21:45:14 raspberrypi systemd[1]: Failed to start Companion Service.
nov 02 21:45:14 raspberrypi systemd[1]: alexa.service: Unit entered failed state.
nov 02 21:45:14 raspberrypi systemd[1]: alexa.service: Failed with result 'exit-code'

sudo systemctl status gassistpi-ok-google.service : 

● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor p
   Active: failed (Result: exit-code) since Fri 2018-11-02 21:45:14 CET; 17h ago
  Process: 453 ExecStart=/home/__USER__/env/bin/python -u /home/__USER__/GassistPi/sr
 Main PID: 453 (code=exited, status=217/USER)

nov 02 21:45:13 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exi
nov 02 21:45:13 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered fai
nov 02 21:45:13 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with resu
nov 02 21:45:14 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off
nov 02 21:45:14 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
nov 02 21:45:14 raspberrypi systemd[1]: gassistpi-ok-google.service: Start request re
nov 02 21:45:14 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.
nov 02 21:45:14 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered fai
nov 02 21:45:14 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with resu

please help me 👍

cnoorman, 2018-11-03:
See [here](https://github.com/shivasiddharth/GassistPi/issues/607#issue-375429734).
Replace `__USER__ `by `pi` in 
`/lib/systemd/system/gassistpi-ok-google.service` and 
`/lib/systemd/system/alexa.service`. 
And restart services:
```
sudo systemctl daemon-restart
sudo systemctl start alexa.service
sudo systemctl start gassistpi-ok-google.service
```

erosalpe02, 2018-11-03:
It don't work

cnoorman, 2018-11-04:
Show us results of:
```
cat /lib/systemd/system/alexa.service 
cat /lib/systemd/system/gassistpi-ok-google.service
sudo systemctl status alexa.service
sudo systemctl status gassistpi-ok-google.service
```
If you want us to solve your problem you have to give more information than _"It don't work"_!

erosalpe02, 2018-11-04:
cat /lib/systemd/system/alexa.service 

[Unit]
Description=Companion Service
After=network.target

[Service]
Environment=PATH=/home/__pi__/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/usr/bin/python -u /home/__pi__/Assistants-Pi/Alexa/alexa.py
WorkingDirectory=/home/__pi__/
StandardOutput=tty,
StandardError=tty
Restart=always
User=__pi__

[Install]
WantedBy=multi-user.target

cat /lib/systemd/system/gassistpi-ok-google.service

[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/__pi__/env
Environment=PATH=/home/__pi__/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/__pi__/env/bin/python -u /home/__pi__/GassistPi/src/main.py --hahahome-93727-gapi-7wiuf2 'saved-model-id'

WorkingDirectory=/home/__pi__/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=__pi__

[Install]
WantedBy=multi-user.target

sudo systemctl status alexa.service

● alexa.service - Companion Service
   Loaded: loaded (/lib/systemd/system/alexa.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2018-11-04 20:48:12 CET; 12min ago
 Main PID: 494 (code=exited, status=217/USER)

nov 04 20:48:12 raspberrypi systemd[1]: alexa.service: Unit entered failed state.
nov 04 20:48:12 raspberrypi systemd[1]: alexa.service: Failed with result 'exit-code'.
nov 04 20:48:12 raspberrypi systemd[1]: alexa.service: Service hold-off time over, scheduling restart.
nov 04 20:48:12 raspberrypi systemd[1]: Stopped Companion Service.
nov 04 20:48:12 raspberrypi systemd[1]: alexa.service: Start request repeated too quickly.
nov 04 20:48:12 raspberrypi systemd[1]: Failed to start Companion Service.
nov 04 20:48:12 raspberrypi systemd[1]: alexa.service: Unit entered failed state.
nov 04 20:48:12 raspberrypi systemd[1]: alexa.service: Failed with result 'exit-code'.
nov 04 20:52:03 raspberrypi systemd[1]: [/lib/systemd/system/alexa.service:9] Failed to parse output specifier, ignoring: tty,
nov 04 20:52:15 raspberrypi systemd[1]: [/lib/systemd/system/alexa.service:9] Failed to parse output specifier, ignoring: tty,

sudo systemctl status gassistpi-ok-google.service

● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2018-11-04 20:48:12 CET; 12min ago
 Main PID: 497 (code=exited, status=217/USER)

nov 04 20:48:12 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=217/USER
nov 04 20:48:12 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
nov 04 20:48:12 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
nov 04 20:48:12 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
nov 04 20:48:12 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
nov 04 20:48:12 raspberrypi systemd[1]: gassistpi-ok-google.service: Start request repeated too quickly.
nov 04 20:48:12 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.
nov 04 20:48:12 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
nov 04 20:48:12 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.

cnoorman, 2018-11-05:
You have to take out the underscores around `pi` also!
Try this again:
> Replace `__USER__ `by `pi` in
> `/lib/systemd/system/gassistpi-ok-google.service` and
> `/lib/systemd/system/alexa.service`.
> And restart services:
> 
> ```
> sudo systemctl daemon-restart
> sudo systemctl start alexa.service
> sudo systemctl start gassistpi-ok-google.service
> ```
In `/lib/systemd/system/gassistpi-ok-google.service` you have to replace:
`ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --hahahome-93727-gapi-7wiuf2 'saved-model-id'`
by:
`ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id hahahome-93727-gapi-7wiuf2 `

assuming that that is your device_model_id.

closed 2018-11-05 by shivasiddharth

erosalpe02, 2018-11-05:
Now i have less errors but they still didn't start now i send the log

commit 49bdfd51 (2018-11-05): FIx service issues

Fixes #79  #72
PR #84 (2018-11-05): Alexa indicator
Fixes #72  Fixes #79