shivasiddharth/Assistants-Pi #80: Assistants don't start at the boot

erosalpe02, 2018-11-02:
I have installed alexa and google assistant and if i start them manually they work but don't start at boot please help me

closed 2018-11-04 by erosalpe02

virtuary, 2020-01-09:
care to share how you solved this since you opened this and closed this yourself?

shivasiddharth, 2020-01-09:
> care to share how you solved this since you opened this and closed this yourself?

Can you please be a bit more polite here in this forum.

virtuary, 2020-01-09:
Sorry- tone may not be conveyed properly but was not being snarky- just
literal and genuinely serious.
His answer might have helped me and others.  Hope nothing happened to the
lad....
I really like this project Shiva! I think it's amazing! I just wish I
wasn't chasing answers in so many different locations. Lol.