shivasiddharth/Assistants-Pi #85: Assistants don't auto start when i power on my raspberry pi i have a raspberry pi 3

erosalpe02, 2018-11-05:
assistant don't start i wrote here because my issue was closed whitout resolving my problem so

this is  : 

cat /lib/systemd/system/gassistpi-ok-google.service

[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/__pi__/env
Environment=PATH=/home/__pi__/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/__pi__/env/bin/python -u /home/__pi__/GassistPi/src/main.py --device_model_id 'hahahome-93727-gapi-7wiuf2'

WorkingDirectory=/home/__pi__/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=__pi__

[Install]
WantedBy=multi-user.target

cat /lib/systemd/system/alexa.service

[Unit]
Description=Companion Service
After=network.target

[Service]
Environment=PATH=/home/__pi__/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/usr/bin/python -u /home/__pi__/Assistants-Pi/Alexa/alexa.py
WorkingDirectory=/home/__pi__/
StandardOutput=tty,
StandardError=tty
Restart=always
User=__pi__

[Install]
WantedBy=multi-user.target

sudo systemctl status gassistpi-ok-google.service

● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2018-11-05 14:57:32 CET; 12min ago
  Process: 456 ExecStart=/home/__pi__/env/bin/python -u /home/__pi__/GassistPi/src/main.py --device_model_id hahahome-93727-gapi-7wiuf2 (code=exited, status=217/USER)
 Main PID: 456 (code=exited, status=217/USER)

nov 05 14:57:31 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=217/USER
nov 05 14:57:31 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
nov 05 14:57:31 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
nov 05 14:57:32 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
nov 05 14:57:32 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
nov 05 14:57:32 raspberrypi systemd[1]: gassistpi-ok-google.service: Start request repeated too quickly.
nov 05 14:57:32 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.
nov 05 14:57:32 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
nov 05 14:57:32 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.

sudo systemctl status alexa.service

● alexa.service - Companion Service
   Loaded: loaded (/lib/systemd/system/alexa.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2018-11-05 14:57:32 CET; 12min ago
  Process: 458 ExecStart=/usr/bin/python -u /home/__pi__/Assistants-Pi/Alexa/alexa.py (code=exited, status=217/USER)
 Main PID: 458 (code=exited, status=217/USER)

nov 05 14:57:31 raspberrypi systemd[1]: alexa.service: Main process exited, code=exited, status=217/USER
nov 05 14:57:31 raspberrypi systemd[1]: alexa.service: Unit entered failed state.
nov 05 14:57:31 raspberrypi systemd[1]: alexa.service: Failed with result 'exit-code'.
nov 05 14:57:32 raspberrypi systemd[1]: alexa.service: Service hold-off time over, scheduling restart.
nov 05 14:57:32 raspberrypi systemd[1]: Stopped Companion Service.
nov 05 14:57:32 raspberrypi systemd[1]: alexa.service: Start request repeated too quickly.
nov 05 14:57:32 raspberrypi systemd[1]: Failed to start Companion Service.
nov 05 14:57:32 raspberrypi systemd[1]: alexa.service: Unit entered failed state.
nov 05 14:57:32 raspberrypi systemd[1]: alexa.service: Failed with result 'exit-code'.

please help ty

cnoorman, 2018-11-05:
You still have underscores in your service files! 
The fact that they print out in bold is an indication for that.
`__pi__` is displayed as __pi__. 

Use: `sudo nano /lib/systemd/system/gassistpi-ok-google.service`
and
`/lib/systemd/system/alexa.service`
to edit and remove the underscores `__` !

erosalpe02, 2018-11-05:
ty now alexa start but google assistant not start

sudo systemctl status gassistpi-ok-google.service

● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2018-11-05 16:30:31 CET; 1min 57s ago
  Process: 517 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id hahahome-93727-gapi-7wiuf2 (code=exited, status=2)
 Main PID: 517 (code=exited, status=2)

nov 05 16:30:31 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
nov 05 16:30:31 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
nov 05 16:30:31 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
nov 05 16:30:31 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
nov 05 16:30:31 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
nov 05 16:30:31 raspberrypi systemd[1]: gassistpi-ok-google.service: Start request repeated too quickly.
nov 05 16:30:31 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.
nov 05 16:30:31 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
nov 05 16:30:31 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.

cat /lib/systemd/system/gassistpi-ok-google.service

[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'hahahome-93727-gapi-7wiuf2'

WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target

cnoorman, 2018-11-05:
Can you still start the Google Assistant from the command line with:
`/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'hahahome-93727-gapi-7wiuf2'`  ? 
At least you can be sure that the device_model_id is correct then.
The service script looks OK now.
Try:
```
sudo systemctl daemon-reload
sudo systemctl reset-failed
sudo systemctl start gassistpi-ok-google
sudo systemctl status gassistpi-ok-google
```
If GA starts now than:
`sudo systemctl enable gassistpi-ok-google`
and `sudo reboot`

erosalpe02, 2018-11-05:
If i do sudo systemctl start gassistpi-ok-google.service than sudi systemctl enable gassistpi-ok-google.service and sudo systemctl status gassistpi-ok-google.service it say service running but if i redo the last command in another terminal it say service not running

cnoorman, 2018-11-05:
What do you get when you reboot and:
`sudo systemctl status gassistpi-ok-google`  ?

erosalpe02, 2018-11-05:
i get this:

● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2018-11-05 21:24:39 CET; 4min 37s ago
  Process: 608 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --hahahome-93727-gapi-7wiuf2 hahahome-93727-gapi-7wiuf2 (code=exited, status=2)
 Main PID: 608 (code=exited, status=2)

nov 05 21:24:39 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
nov 05 21:24:39 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
nov 05 21:24:39 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
nov 05 21:24:39 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
nov 05 21:24:39 raspberrypi systemd[1]: gassistpi-ok-google.service: Start request repeated too quickly.
nov 05 21:24:39 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.
nov 05 21:24:39 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
nov 05 21:24:39 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.

shivasiddharth, 2018-11-05:
post the output of 
`journalctl -u gassistpi-ok-google.service`

erosalpe02, 2018-11-05:
It say no entries

shivasiddharth, 2018-11-05:
use:
`systemctl restart systemd-journald.service`
and then
`journalctl -u gassistpi-ok-google.service`

erosalpe02, 2018-11-05:
When i do systemctl restart systemd-journald.service than i choose pi the i use the second command it say no entries if i choose root i don't know what is the password

erosalpe02, 2018-11-05:
And if i select root after i set the root pw it say no entries

cnoorman, 2018-11-05:
use `sudo journalctl -u gassistpi-ok-google.service`

erosalpe02, 2018-11-05:
It say no entries

shivasiddharth, 2018-11-06:
manually start the google assistant. If the manual start works the auto start should also work. There must be something wrong going on.

shivasiddharth, 2018-11-06:
https://github.com/shivasiddharth/Assistants-Pi#manually-start-the-google-assistant

erosalpe02, 2018-11-06:
I use this command /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --hahahome-93727-gapi-7wiuf2 
And it say no such file or directory

shivasiddharth, 2018-11-06:
While running the installer, what option did you choose?

erosalpe02, 2018-11-06:
Both

shivasiddharth, 2018-11-06:
`/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --hahahome-93727-gapi-7wiuf2` syntax is wrong. Please the README for correct syntax.

shivasiddharth, 2018-11-10:
The GA not working is due to #87 . I have fixed that in the installer script.

closed 2018-11-10 by shivasiddharth

erosalpe02, 2018-11-10:
So  now what u have to do

shivasiddharth, 2018-11-10:
Check if the GassistPi directory is within the Alexa folder. Most likely it will be there only. Right click on the GassistPi folder choose cut and paste it in the /home/pi/ directory.