shivasiddharth/Assistants-Pi #86: Unable to install assistants installer

MrsCooky, 2018-11-05:
the installation process pause at [ 60% ] Building CXX object CapabilityAgents/Bluetooth/src/CMakeFiles/Bluetooth.dir/BlueTooth.cpp.o.
Im a beginner of Pi, how can I solve this issues, thank you!

bpmittal, 2018-11-18:
> the installation process pause at [ 60% ] Building CXX object CapabilityAgents/Bluetooth/src/CMakeFiles/Bluetooth.dir/BlueTooth.cpp.o.
> Im a beginner of Pi, how can I solve this issues, thank you!

i am also having this same issue, any solution , and y is happending 
, 
i have tried the 8th step many time but my raspberry pi hangs at the same state .

[ 60%] Building CXX object CapabilityAgents/Bluetooth/src/CMakeFiles/Bluetooth.dir/Bluetooth.cpp.o

deniscubic, 2019-01-12:
Have the same issue. Raspi is completly freezing no responding only power off/on is helping after that line.
Whats the reason?

bpmittal, 2019-01-13:
It is a problem related to old raspberry pi , i was using old version

rsanjeev525, 2019-01-21:
sudo /home/pi/Assistants-Pi/Alexa/startsample.sh  
sudo: /home/pi/Assistants-Pi/Alexa/startsample.sh: command not found

I am facing this error