shivasiddharth/Assistants-Pi #94: Google Assistant cannot play music

BaldEagleX02, 2018-12-23:
When I ask Google Assistant to play music it says: "I can't do that here, but you can ask me to do it on another device" or something similar.
How can I fix this? 
I have configured Spotify correctly.

BaldEagleX02, 2018-12-24:
Yeah, you are right. I haven't thought it was considered a customization.

BaldEagleX02, 2018-12-24:
Thank you for your help

BaldEagleX02, 2018-12-24:
It doesn't work. If I say: "Play <my_playlist> from Spotify" it says: "Obtaining music links" and, after a few seconds, it reboots.

BaldEagleX02, 2018-12-24:
Playing from YouTube works very well instead.

shivasiddharth, 2018-12-24:
Have you input the spotify user-id in the config.yaml ?

BaldEagleX02, 2018-12-24:
Yes

BaldEagleX02, 2018-12-24:
I inserted Client ID, Client secret, my username and the Google API key for Youtube