shivasiddharth/Assistants-Pi #98: Alexa fails to install

jmfrancofraiz, 2019-01-08:
I have a Pi 3 Model B, with an AIY-HAT. Fresh install of Raspbian Stretch Lite. The first problem I encoutered said 'pip command not found', so I had to manually install it with 'sudo apt-get install python-pip'. After that, Alexa built OK but failed at tests:

```
The following tests FAILED:
	253 - TimeTest.testStringConversion (Failed)
	256 - TimeTest.testTimeConversion (Failed)
	846 - MediaPlayerTest.testStartPlayWaitForEnd (Failed)
	847 - MediaPlayerTest.testStartPlayForUrl (Failed)
	848 - MediaPlayerTest.testConsecutiveSetSource (Failed)
	849 - MediaPlayerTest.testStartPlayWaitForEndStartPlayAgain (Failed)
	850 - MediaPlayerTest.testStopPlay (Failed)
	851 - MediaPlayerTest.testStartPlayCallAfterStopPlay (Failed)
	852 - MediaPlayerTest.testStartPlayCallAfterStopPlayDifferentSource (Failed)
	853 - MediaPlayerTest.testPauseDuringPlay (Failed)
	854 - MediaPlayerTest.testResumeAfterPauseThenStop (Failed)
	855 - MediaPlayerTest.testStopAfterPause (Failed)
	856 - MediaPlayerTest.testPauseAfterPause (Failed)
	857 - MediaPlayerTest.testResumeAfterPlay (Failed)
	858 - MediaPlayerTest.testGetOffsetInMilliseconds (Failed)
	860 - MediaPlayerTest.testGetOffsetWhenStoppedFails (Failed)
	861 - MediaPlayerTest.testGetOffsetWhenPaused (Failed)
	862 - MediaPlayerTest.testPlayingTwoAttachments (Failed)
	863 - MediaPlayerTest.testUnsteadyReads (Failed)
	864 - MediaPlayerTest.testRecoveryFromPausedReads (Failed)
	865 - MediaPlayerTest.testStartPlayWithUrlPlaylistWaitForEnd (Failed)
	866 - MediaPlayerTest.testSetOffsetSeekableSource (Failed)
	869 - MediaPlayerTest.testRepeatAttachment (Failed)
	870 - MediaPlayerTest.testSetVolumePlays (Failed)
	871 - MediaPlayerTest.testAdjustVolumePlaysDuringPlay (Failed)
	872 - MediaPlayerTest.testAdjustVolumePlays (Failed)
	873 - MediaPlayerTest.testAdjustVolumeOutOfBounds (Failed)
	874 - MediaPlayerTest.testSetMutePlays (Failed)
	877 - MediaPlayerTest.testReadTags (Failed)
	878 - MediaPlayerTest.testConsecutiveSameApiCalls (Failed)
	879 - MediaPlayerTest.testImmediatePause (Failed)
	880 - MediaPlayerTest.multiplePlayAndSetSource (Failed)
	882 - MediaPlayerTest.doublePause (Failed)
	883 - MediaPlayerTest.resumeWhenPlaying (Failed)
	884 - MediaPlayerTest.resumeWhenStopped (Failed)
	885 - MediaPlayerTest.newSetSourceLeadsToStoppedCallback (Failed)
	886 - MediaPlayerTest.resumeAfterPauseWithPendingPlay (Failed)
Errors while running CTest
Makefile:116: fallo en las instrucciones para el objetivo 'test'
make: *** [test] Error 8
```

shivasiddharth, 2019-01-09:
Are you using Raspbian Lite version ? If yes, then please switch to Raspbian desktop.

jatt01ankit, 2019-05-21:
Please help me with these errors.......
The following tests FAILED:
        846 - MediaPlayerTest.testStartPlayWaitForEnd (Failed)
        847 - MediaPlayerTest.testStartPlayForUrl (Failed)
        848 - MediaPlayerTest.testConsecutiveSetSource (Failed)
        849 - MediaPlayerTest.testStartPlayWaitForEndStartPlayAgain (Failed)
        854 - MediaPlayerTest.testResumeAfterPauseThenStop (Failed)
        862 - MediaPlayerTest.testPlayingTwoAttachments (Failed)
        863 - MediaPlayerTest.testUnsteadyReads (Failed)
        864 - MediaPlayerTest.testRecoveryFromPausedReads (Failed)
        865 - MediaPlayerTest.testStartPlayWithUrlPlaylistWaitForEnd (Failed)
        866 - MediaPlayerTest.testSetOffsetSeekableSource (Failed)
        869 - MediaPlayerTest.testRepeatAttachment (Failed)
        870 - MediaPlayerTest.testSetVolumePlays (Failed)
        871 - MediaPlayerTest.testAdjustVolumePlaysDuringPlay (Failed)
        872 - MediaPlayerTest.testAdjustVolumePlays (Failed)
        873 - MediaPlayerTest.testAdjustVolumeOutOfBounds (Failed)
        874 - MediaPlayerTest.testSetMutePlays (Failed)
        877 - MediaPlayerTest.testReadTags (Failed)
        879 - MediaPlayerTest.testImmediatePause (Failed)
        880 - MediaPlayerTest.multiplePlayAndSetSource (Failed)
        886 - MediaPlayerTest.resumeAfterPauseWithPendingPlay (Failed)
Errors while running CTest
Makefile:116: recipe for target 'test' failed
make: *** [test] Error 8
thank you so much in advance.....

jatt01ankit, 2019-05-21:
> Please help me with these errors.......
> The following tests FAILED:
> 846 - MediaPlayerTest.testStartPlayWaitForEnd (Failed)
> 847 - MediaPlayerTest.testStartPlayForUrl (Failed)
> 848 - [...]

I'm using Raspbean desktop stretch

PR #136 (2020-03-13): Update SDK to 1.80.0
Fixes #63  Fixes #83 Fixes #90 Fixes #96 Fixes #98  Fixes #99  Fixes #122  Fixes #123  Fixes #124 Fixes #125 Fixes #126  Fixes #130 Fixes #131
closed 2020-03-13 by shivasiddharth