shivasiddharth/Assistants-Pi #99: Alexa.service still overloading CPU

rvsphinx, 2019-01-24:
I edited Alexa.service just as @cnoorman suggested.  But my CPU is stilling being overloaded. I have an RPI 3 B and just enable Alexa not both services for now.  If I run startsample.sh run with minimal load on CPU. when i start service runs from 80% and up till completely freezes most likely because of temp.

shivasiddharth, 2019-01-24:
There is nothing called startsample in this project. Looks like you are on an old and outdated version.

rvsphinx, 2019-01-25:
Sorry but I did grab the latest file.  I also backed it up created a new file with the syntax in the alexa.service file and once I sudo systemctl start alexa.service cpu from 7% jumps to 75 and 85% usage. once I stop service everything drops.  
Is there any thing else you think I can do.

shivasiddharth, 2019-01-25:
What board are you using?

vijayrraut, 2019-01-26:
I also did the changes as per William's comment, but still the cpu goes to 85%, Please help

shivasiddharth, 2019-01-26:
try the updated alexa.py file

shivasiddharth, 2019-01-27:
For time being in the alexa service file change 

`ExecStart=/usr/bin/python -u /home/__USER__/Assistants-Pi/Alexa/alexa.py`

to 

`ExecStart=/bin/bash /home/__USER__/Assistants-Pi/Alexa/startsample.sh`

Then run:
```
sudo systemctl stop alexa.service
sudo systemctl disable alexa.service
```

Then go through the service installer process again.

rvsphinx, 2019-01-30:
Hello, I tried go back to something simple that @HardTYZ commented.  Adding screen -dmS -D and add RemainAfterExit=yes to the alexa.service file and that worked!  From 85% down to 8 a 9% .  I recommend this for anyone else.  Just install screen sudo apt-get install screen then put service file like this:

[Unit]
Description=Companion Service
After=network.target

[Service]
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/usr/bin/screen -dmS -D bash -c '/usr/bin/python -u /PATH/TO/YOUR/SCRIPT.py'
WorkingDirectory=/home/pi/
StandardOutput=tty,
StandardError=tty
Restart=always
User=pi
RemainAfterExit=yes

[Install]
WantedBy=multi-user.target

grvweb, 2019-02-03:
> Hello, I tried go back to something simple that @HardTYZ commented. Adding screen -dmS -D and add RemainAfterExit=yes to the alexa.service file and that worked! From 85% down to 8 a 9% . I recommend [...]

I tried this but, it keeps restarting.
RAM Size increasing from 100 mb to 900 mb and processor jumps from 10% to 80%, After sometime pi stops responding.

> For time being in the alexa service file change
> 
> `ExecStart=/usr/bin/python -u /home/__USER__/Assistants-Pi/Alexa/alexa.py`
> 
> to
> 
> `ExecStart=/bin/bash /home/__USER__/Assistants-Pi/Alexa/startsample.sh`
> 
> Then run:
> 
> ```
> sudo systemctl stop alexa.service
> sudo systemctl disable alexa.service
> ```
> Then go through the service installer process again. This will start the startsample as service, but you will not have any indicators or audio tones.

Didn't found useful, I have attached a screenshot. 
[image]

shivasiddharth, 2019-02-03:
> I tried this no effect on ram and cpu but, service keeps restarting every 2 mins. 

If you have not installed screen, then obviously it will restart.

williamhemmingsen, 2019-02-04:
shivasiddharth, see my new pull request.  There was an error in alexa.service.  It now does not cause any high cpu.

Regards,

shivasiddharth, 2019-02-04:
@williamhemmingsen , can you raise a PR?

grvweb, 2019-02-04:
> [Unit]
> Description=Companion Service
> After=network.target
> 
> [Service]
> Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
> ExecStart=/usr/bin/python -u /home/pi/Assistants-Pi/alexa.py
> WorkingDirectory=/home/pi/
> TTYPath=/dev/tty12
> StandardInput=tty
> StandardOutput=tty
> StandardError=null
> Restart=always
> User=pi
> 
> [Install]
> WantedBy=multi-user.target
> 
> Screen works, but it mask systemd from restarting on failure. Try the above, it works perfectly for me.
> 
> William

I ran it by changing line no: 7 from /home/pi/Assistants-Pi/alexa.py to /home/pi/Assistants-Pi/Alexa/alexa.py
Still Same issue..! 
Should i change alexa.py from Assistants-Pi/Alexa/ to Assistants-Pi/ directory ?

williamhemmingsen, 2019-02-05:
> I ran it by changing line no: 7 from /home/pi/Assistants-Pi/alexa.py to /home/pi/Assistants-Pi/Alexa/alexa.py
> Still Same issue..!
> Should i change alexa.py from Assistants-Pi/Alexa/ to Assistants-Pi/ [...]

/home/pi/Assistants-Pi/Alexa/alexa.py is correct.  Above is a typo.  The code is correct in github.  

But changing the script is not enough.  you MUST update systemd.  

steps--

pull down new copy of alexa.service from git and overwrite your current alexa.service. then run the following commands to update systemd.

sudo systemctl stop alexa.service
sudo systemctl disable alexa.service
sudo /home/pi/Assistants-Pi/scripts/installer.sh  
sudo systemctl enable alexa.service
sudo systemctl start alexa.service

williamhemmingsen, 2019-02-05:
Also just so you know the high CPU is caused by stdin/stdout/sterr.

grvweb, 2019-02-05:
> Also just so you know the high CPU is caused by stdin/stdout/sterr.

True, it is caused by stdin/stdout. Issue is not related to service file.
I tried by running it in background via crontab, init.d and rc.local.
I also tried running this command in crontab `@reboot /home/pi/Assistants-Pi/Alexa/build/SampleApp/src/SampleApp '/home/pi/Assistants-Pi/Alexa/build/Integration/AlexaClientSDKConfig.json' '/home/pi/Assistants-Pi/Alexa/third-party/alexa-rpi/models' > /dev/null 2>&1 &` I got same response, nothing really worked.
I think maybe this issue is not related to this repo.

shivasiddharth, 2019-02-05:
Is the sampleapp also driving the CPU crazy ?

grvweb, 2019-02-05:
Yes

shivasiddharth, 2019-02-05:
Check issue #539 in the official avs-device-sdk we are not alone on this. Amazon themselves don't recommend running Sampleapp as a service. They have designed Alexa to work with stdin causing this issue.

grvweb, 2019-02-06:
I fixed it.
Open develop branch on your amazon voice services repo, I will make a pr.
😁🎉🎊

shivasiddharth, 2019-02-06:
Develop branch holds good only for the google assistant, where constantly changes are happening. raise a PR to the master for Alexa.

grvweb, 2019-02-06:
Ok.. I Will.

grvweb, 2019-02-07:
Issue fixed on Raspberry Pi, Tested with both fresh and old installations.

For old installation delete all folders in /home/pi/Assistants-Pi/Alexa **(Only folders, not files).**

Move to Installation directory

`cd /home/pi/Assistants-Pi`

Pull Changes

`git pull origin master`

Run Installer

`sudo /home/pi/Assistants-Pi/scripts/installer.sh`

Some tests may fail ignore them.

Then, you need to update systemd.service file manually:

`sudo nano /lib/systemd/system/alexa.service`

Enter the following inside alexa.service file:

```
[Unit]
Description=Amazon Alexa
After=network.target network-online.target

[Service]
Type=forking
ExecStart=/home/pi/Assistants-Pi/Alexa/start.sh

[Install]
WantedBy=multi-user.target
```

```
sudo systemctl stop /lib/systemd/system/alexa.service
sudo systemctl daemon-reload
sudo systemctl enable /lib/systemd/system/alexa.service
sudo systemctl start /lib/systemd/system/alexa.service
```
Should work fine. Only 5% to 10% CPU should be use by Alexa.

vijayrraut, 2019-02-07:
Hello Gowthamraj,
I tried git pull, but while enabling the service its failing with below error, please help.

pi@raspberrypi:~ $ sudo systemctl daemon-reload
pi@raspberrypi:~ $ sudo systemctl enable /lib/systemd/system/alexa.service
Created symlink /etc/systemd/system/multi-user.target.wants/alexa.service -> /lib/systemd/system/alexa.service.
pi@raspberrypi:~ $ sudo systemctl start /lib/systemd/system/alexa.service
Failed to start lib-systemd-system-alexa.service.mount: Unit lib-systemd-system-alexa.service.mount not found.

grvweb, 2019-02-07:
> Hello Gowthamraj,
> I tried git pull, but while enabling the service its failing with below error, please help.
>
> pi@raspberrypi:~ $ sudo systemctl daemon-reload
> pi@raspberrypi:~ $ sudo systemctl [...]
Try
`sudo systemctl start alexa.service`

PR #136 (2020-03-13): Update SDK to 1.80.0
Fixes #63  Fixes #83 Fixes #90 Fixes #96 Fixes #98  Fixes #99  Fixes #122  Fixes #123  Fixes #124 Fixes #125 Fixes #126  Fixes #130 Fixes #131
closed 2020-03-13 by shivasiddharth