shivasiddharth/Assistants-Pi #108: ${USER}=root not pi

iwwork, 2019-02-12:
Every time, the response of ${USER} is root, not pi.
How to make the setting rightly?

Thanks

grvweb, 2019-02-15:
> Every time, the response of ${USER} is root, not pi.
> How to make the setting correctly?
> 
> Thanks

Please post complete issue.
provide some Screenshots to understand.

iwwork, 2019-02-16:
Assistants-Pi/installer.sh at master · shivasiddharth/Assistants-Pi
https://github.com/shivasiddharth/Assistants-Pi/blob/master/scripts/installer.sh

line 128: sudo chmod +x /home/${USER}/GassistPi/scripts/gassist-installer.sh

it is /home/root/ not /home/pi/

shivasiddharth, 2019-02-16:
Still i dont get your issue..

grvweb, 2019-02-16:
> Assistants-Pi/installer.sh at master · shivasiddharth/Assistants-Pi
> https://github.com/shivasiddharth/Assistants-Pi/blob/master/scripts/installer.sh
> 
> line 128: sudo chmod +x /home/${USER}/GassistPi/scripts/gassist-installer.sh
> 
> it is /home/root/ not /home/pi/

Settings are correct, you dont need to do anything. 
data willl be stored to /home/pi/.
User will be pi and not root.

iwwork, 2019-02-18:
@grvweb You got what I want to say. But the terminal shows /home/root/... and could not install

kmpathak1610, 2019-03-26:
Same problem I am also facing.