shivasiddharth/Assistants-Pi #109: in a pickle even after authorizing alexa

mananhiteshkataria, 2019-02-13:
# UNRECOVERABLE AUTHORIZATION ERROR: INVALID_REQUEST
# Entering limited interaction mode.

##################################################################

+----------------------------------------------------------------------------+
| In Limited Mode: |
+----------------------------------------------------------------------------+
| Status : Unrecoverable authorization failure. |
+----------------------------------------------------------------------------+
| Info: |
| Press 'i' followed by Enter at any time to see the help screen. |
| Stop an interaction: |
| Press 's' and Enter to stop an ongoing interaction. |
| Privacy mode (microphone off): |
| Press 'm' and Enter to turn on and off the microphone. |
| Speaker Control: |
| Press 'p' followed by Enter at any time to adjust speaker settings. |
| Reset device: |
| Press 'k' followed by Enter at any time to reset your device. This |
| will erase any data stored in the device and you will have to |
| re-register your device. |
| This option will also exit the application. |
| Quit: |
| Press 'q' followed by Enter at any time to quit the application. |
+----------------------------------------------------------------------------+

mananhiteshkataria, 2019-02-13:
this is the mode i am stuck in even after registering device

shivasiddharth, 2019-02-13:
Duplicate of #89

closed 2019-02-13 by shivasiddharth