shivasiddharth/Assistants-Pi #110: not able to get google assistant to start on boot

mananhiteshkataria, 2019-02-13:
made necessary changes to exec start stamement replacing model no removing underscore and all but still cannot start google assistant after boot 

alexa works fine

mananhiteshkataria, 2019-02-14:
https://drive.google.com/file/d/1FbVhwkPA0LgHzk3SfvNNZJ2TKaZq7zSd/view?usp=drivesdk

link to the file

grvweb, 2019-02-15:
Try this 
`ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id mananspeaker-mananspeaker-jjedav`
Instead of this 
`ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'mananspeaker-mananspeaker-jjedav'`

mananhiteshkataria, 2019-02-16:
● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2019-02-16 09:52:48 IST; 54s ago
  Process: 493 ExecStart=/home/__USER__/env/bin/python -u /home/__USER__/GassistPi/src/main.py --device_model_id mananspeaker-mananspeaker-jjedav (code=exited, status=203/EXEC)
 Main PID: 493 (code=exited, status=203/EXEC)

Feb 16 09:52:48 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Feb 16 09:52:48 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Feb 16 09:52:48 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Feb 16 09:52:48 raspberrypi systemd[1]: gassistpi-ok-google.service: Start request repeated too quickly.
Feb 16 09:52:48 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.
Feb 16 09:52:48 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Feb 16 09:52:48 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.

mananhiteshkataria, 2019-02-16:
done as said still get this issue

after the edits and reenabling it

mananhiteshkataria, 2019-02-16:
it is bolding at user which i have mentioned as pi as i kept everything default for simplicity and still face this issue

mananhiteshkataria, 2019-02-16:
[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'mananspeaker-mananspeaker-jjedav'

WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target

mananhiteshkataria, 2019-02-16:
dropped gatspi if anyone else has suggestions running out of ideas tbh

grvweb, 2019-02-16:
Drop output for 
`sudo cat /lib/systemd/system/gassistpi-ok-google.service`

mananhiteshkataria, 2019-02-16:
pi@raspberrypi:~ $ sudo cat /lib/systemd/system/gassistpi-ok-google.service
[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/__USER__/env/bin/python -u /home/__USER__/GassistPi/src/main.py --device_model_id 'mananspeaker-mananspeaker-jjedav'

WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target

mananhiteshkataria, 2019-02-16:
> Drop output for
> `sudo cat /lib/systemd/system/gassistpi-ok-google.service`

pi@raspberrypi:~ $ sudo cat /lib/systemd/system/gassistpi-ok-google.service
[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/__USER__/env/bin/python -u /home/__USER__/GassistPi/src/main.py --device_model_id 'mananspeaker-mananspeaker-jjedav'

WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target

shivasiddharth, 2019-02-16:
First manually start the assistant and see if it's starting normally. 90% of the time auto start issues are due to the assistant and not the service file. I have given instructions in the README to manually start the Google assistant.

mananhiteshkataria, 2019-02-16:
sir if i made this thread it was pretty obvious that manual is working and auto is not working as i stated that in the title

mananhiteshkataria, 2019-02-16:
so if its an assistant issue cant be helped i guess

shivasiddharth, 2019-02-16:
You have given me no time to respond at all.....

1. You are **not supposed to remove the underscore**. In no place, I have asked you to remove that underscore. Only instructions are to add the model id. I dont understand why you made that change in first place. This issue would not even have occurred if you did not remove that underscore.
2. Any time you make changes to the systemd file, you need to run the service installer again.

Open a terminal and run:
```
sudo systemctl stop gassistpi-ok-google.service
sudo systemctl disable gassistpi-ok-google.service
sudo /home/pi/Assistants-Pi/scripts/service-installer.sh  
sudo systemctl enable gassistpi-ok-google.service
```

grvweb, 2019-02-16:
> pi@raspberrypi:~ $ sudo cat /lib/systemd/system/gassistpi-ok-google.service
> [Unit]
> Description=Google Assistant-Voice
> Wants=network-online.target
> After=network-online.target
>
> [Service] [...]

Your are facing file not found error, do this .

```
sudo systemctl stop gassistpi-ok-google.service
sudo systemctl disable gassistpi-ok-google.service
```

**Replace this**
`ExecStart=/home/**USER**/env/bin/python -u /home/**USER**/GassistPi/src/main.py --device_model_id 'mananspeaker-mananspeaker-jjedav'`

**With**
`ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id mananspeaker-mananspeaker-jjedav`

and

```
sudo systemctl enable gassistpi-ok-google.service
sudo systemctl start gassistpi-ok-google.service
```

shivasiddharth, 2019-02-16:
@Gowtham he is making your changes in the service file located in project directory and not in the systemd directory. That's why it did not work.

grvweb, 2019-02-18:
> @gowtham he is making your changes in the service file located in project directory and not in the systemd directory. That's why it did not work.
@shivasiddharth 
yes.

closed 2019-02-18 by mananhiteshkataria