shivasiddharth/Assistants-Pi #114: Alexa isn't speaking german

jsloot, 2019-03-10:
What do I have to do to make alexa speak German?

pimseb, 2019-03-19:
I'd like also to know how to change language.
If I use startsample.sh I see I should type c to access the settings and change language. But no key is working.
Is there a file where I can change the language ? I tried to modify AlexaClientSDKConfig.json with no luck

pimseb, 2019-03-22:
I answer to my own question, I managed to run it in french but had to start with a fresh and new install.
I modified the file Assistants-Pi/Alexa/setup.sh and changed the line 33 from LOCALE=${LOCALE:-'en-US'} to LOCALE=${LOCALE:-'fr-FR'}
and compiled it once again

FiveKnive, 2019-05-20:
Does this solution work if I want to install both assistants or do I need to modify a different file?