shivasiddharth/Assistants-Pi #120: Installer script issue

kumarullal, 2019-06-28:
It only asks for Alexa Client ID and product ID. Does not ask client secret.
What ami I doing wrong?

kumarullal, 2019-06-30:
Also,
sudo /home/pi/Assistants-Pi/Alexa/startsample.sh  
startsample.sh  does not exist. on pi.sh, start.sh and setup.sh is included.
Has the instruction code changed?

syedsulaiman07, 2019-12-24:
same with me
there is some problem the project is changed

virtuary, 2020-01-07:
....and.....we continue to wait.....
so- i've run into similar issues;  either DIRs don't exist, of FILENAMES are different in the setup/installer than in the payload.  Another issue installing a different way (like either one at a time, or in a different OS version of raspbian, like buster, stretch lite, and original stretch version), there's a point is needs to install something in path %home/%root- and it throws error 'no such dir' (root).
If everything mostly goes in  the /home/pi/ DIR (and the appropriate env and bin folders) why and what is it trying to install in ROOT?
I also get the "startsample.sh does not exist" error depending on the order I do things. And can also confirm the package/setup/installer being the following "start.sh and setup.sh is included".  what is the deal?  Seems like the original GIT project is broken/outdated/abandoned, and the solutions are scattered between 3 other GIT projects- which I've tried using in 1 of my 5 altered attempts already, to piece together a decent smooth install.  Gassist, AlexaAVS, Asssistants-Pi......can somebody please simplify this and compile ONE working install for Raspbian, please?  I don't care if it's and older version like stretch or jessie- just.....something that works and not frankensteined :)
Gonna try one more time and attempts before marking this as a lost cause.
Thanks.
OH and PS- a clear sentence of what the 'json' file name syntax that we are supposed to place in /pi DIR would be grrrrrrreat.
cause you say 'rename it in the video, then DO NOT rename it in the abandoned GIT project instructions, and there are 3 ways it can be named - 
original sysntax - 'assistant.json'
then, client_id/json
then, client)secret_23891268longAssHexablabber78934.
What syntax is it looking for?  i've tried a deep dive looking at your code and cannot see or understand where/when it's querying the file, in what DIR, and what syntax is the correct one.
Good work, I'm not being sarcastic.  However the upkeep.....yeah, not so much.
I'd gladly help put together ONE working compiled project if it was clear what pieces of all these 3 projects and different instructions to go by :)

shivasiddharth, 2020-01-08:
@virtuary, I have never mentioned startsample.sh in the git readme instructions. Have I ? This project still works with raspbian stretch desktop version. This is not a project but a bunch of scripts to "aid" the users to install the audio drivers, indicators etc for both the assistants.

virtuary, 2020-01-08:
@shivasiddharth Well, first of all- Thanks for your quick reply!  A lot of contributors don't even respond to old projects so that's very good you did, Shiva!
I understand what you say, but there are just several other little issues that the results are the same and it just doesn't work for some reason.  
I am doing a fresh new install, documenting all the steps and outputs, and putting together a N++ doc with the specs and devices used, to hopefully have one smooth instructions for myself- if it works this time haha- and if not, so you (if you have time) take a look at it and see if there's anything you spot that could be the reason.  I know some people have success, I just don't know why some of us don't.
Here's hoping for one final good run and that it goes smoothly!

Thanks, Shiva- I will def share my results.

virtuary, 2020-01-09:
RESULTS- meh.  Google works with hotword after the install- but doing the change to get it to auto-start on boot (replacing model id)  does nothing because it doesn't work.  Only way to get GA to run again was doing the entire ga and service installer scripts.  Trying to 'start' or 'restart' the service throws errors like if it is not even installed or running.
I guess there is no point to my previous comments because I did not put it in context.  
Allow me if i may, but I originally stumbled upon this on THIS site:
(https://www.hackster.io/shiva-siddharth/simultaneously-run-alexa-and-google-assistant-on-pi-12e8df)
From that jumpoff point is how I ended up at the first attempts being duds.
Had it been through github _first_, that might have been different.  So from that article and your video, to then using the links through the terminal scripts to finding this on Ghub, to then finding out the project migrated to something else, then referencing the Gassist project for the 1st couple of steps in the Assistants-Pi project......all after already another fresh image ruined, and flashing SD cards- this is how I have come to the point of being completely lost.
Now I have started with the Github instructions- and have been searching through the Issues (open/closed) as you can tell by now....my 'point' is that in order to get it running as it should, the pieces of information necessary for the correct DIRS, paths, filenames, syntax, order of steps, etc. are not all in 1 place.  That was it.  I'm just super OCD! lol Forgive me for that, I know it can come across as impolite- however the frustration is still real.  Think of it like this- I have spent almost 2 entire days messing with this and sort of at this point working for you lol.
-Peace bro-

shivasiddharth, 2020-03-13:
Fixed with the last update

closed 2020-03-13 by shivasiddharth