shivasiddharth/Assistants-Pi #126: Failed to create sample application

Friday8229, 2019-07-26:
Hi I installed alexa on a fresh resbian stretch by chosinv the option 2 all of the things went well until I got this error 

pi@raspberrypi:~ $ sudo /home/pi/Assistants-Pi/Alexa/startsample.sh  
2019-07-26 18:13:45.305 [  1] I sdkVersion: 1.9.0
configFile /home/pi/Assistants-Pi/Alexa/build/Integration/AlexaClientSDKConfig.json
Running app with log level: INFO
#################################################
#       NOTIFICATION INDICATOR STATE: OFF       #
#################################################

Expression 'alsa_snd_config_update()' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1353
Expression 'BuildDeviceList( alsaHostApi )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 766
2019-07-26 18:13:45.581 [  1] C PortAudioMicrophoneWrapper:Failed to initialize PortAudio:errorCode=-9999
2019-07-26 18:13:45.581 [  1] C PortAudioMicrophoneWrapper:Failed to initialize PortAudioMicrophoneWrapper
2019-07-26 18:13:45.581 [  1] C SampleApplication:Failed to create PortAudioMicrophoneWrapper!
2019-07-26 18:13:46.327 [  1] C SampleApplication:Failed to initialize SampleApplication
2019-07-26 18:13:46.408 [  2] E HttpPost:doPostFailed:reason=curl_easy_performFailed,result=35,error=SSL connect error
2019-07-26 18:13:46.410 [  2] E CBLAuthDelegate:parseLWAResponseFailed:reason=parseJsonFailed,position=0,error=The document is empty.
Failed to create to SampleApplication!

And i think it is related to the microphone or the sound.

Output of .asoudrc

pcm.!default {
             type hw
             card -1
             device 0
        }
}

Output of asound.conf

pcm.!default {
             type hw
             card -1
             device 0
        }
}

Friday8229, 2019-07-26:
I tried installing portaudio but nothing helped.

HomeLuch, 2019-10-31:
Hi, 
After looking for so long today, I was able to resolve this issue in my device by modifying **Startsample.sh**. 
Added `env PA_ALSA_PLUGHW=1 `in front of the  ` ./SampleApp.`
thus,
`env PA_ALSA_PLUGHW=1 ./SampleApp "/home/pi/Assistants-Pi/Alexa/build/Integration/AlexaClientSDKConfig.json" "/home/pi/Assistants-Pi/Alexa/third-party/alexa-rpi/models" INFO
`

Found it here:
[https://github.com/alexa/avs-device-sdk/issues/1413](url)

PR #136 (2020-03-13): Update SDK to 1.80.0
Fixes #63  Fixes #83 Fixes #90 Fixes #96 Fixes #98  Fixes #99  Fixes #122  Fixes #123  Fixes #124 Fixes #125 Fixes #126  Fixes #130 Fixes #131
closed 2020-03-13 by shivasiddharth