shivasiddharth/Assistants-Pi #129: Ok.  Can't move past Step 9. The next steps don't work....

sdabiker, 2019-11-12:
I have tried this several times. Google assistant works great, but once it authorize and starts its test, nothing happens. I try a second terminal to do the next steps. "sudo chmod +x /home/pi/Assistants-Pi/scripts/service-installer.sh" and beyond and nothing happens. I can't authorize Alexa either. Which is why I wanted to try this project to begin with... Can you help?

virtuary, 2020-01-08:
Same here- what error were you getting.  I've tried following 3 different set of instructions in 3 separate attempts.  
The last error I get when 'authorizing' alexa in that step is:
pi@raspberrypi:~ $ sudo systemctl enable alexa.service
Failed to enable unit: File alexa.service: No such file or directory

That and when I actually say "Alexa" it starts running this (and I know for a FACT that I have network connectivity).  This is all after I actually got a chance to run the auth of this step, go to amazon/com and input the code it output in the terminal script instructions, and register successfully.

2020-01-08 22:28:36.293 [  3] I DefaultClient:notifyOfWakeWord:action=ignoreAlexaWakeWord,reason=networkDisconnected
2020-01-08 22:31:47.563 [  3] I DefaultClient:notifyOfWakeWord:action=ignoreAlexaWakeWord,reason=networkDisconnected

shivasiddharth, 2020-01-08:
Without running service installer if you give systemd commands it will return with service not found only. It needs to be done only after authorizing Alexa. I have mentioned that in the readme clewrly. If Alexa is not authorized do not give the systemd commands.

virtuary, 2020-01-09:
Issue#89 did provide better detail- and the client id entered was not from 'otehr devices and platforms' so this one was on me.  Not sure about the other dude, though.
thanks for answering.