shivasiddharth/Assistants-Pi #133: Alexa and Google both answering my question

MrPestle, 2020-01-28:
If I ask Google a question it gives the answer.
If I ask Alexa a question she answers but at almost the same time Google is also answering the question??? 
Also how can I set the google voice to be male all the time. I can ask it to change but on reboot it's back to female voice?
Any help gratefully received
Thanks for this great piece of work - loving it.

Arunkumar2712, 2020-02-19:
Pls tell what is you used in raspberry pi noods or raspbian os pls tell me I finish till last step the sample is not running in Google assistant so Iam asking

MrPestle, 2020-02-19:
Running latest version of raspbian stretch. Solved my problem I think. I
was running two Google services, the single project and the combined
project.

shivasiddharth, 2020-03-12:
closing this

closed 2020-03-12 by shivasiddharth