shivasiddharth/Assistants-Pi #135: Bad Setting,google-assistant.service: Service has no ExecStart=,

xxue12, 2020-03-12:
Hi,when I did the last step tried to enable google assistant,and I checked the status of google-assistant.service,I got error.

shivasiddharth, 2020-03-12:
Post the error

xxue12, 2020-03-12:
I'll post that soon

xxue12, 2020-03-12:
alexa.service - Amazon Alexa
   Loaded: loaded (/lib/systemd/system/alexa.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2020-03-11 21:51:21 EDT; 34min ago

Mar 11 21:51:21 raspberrypi sudo[565]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/
Mar 11 21:51:21 raspberrypi sudo[565]: pam_unix(sudo:session): session opened for user root by (u
Mar 11 21:51:21 raspberrypi sudo[565]: pam_unix(sudo:session): session closed for user root
Mar 11 21:51:21 raspberrypi sudo[581]:     root : TTY=unknown ; PWD=/home/pi/Assistants-Pi/Alexa 
Mar 11 21:51:21 raspberrypi sudo[581]: pam_unix(sudo:session): session opened for user root by (u
Mar 11 21:51:21 raspberrypi sudo[581]: pam_unix(sudo:session): session closed for user root
Mar 11 21:51:21 raspberrypi start.sh[458]: Must be connected to a terminal.
Mar 11 21:51:21 raspberrypi systemd[1]: alexa.service: Control process exited, code=exited, statu
Mar 11 21:51:21 raspberrypi systemd[1]: alexa.service: Failed with result 'exit-code'.
Mar 11 21:51:21 raspberrypi systemd[1]: Failed to start Amazon Alexa.

xxue12, 2020-03-12:
btw,when I was installing the alexa,I got error said that startsample.sh command could not found

xxue12, 2020-03-12:
I dont change anything,all file are default

shivasiddharth, 2020-03-12:
Its difficult to says whats gone wrong without actually seeing the error. I need to look at the terminal contents etc.

xxue12, 2020-03-12:
what do u wanna look at

xxue12, 2020-03-12:
How can i assist you

shivasiddharth, 2020-03-17:
Can you manually run the startsample.sh and post the terminal contents ? Have you authorized the alexa ?