shivasiddharth/Assistants-Pi #141: /home/pi/env/bin/python: No such file or directory

Zenfather, 2020-04-10:
Installed from this Assistants-Pi and went to start Google assistant manually and got the No such file or directory error

/home/pi/env/bin/python -u /home/pi/Assistants-Pi/Google-Assistant/src/main.py --project_id 'Project'--device_model_id 'Model'                   -bash:

sagarumathe, 2020-04-16:
First go to that directry and manually check if main.py exists there! If it does then make sure while entering this command you put space before and after --project_id and --device_model_id

marine1988, 2020-04-20:
im getting same error in orange pi zero :(

shivasiddharth, 2020-04-20:
If you are using some other username instead of "pi" you will get the error.  Change your username in /home/pi/env/bin/python from pi to your actual username.