shivasiddharth/Assistants-Pi #144: Which version of PI OS

MrPestle, 2020-06-01:
I as recall you used to say that we needed Stretch to run this project. I cannot see any advice on this now - will the project work with the latest Buster?

Thank you

sagarumathe, 2020-06-01:
Yes, it does work on latest buster! Go ahead and try it out!

ShaGGyUK, 2020-06-17:
Do these instructions only work on the full installed versions of the OS's.

Ive tried Lite and Desktop (not additional software) versions but sound test do not work.

Standard asound.conf works for playback (3.5 jack) but if you run system-prep.sh it replaces the asoundrc and asond.conf files with the ones from USB-MIC-JACK which stops sound working.

testing on PI3b

shivasiddharth, 2020-06-18:
You need to check the card ids and make suitable changes to the audio conf files.

ShaGGyUK, 2020-06-18:
> 
> 
> You need to check the card ids and make suitable changes to the audio conf files.

I have,
result = playback is fine but recording does not work through script?
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 2: CameraB409241 [USB Camera-B4.09.24.1], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

config = 
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:2,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:1,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

if i try
pi@raspberrypi:~ $ arecord -d 5 -c 1 -r 16000 test-mic.wav
Recording WAVE 'test-mic.wav' : Unsigned 8 bit, Rate 16000 Hz, Mono
it records the file

but
pi@raspberrypi:~ $ aplay test-mic.wav
Playing WAVE 'test-mic.wav' : Unsigned 8 bit, Rate 16000 Hz, Mono
there is no output

if i run sudo
pi@raspberrypi:~ $ sudo aplay test-mic.wav
Playing WAVE 'test-mic.wav' : Unsigned 8 bit, Rate 16000 Hz, Mono
it plays?

I am running 
pi@raspberrypi:~ $ sudo /home/pi/Assistants-Pi/scripts/audio-test.sh for the script?

MrPestle, 2020-09-26:
As I opened this query and I am happy with the answer I am closing it.

closed 2020-09-26 by MrPestle