shivasiddharth/Assistants-Pi #151: Alexa Installation Freeze at 92%, How to fix it?

iamsujay, 2020-11-15:
I have force shut down the Pi after it stops responding. Then I wait 10-20 seconds and again power-up my Pi and use the same command `sudo /home/pi/Assistants-Pi/scripts/installer.sh` and Select > **2 for Alexa installation**.

after that, I getting the below logs in my terminal. And it is still stuck in 95%. I have shut down and started the Pi 5 times and repeat the process but nothing seemed to works. Please give me a solution.

[ 32%] Built target AVSCommon
[ 37%] Built target RegistrationManager
[ 37%] Built target SQLiteStorage
[ 39%] Built target ADSL
[ 44%] Built target ACL
[ 44%] Built target InterruptModel
[ 44%] Built target Captions
[ 44%] Built target AVSGatewayManager
[ 44%] Built target Alexa
[ 46%] Built target EqualizerImplementations
[ 46%] Built target ContextManager
[ 48%] Built target PlaylistParser
[ 51%] Built target KWD
[ 51%] Built target SystemSoundPlayer
[ 53%] Built target SpeechEncoder
[ 55%] Built target SpeakerManager
[ 58%] Built target AudioResources
[ 58%] Built target ApiGateway
[ 60%] Built target AudioPlayer
[ 60%] Built target Bluetooth
[ 62%] Built target Equalizer
[ 62%] Built target InteractionModel
[ 65%] Built target Notifications
[ 67%] Built target PlaybackController
[ 67%] Built target SpeechSynthesizer
[ 67%] Built target TemplateRuntime
[ 69%] Built target CBLAuthDelegate
[ 72%] Built target SynchronizeStateSender
[ 74%] Built target MediaPlayer
[ 74%] Built target CertifiedSender
[ 76%] Built target AFML
[ 76%] Built target Endpoints
[ 76%] Built target SENSORY
[ 79%] Built target ExternalMediaPlayer
[ 83%] Built target DeviceSettings
[ 83%] Built target CapabilitiesDelegate
[ 83%] Built target KeywordDetectorProvider
[ 83%] Built target AIP
[ 86%] Built target Alerts
[ 90%] Built target AVSSystem
[ 93%] Built target DoNotDisturbCA
[ 95%] Built target DefaultClient
[ 95%] Building CXX object SampleApp/src/CMakeFiles/LibSampleApp.dir/UIManager.cpp.o
[ 95%] Building CXX object SampleApp/src/CMakeFiles/LibSampleApp.dir/InteractionManager.cpp.o

cnoorman, 2020-11-16:
I had the same problem. With me it was caused by overheating of the Pi SOC.
Remedy: 
1 - Provide good cooling of the processor.
or
2 - Only use one core to build the Alexa software.
In the current software the number of cores is limited to two by commands such as: `make SampleApp -j2`.
After you have run `sudo /home/pi/Assistants-Pi/scripts/installer.sh` for the first time a number of command files (*.sh) are created that contain the `make` command. 
To find these use: `find /home/pi/Assistants-Pi/ -type f -name "*.sh" -exec grep -l "j2" {} \;` 
In my case five files were found:
> /home/pi/Assistants-Pi/Alexa/setup.sh
> /home/pi/Assistants-Pi/Alexa/avs-device-sdk/tools/Install/setup.sh
> /home/pi/Assistants-Pi/Alexa/avs-device-sdk/tools/Install/pi.sh
> /home/pi/Assistants-Pi/Alexa/test.sh
> /home/pi/Assistants-Pi/Alexa/pi.sh

In all these files I replaced -j2 by -j1 and ran the command `sudo /home/pi/Assistants-Pi/scripts/installer.sh` again. 
Now the processor load was low enough to avoid overheating and to successfully complete the installation. 
I'm not sure which of the *.sh files are actually used, but who cares. After these changes of course it will take even longer than before to finish the install. 

PS
Note that you also have to issue the command: `sudo chmod +x /home/pi/Assistants-Pi/Alexa/start.sh` to enable proper start-up of the _alexa.service_ after reboot.

Success...

iamsujay, 2020-11-16:
Problem solved using this > []https://github.com/shivasiddharth/Assistants-Pi/issues/139#issuecomment-641235505

closed 2020-11-16 by iamsujay