shivasiddharth/Assistants-Pi #153: Google Assistant is not responding to voice input

drkostas, 2020-12-06:
Hi @shivasiddharth ! I successfully finished the installation of the project on my RaspberryPi 3 for the Google Assistant only, and the assistant **doesn't respond to my voice**.

- I configured it to use USB input (from USB headphones) and Jack output (speakers). When I am running  `audio-test.sh` it finishes with no problems (I hear the sound on the speakers and the recording is played back to me correctly). 
I only get this warning: `Warning: rate is not accurate (requested = 16000Hz, got = 48000Hz) please, try the plug plugin`

- The outputs of `/etc/asound.conf` and `~/.asoundrc` are the same:
    ```
    pcm.dsnooper {
        type dsnoop
        ipc_key 816357492
        ipc_key_add_uid 0
        ipc_perm 0666
        slave {
            pcm "hw:1,0"
            channels 1
        }
    }

    pcm.!default {
            type asym
            playback.pcm {
                    type plug
                    slave.pcm "hw:0,0"
            }
            capture.pcm {
                    type plug
                    slave.pcm "dsnooper"
            }
    }
    ```
- The outputs of `aplay`, `arecord` and `cat /proc/asound/cards` are the following:
    ```
    $ aplay -l
    **** List of PLAYBACK Hardware Devices ****
    card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
      Subdevices: 6/7
      Subdevice #0: subdevice #0
      Subdevice #1: subdevice #1
      Subdevice #2: subdevice #2
      Subdevice #3: subdevice #3
      Subdevice #4: subdevice #4
      Subdevice #5: subdevice #5
      Subdevice #6: subdevice #6
    card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
      Subdevices: 1/1
      Subdevice #0: subdevice #0
    card 0: ALSA [bcm2835 ALSA], device 2: bcm2835 IEC958/HDMI1 [bcm2835 IEC958/HDMI1]
      Subdevices: 1/1
      Subdevice #0: subdevice #0
    card 1: Device [USB PnP Audio Device], device 0: USB Audio [USB Audio]
      Subdevices: 1/1
      Subdevice #0: subdevice #0

    $ arecord -l
    **** List of CAPTURE Hardware Devices ****
    card 1: Device [USB PnP Audio Device], device 0: USB Audio [USB Audio]
      Subdevices: 1/1
      Subdevice #0: subdevice #0

    $ cat /proc/asound/cards
     0 [ALSA           ]: bcm2835_alsa - bcm2835 ALSA
                          bcm2835 ALSA
     1 [Device         ]: USB-Audio - USB PnP Audio Device
                          USB PnP Audio Device at usb-3f980000.usb-1.1.2, full speed

    ```
- And finally, when starting `/home/pi/Assistants-Pi/Google-Assistant/src/main.py` manually, the output is the following:
    ```
    $ /home/pi/env/bin/python -u /home/pi/Assistants-Pi/Google-Assistant/src/main.py --project_id '<My project id>' --device_model_id '<My model id>'

[... 63 lines cut ...]
    ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
    ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
    ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
    ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
    ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
    ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
    Cannot connect to server socket err = No such file or directory
    Cannot connect to server request channel
    jack server is not running or cannot be started
    JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
    JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
    /home/pi/env/lib/python3.5/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
      warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
    device_model_id: raspberry-roogle-assistant-raspistant-wxr2k4
    device_id: DDAA8FD2FFD12EAD4DC621C4FC87CE07

    ON_MUTED_CHANGED:
      {"is_muted": false}

    ON_START_FINISHED

    ON_MEDIA_STATE_IDLE
    ```

Can you help me identify the issue? Thanks in advance!

EmadMokhtar, 2021-01-28:
I'm facing the same issue.

shivasiddharth, 2021-01-31:
If you are on the Dec release, try the following:

`sudo nano /etc/pulse/client.conf.d/00-disable-autospawn.conf`

Comment the autospawn (add a # infront of the autospawn line).

Restart and see.