shivasiddharth/Assistants-Pi #158: Google Assistant on Raspberry Pi Zero

EmadMokhtar, 2021-01-31:
I followed your step and install the driver using your script. The google assistant service is running, I can hear the welcome (Hello) but after that nothing.

``` bash
pi@raspberrypi:~ $ sudo systemctl status google-assistant.service
● google-assistant.service - Google Assistant Voice Service
   Loaded: loaded (/lib/systemd/system/google-assistant.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2021-01-29 14:38:57 GMT; 2min 37s ago
 Main PID: 828 (python)
    Tasks: 21 (limit: 881)
   CGroup: /system.slice/google-assistant.service
           └─828 /home/pi/env/bin/python -u /home/pi/Assistants-Pi/Google-Assistant/src/main.py --project_id peerless-voyage-299414 -

Jan 29 14:39:07 raspberrypi python[828]: jack server is not running or cannot be started
Jan 29 14:39:07 raspberrypi python[828]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Jan 29 14:39:08 raspberrypi python[828]: /home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py:90: Deprecati
Jan 29 14:39:08 raspberrypi python[828]:   warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
Jan 29 14:39:08 raspberrypi python[828]: device_model_id: peerless-voyage-299414-officeamp-m8ak3t
Jan 29 14:39:08 raspberrypi python[828]: device_id: 9E0C8CA68AEDCFDB17FC8862CE8F9A7C
Jan 29 14:39:09 raspberrypi python[828]: ON_MUTED_CHANGED:
Jan 29 14:39:09 raspberrypi python[828]:   {"is_muted": false}
Jan 29 14:39:12 raspberrypi python[828]: ON_MEDIA_STATE_IDLE
Jan 29 14:39:12 raspberrypi python[828]: ON_START_FINISHED
```

EmadMokhtar, 2021-01-31:
Update:

I installed the reseeed HAT on Raspberry Pi 3 and wake word is working. I wondering is it the processor power of the Zero can't handle this? and if yes, how to use the Raspberry Pi zero with push to talk via the button provided by the HAT?

shivasiddharth, 2021-01-31:
Pi Zero has an upstream audio issue that has not been fixed yet. Thats why i have left the Pi Zero pushbutton script untouched for sometime in GassistPi and have also removed the support for Pi Zero in this git. 

Try this:
`sudo nano /etc/pulse/client.conf.d/00-disable-autospawn.conf`
Comment the autospawn (add a # infront of the autospawn line).
Also run this command
`sudo rm /lib/udev/rules.d/91-pulseaudio-rpi.rules`
Restart and see.

EmadMokhtar, 2021-02-03:
I will give it a try this weekend.

okp81, 2021-02-08:
I also facing the same issue even after try this

EmadMokhtar, 2021-02-25:
I will close this issue because I will stop using the zero borad

closed 2021-02-25 by EmadMokhtar