shivasiddharth/Assistants-Pi #159: how to use Kitt Ai for default amazon alexa wake engine

okp81, 2021-02-08:
i want to use Kitt ai wake word engine instead of default alexa engine

okp81, 2021-02-08:
@shivasiddharth i used your installation guide. I want to change my wake word engine to Kitt Ai or sensory for alexa using raspberry pi. please help me out from this sir.

shivasiddharth, 2021-02-08:
There nothing called Alexa Engine. There is just Sensory Engine and Kitt AI engine. This project uses Sensory Engine with "Alexa" as the wakeword by default and it should work straight away (The wakeword has already been incorporated).

closed 2021-02-08 by shivasiddharth

okp81, 2021-02-08:
[image: 24caa35d8d58d0a12979a521db250547e2a78124]Sir
i know that sir. For raspberry pi zero w
sensory
​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​engine
does not work shows illegal so that i want to use kitt Ai. so please give
steps to change sensory engine to Kitt Ai. Please reply soon sir. Please
help me out from this sir

okp81, 2021-02-08:
[image: f3975fcf1dcee7e319491e2fd3d632cc0eece9a3]
​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​Sir
  i installed GASSIST pi in raspberry pi zero w. after installing the
Google assistant doesn't respond for voice it shows ON_MEDIA_STATE_IDLE

okp81, 2021-02-08:
[image: 0ff1f708c66e94ef6ba73eb7b026e8600ab2dc21]please help me out from
this sir. i tried lot of things for past 2 days but nothing happened.
please help me out from this sir.
​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​

shivasiddharth, 2021-02-08:
There are some fake youtube videos that claim to run voice assistants on Pi Zero. I do not support Pi Zero Boards. Pi Zero is not ideal for voice assistant applications.

okp81, 2021-02-08:
[image: 631295d8fbf8acdf06301b58a68dc6ba882040a3]ok sir please atleast tell
me the steps
​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​.
Atleast i tried that. because without sleep i tried for past 2 days. please
tell steps to use Kitt Ai sir.

shivasiddharth, 2021-02-08:
Stop wasting your time using Pi Zero. It will take 7-8 hours to compile Alexa on Pi Zero.  What does the Hardware Requirement say here https://developer.amazon.com/en-US/docs/alexa/avs-device-sdk/raspberry-pi.html ? People are giving false directions in YouTube for making money. I have been in this for long enough to give correct directions to people.

There is no directions or steps because KITT.AI snowboy was shutdown down last year. Their official website has also been shut down. Check their git page https://github.com/Kitt-AI/snowboy/ Amazon has removed support for snowboy as well.

okp81, 2021-02-08:
Thank you for your response.