shivasiddharth/Assistants-Pi #162: Alexa is not responding to my voice (Wake word)

EmadMokhtar, 2021-02-25:
I installed the Alexa (with some tweaks to support the stream and pulseAudio). I ran the `startsample.sh` and this is the output

```
2021-02-25 16:56:29.675 [  1] I sdkVersion: 1.19.1
configFile /home/pi/Assistants-Pi/Alexa/build/Integration/AlexaClientSDKConfig.json
Running app with log level: INFO
2021-02-25 16:56:30.692 [  1] I DeviceInfo:create:result=skipDeviceType,key=deviceType
2021-02-25 16:56:30.692 [  1] I DeviceInfo:create:result=skipFriendlyName,key=friendlyName
2021-02-25 16:56:30.692 [  1] I DeviceInfo:create:result=skipRegistrationKey,key=registrationKey
2021-02-25 16:56:30.692 [  1] I DeviceInfo:create:result=skipProductIdKey,key=productIdKey
2021-02-25 16:56:30.734 [  1] I Setting:handleSetValue:value=false3990065800
2021-02-25 16:56:30.736 [  1] I Setting:handleSetValue:value="NONE"3990065800
2021-02-25 16:56:30.737 [  1] I Setting:handleSetValue:value="NONE"3990065800
2021-02-25 16:56:30.738 [  1] I Setting:handleSetValue:value="NONE"3990065800
2021-02-25 16:56:30.739 [  1] I Setting:handleSetValue:value="Europe/Amsterdam"3990065800
2021-02-25 16:56:30.741 [  1] I Setting:handleSetValue:value={}3990065800
2021-02-25 16:56:30.743 [  5] I LocaleAssetsManager:changeAssets:Locale=["en-US"],WakeWords=["ALEXA"]
2021-02-25 16:56:30.766 [  1] I SpeakerManager:mapCreated:numSpeakerVolume=6,numAlertsVolume=2
2021-02-25 16:56:30.786 [  1] W UUIDGeneration:low entropy on call to generate UUID:current entropy=32
#################################################
#       NOTIFICATION INDICATOR STATE: OFF       #
#################################################

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
2021-02-25 16:56:35.313 [  1] I SensoryKeywordDetector:License expires on 28 Apr 2021 00:00:00 GMT
2021-02-25 16:56:35.313 [  1] I SensoryKeywordDetector:Sensory library license does not expire for at least 60 more days.
###########################
#       Authorized!       #
###########################

2021-02-25 16:56:35.457 [  1] I PortAudioMicrophoneWrapper:startStreamingMicrophoneData
#############################
#       Connecting...       #
#############################

                  #    #     #  #####      #####  ######  #    #              
                 # #   #     # #     #    #     # #     # #   #               
                #   #  #     # #          #       #     # #  #                
               #     # #     #  #####      #####  #     # ###                 
               #######  #   #        #          # #     # #  #                
               #     #   # #   #     #    #     # #     # #   #               
               #     #    #     #####      #####  ######  #    #              
                                                                              
       #####                                           #                      
      #     #   ##   #    # #####  #      ######      # #   #####  #####      
      #        #  #  ##  ## #    # #      #          #   #  #    # #    #     
       #####  #    # # ## # #    # #      #####     #     # #    # #    #     
            # ###### #    # #####  #      #         ####### #####  #####      
      #     # #    # #    # #      #      #         #     # #      #          
       #####  #    # #    # #      ###### ######    #     # #      #          

       SDK Version 1.19.1

       WARNING! THIS DEVICE HAS BEEN COMPILED IN DEBUG MODE.

       RELEASING A PRODUCTION DEVICE IN DEBUG MODE MAY IMPACT DEVICE PERFORMANCE,
       DOES NOT COMPLY WITH THE AVS SECURITY REQUIREMENTS,
       AND COULD RESULT IN SUSPENSION OR TERMINATION OF THE ALEXA SERVICE ON YOUR DEVICES.

+----------------------------------------------------------------------------+
|                                  Options:                                  |
| Wake word:                                                                 |
|       Simply say Alexa and begin your query.                               |
| Tap to talk:                                                               |
|       Press 't' and Enter followed by your query (no need for the 'Alexa').|
| Hold to talk:                                                              |
|       Press 'h' followed by Enter to simulate holding a button.            |
|       Then say your query (no need for the 'Alexa').                       |
|       Press 'h' followed by Enter to simulate releasing a button.          |
| Stop an interaction:                                                       |
|       Press 's' and Enter to stop an ongoing interaction.                  |
| Privacy mode (microphone off):                                             |
|       Press 'm' and Enter to turn on and off the microphone.               |
| Playback Controls:                                                         |
|       Press '1' for a 'PLAY' button press.                                 |
|       Press '2' for a 'PAUSE' button press.                                |
|       Press '3' for a 'NEXT' button press.                                 |
|       Press '4' for a 'PREVIOUS' button press.                             |
| Settings:                                                                  |
|       Press 'c' followed by Enter at any time to see the settings screen.  |
| Speaker Control:                                                           |
|       Press 'p' followed by Enter at any time to adjust speaker settings.  |
| Firmware Version:                                                          |
|       Press 'f' followed by Enter at any time to report a different        |
|       firmware version.                                                    |
| Info:                                                                      |
|       Press 'i' followed by Enter at any time to see the help screen.      |
| Reset device:                                                              |
|       Press 'k' followed by Enter at any time to reset your device. This   |
|       will erase any data stored in the device and you will have to        |
|       re-register your device.                                             |
|       This option will also exit the application.                          |
| Reauthorize device:                                                        |
|       Press 'z' followed by Enter at any time to re-authorize your device. |
|       This will erase any data stored in the device and initiate           |
|       re-authorization.                                                    |
|                                                                            |
| Quit:                                                                      |
|       Press 'q' followed by Enter at any time to quit the application.     |
+----------------------------------------------------------------------------+

2021-02-25 16:56:35.496 [ 10] W UUIDGeneration:low entropy on call to generate UUID:current entropy=32
########################################
#       Alexa is currently idle!       #
########################################

2021-02-25 16:56:36.702 [  6] W UUIDGeneration:low entropy on call to generate UUID:current entropy=32
```

Note: The Google assistance is working fine and can reply to me.

shivasiddharth, 2021-02-26:
Does pressing on the keys work ?

EmadMokhtar, 2021-02-27:
I'm trying to press the keys on the SSH session, nothing is happening.

shivasiddharth, 2021-02-27:
You should be physically connected.

VNMan148, 2021-03-17:
Pulse Audio on the pi by default is set for single user, that is why you have problem with Alexa. You need to set Pulse Audio to system-wide service.

Search for how to do this on the Alexa trouble shooting side, note that you just need to "convert" pulse Audio to service and nothing else, ie. you do not need to create extra user.

rajat8888, 2021-04-20:
@EmadMokhtar Did you find a solution ?

commit 23b51651 (2021-04-20): Update alexa.service

Fixes #164  Fixes #162
closed 2021-04-20 by shivasiddharth