shivasiddharth/Assistants-Pi #164: Wake word seems broken on auto startup

adarshjayan7574, 2021-04-02:
**Problem**

When executing Alexa start.sh script or startsample.sh, Alexa is working as expected, it responds to my voice commands flawlessly.

But on reboot/fresh boot I can hear only the Alexa startup tones, it's not responding to my commands, it should be an issue with some of my configuration or something,

"sudo systemctl enable alexa.service" is the command I executed for running  Alexa on boot

I can share more informations, but please share the commands for taking necessary logs, thank you..

rajat8888, 2021-04-20:
@adarshjayan7574 Did you find a solution ?

adarshjayan7574, 2021-04-20:
Yes i updated the raspbian os to the latest version and Alexa started working again, but still google assistant is not running properly with alexa on boot, but the sample app works

commit 23b51651 (2021-04-20): Update alexa.service

Fixes #164  Fixes #162
closed 2021-04-20 by shivasiddharth